DerAndere1/Marlin2ForPipetBot #1: [BUG][bf2_6axis_dev] Axes I, J and K do not move

DerAndere1, 2019-02-20:
### Description

With the files from the bf_6axes_dev branch downloaded today, when specifying `#define NON_E_AXES = 4` or `#define NON_E_AXES = 5` or `#define NON_E_AXES = 6` only axes XYZ can be moved by issuing G-code G1.   

### Steps to Reproduce

1. Use the original PipetBot-A8 as detailed at https://it-by-derandere.blogspot.com/p/pipetbot-a8.html.
2. visit https://github.com/DerAndere1/Marlin/tree/bf2_6axis_dev and download the files from that branch.
3. Copy the content of Marlin/src/config/examples/DerAndere/PipetBot-A8 to Marlin/ and confirm the replacement of the existing files Marlin/Configuration.h and Marlin/Configuration_adv.h
4. Upload the configured firmware from step 3 to the microcontroller of the PipetBot-A8
5. Issue G-code command `G28 I` or `G28` G-code commands like `G1 I10 F100` or `G1 X5 Y6 Z3 I40 J50 K60 E70 F100` (e.g. via USB serial connection using either printrun or Universal G-code sender (UGS))

**Expected behavior:** 
The stepper motors connected to the pins for AXIS_I, AXIS_J, AXIS_K should move similar to those connected to pins for AXIS_X, AXIS_Y and AXIS_Z.
 
**Actual behavior:** 
Only stepper motors connected to the pins for AXIS_X, AXIS_Y and AXIS_Z move. Steppers for I,J and K cannot be moved manually without applying high load. This indicates that these steppers are enabled.

#### Additional Information

None

DerAndere1, 2019-02-21:
This might be related to #2

vector76, 2019-09-19:
Hello I am interested in this functionality and I am unclear on the current status.  I can see multiple branches, the most recent of which was touched in June, but it appears these are intending to get E-axis homing merged into upstream.

I thought I might rebase the main commit from bf2_6axis_dev on to the current Marlin but I am not so fluent in conflict resolution so I might have to work off of the slightly stale bf2_6axis_dev.  What should I expect?

My alternative, which might "work" but which I don't like, is to perform CAM with A and B axes and then post-process into a fake mixing extruder, enabling MIXING_EXTRUDER and performing M165 between each movement.  I'm not sure if the mix factors can be negative and in any case this would be an ugly approach.

Anyway the main question is, if I were to use this bf2_6axis_dev as a starting point, what should I expect and is there anything else I should know before trying to go forward?

DerAndere1, 2020-01-07:
I never could make it work. This was a toy project for me to learn programming, so I did not research but there is a hight chance that 8 bit AVRs are to slow to process more than 4 Axes (including extruder) in parallel. Note that my Idea was to support cartesian robots where I,J,K move additional steppers that do not influence positioning of the tool in any way. If you need support for a specific kinematic model (e.g. 5 axis arm), I don't know much code you have to add. 
If you need more than 3 axes for a serious project , currently the best is to base your project on Smoothieware + their supported hardware. Smoothieware has support for 6 axes. Also, check if https://github.com/fra589/grbl-Mega-5X/ meets your requirements. To give more feedback, I would need to know what your requirements are: What type of kinematic model? do you need endstops on all axes? do you need auto-homing on all axes? do you have a printer or engraver where extrusion/laser power needs to be calculated based on XYZ-movement speed or is GRBL-5X enough?

vector76, 2020-01-09:
Thank you for the clarification.  I don't really "need" 5-axis, I am mostly curious to play with it.  I think I will try 3+2 and postprocess A and B into fake extruder movements.  I have no extruder so I have 2 free stepper drivers anyway, and I am wanting to play with 3-axis milling with 2 rotary axes.

I appreciate the work that you put into this.  It's definitely a challenge.  In the end I think I can achieve my objective with other means.

DerAndere1, 2020-01-14:
If grbl-Mega-5X or Smoothieware is not a solution, Repetier Firmware 2.0 (https://github.com/repetier/Repetier-Firmware/tree/dev2) supports 7 axes and CNC milling tools. It is currently in development but you can already try it on Ardunino Due-based boards.

nguyennhattam268, 2020-02-10:
Hello, I control the axes i, j, k but they run very slowly :-D there is still much work to do too

DerAndere1, 2020-02-10:
@nguyennhattam268 Hi, thats great news. Could You send me your source code for inspection? If you don't want to share certain parts of your code, leave those parts out. If you are not familiar with github, you can put the source code into a zip-compressed folder and drag and drop that zip-compressed folder into a comment here.

nguyennhattam268, 2020-02-11:
Yes, I will post here because I am not good at English
2 is that I don't have much time for this because I have to go to work
I did 6asix on Repetier_firmware last year but I have to use arduino due and RAMPS_FD
I wanted to use ramps1.4 and marlin so I made it with passion
This is my test video      and I only changed a few parameters in the stepper file the most important is to change LOOP_NUM_AXIS_N in CONFIGRUATION_STORE.CPP
https://www.youtube.com/channel/UCnRniVi-9ObhQ87oIsQxihQ?view_as=subscriber

DerAndere1, 2020-02-12:
@nguyennhattam268 Can you copy of your stepper.cpp file into a comment here?

nguyennhattam268, 2020-02-13:
O of course I will send it right here
stepper_indirection.cpp
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * stepper_indirection.cpp
 *
 * Stepper motor driver indirection to allow some stepper functions to
 * be done via SPI/I2c instead of direct pin manipulation.
 *
 * Part of Marlin
 *
 * Copyright (c) 2015 Dominik Wenger
 */

#include "stepper_indirection.h"

#include "../inc/MarlinConfig.h"

#include "stepper.h"

#if HAS_DRIVER(L6470)
  #include "L6470/L6470_Marlin.h"
#endif

//
// TMC26X Driver objects and inits
//
#if HAS_DRIVER(TMC26X)
  #include <SPI.h>

  #ifdef STM32F7
    #include "../HAL/HAL_STM32F7/TMC2660.h"
  #else
    #include <TMC26XStepper.h>
  #endif

  #define _TMC26X_DEFINE(ST) TMC26XStepper stepper##ST(200, ST##_CS_PIN, ST##_STEP_PIN, ST##_DIR_PIN, ST##_MAX_CURRENT, ST##_SENSE_RESISTOR)

  #if AXIS_DRIVER_TYPE_X(TMC26X)
    _TMC26X_DEFINE(X);
  #endif
  #if AXIS_DRIVER_TYPE_X2(TMC26X)
    _TMC26X_DEFINE(X2);
  #endif
  #if AXIS_DRIVER_TYPE_Y(TMC26X)
    _TMC26X_DEFINE(Y);
  #endif
  #if AXIS_DRIVER_TYPE_Y2(TMC26X)
    _TMC26X_DEFINE(Y2);
[... 1080 lines cut ...]
    #endif
    #if AXIS_DRIVER_TYPE_Z3(L6470)
      _L6470_INIT_CHIP(Z3);
    #endif
    #if AXIS_DRIVER_TYPE_E0(L6470)
      _L6470_INIT_CHIP(E0);
    #endif
    #if AXIS_DRIVER_TYPE_E1(L6470)
      _L6470_INIT_CHIP(E1);
    #endif
    #if AXIS_DRIVER_TYPE_E2(L6470)
      _L6470_INIT_CHIP(E2);
    #endif
    #if AXIS_DRIVER_TYPE_E3(L6470)
      _L6470_INIT_CHIP(E3);
    #endif
    #if AXIS_DRIVER_TYPE_E4(L6470)
      _L6470_INIT_CHIP(E4);
    #endif
    #if AXIS_DRIVER_TYPE_E5(L6470)
      _L6470_INIT_CHIP(E5);
    #endif
  }

#endif // L6470

nguyennhattam268, 2020-02-13:
stepper.cpp
#include "stepper.h"

Stepper stepper; // Singleton

#ifdef __AVR__
  #include "speed_lookuptable.h"
#endif

#include "endstops.h"
#include "planner.h"
#include "motion.h"

#include "../core/macros.h"
#include "temperature.h"
#include "../lcd/ultralcd.h"
#include "../core/language.h"
#include "../gcode/queue.h"
#include "../sd/cardreader.h"
#include "../Marlin.h"
#include "../HAL/shared/Delay.h"

#if MB(ALLIGATOR)
  #include "../feature/dac/dac_dac084s085.h"
#endif

#if HAS_DIGIPOTSS
  #include <SPI.h>
#endif

#if ENABLED(MIXING_EXTRUDER)
  #include "../feature/mixing.h"
#endif

#if FILAMENT_RUNOUT_DISTANCE_MM > 0
  #include "../feature/runout.h"
#endif

#if HAS_DRIVER(L6470)
  #include "../libs/L6470/L6470_Marlin.h"
#endif

// public:

#if HAS_EXTRA_ENDSTOPS || ENABLED(Z_STEPPER_AUTO_ALIGN)
  bool Stepper::separate_multi_axis = false;
#endif

#if HAS_MOTOR_CURRENT_PWM
  uint32_t Stepper::motor_current_setting[3]; // Initialized by settings.load()
#endif

// private:

block_t* Stepper::current_block; // (= NULL) A pointer to the block currently being traced Một con trỏ tới khối hiện đang được theo dõi

uint8_t Stepper::last_direction_bits, // = 0
        Stepper::axis_did_move; // = 0

bool Stepper::abort_current_block;

#if DISABLED(MIXING_EXTRUDER) && EXTRUDERS > 1
  uint8_t Stepper::last_moved_extruder = 0xFF;
#endif

#if ENABLED(X_DUAL_ENDSTOPS)
  bool Stepper::locked_X_motor = false, Stepper::locked_X2_motor = false;
#endif
#if ENABLED(Y_DUAL_ENDSTOPS)
  bool Stepper::locked_Y_motor = false, Stepper::locked_Y2_motor = false;
[... 3153 lines cut ...]
      SERIAL_CHAR('0' + READ(E3_MS1_PIN));
      SERIAL_CHAR('0' + READ(E3_MS2_PIN));
      #if PIN_EXISTS(E3_MS3)
        SERIAL_ECHOLN((int)READ(E3_MS3_PIN));
      #endif
    #endif
    #if HAS_E4_MICROSTEPS
      SERIAL_ECHOPGM("E4: ");
      SERIAL_CHAR('0' + READ(E4_MS1_PIN));
      SERIAL_CHAR('0' + READ(E4_MS2_PIN));
      #if PIN_EXISTS(E4_MS3)
        SERIAL_ECHOLN((int)READ(E4_MS3_PIN));
      #endif
    #endif
    #if HAS_E5_MICROSTEPS
      SERIAL_ECHOPGM("E5: ");
      SERIAL_CHAR('0' + READ(E5_MS1_PIN));
      SERIAL_ECHOLN((int)READ(E5_MS2_PIN));
      #if PIN_EXISTS(E5_MS3)
        SERIAL_ECHOLN((int)READ(E5_MS3_PIN));
      #endif
    #endif
  }

#endif // HAS_MICROSTEPS

DerAndere1, 2020-02-17:
Thanks, I cannot find the changes you made. I guess you have to change something in configuration.h . But I would need your complete Marlin source code as a zip to help you. You can make a new comment here using the github web interface. While typing, you will see a message "Attach files by dragging & dropping, selecting or pasting them" Below the text input field. Click on that message and a file browser will open.

hobiseven, 2020-02-21:
@nguyennhattam268 
Could you post the full marlin code either in a git repository, or here, as a zipfile / drag and drop below please? This would help me, you, and DerAndere1! 

Bạn có thể gửi mã marlin đầy đủ trong kho git hoặc ở đây, dưới dạng zipfile / kéo và thả bên dưới không? Điều này sẽ giúp tôi, bạn và DerAndere1!

hobiseven, 2020-02-22:
@DerAndere1 I know this will not give 6 axis capabilities, but it seems latest 2.0.1 upwards Marlin release has X2 capability, and that X2 can be fully controlled by Gcode. Does it means that this could be the real 4 axis I look for? 
There is a Y2 also, but this one is "just" a Y replication.

DerAndere1, 2020-02-23:
@hobiseven : DUAL_X_CARRIAGE can be used for two independent extruders (IDEX). In the following discussion you can find an example config: https://github.com/MarlinFirmware/Marlin/issues/12464 . However you will have to switch between extruders using the G-code comand `T0` for tool 0 and `T1` for tool 1. So I guess you cannot move axes X1 and X2 simultaneously. If you have more questions regarding existing options, please refer to the following information: https://github.com/MarlinFirmware/Marlin/issues/14694

hobiseven, 2020-02-23:
@DerAndere1 
I rebuilt the files posted above, as there was many tiny format changes in the comments, highlighting differences, and also some other ones which are rather strange to me like removing the _ in front of AVR ... 
You will see that he added a few lines of code in both cPP files. 
[New_files.zip](https://github.com/DerAndere1/Marlin/files/4242637/New_files.zip)

DerAndere1, 2020-02-24:
I added the changes I found plausible. Note that I started adding 6 axis support for a minimal setup (no Trinamic drivers, no delta kinematics, no CoreXY kinematics, no EEPPROM, no muti endstops). The latest changes in stepper.cpp will have to be acompanied with changes in stepper.h for multi enstops and other parts might have become rather more broken than fixed. 

@nguyennhattam268 : I added you as the author of the last commit in the bf2_6axis_dev branch

hobiseven, 2020-02-24:
I started to bring in your code into the Alfawise STM32 code. Will report to you once I will download the code, and give it a try. At least, I will not have a CPU speed issue. it runs at 72Mhz. 
I renamed all I,J, K axis to U,V,W as those are meant to be secondary axis, and I also saw the issue posted about the names conflicting with some Gcode commands.

DerAndere1, 2020-02-24:
@hobiseven thanks that you take over and incorporate the 6 axis code into a more updated code basis. Don't expect it to work. It will require more work to get the basic test setup going and a huge amount of work to get anything beyond G28 and G1 working with 6 axes. Note that most other firmwares (g2core, smoothie, RRF, Repetier Firmware, grbl-5X etc.) already have 6 axes working, so consider using one of those

hobiseven, 2020-02-25:
@DerAndere1 Thank you for the clear warning. Actually, G28 is definitely needed to me, and my Gcode is G1 based. 
There are a few commands with no 6 axis arguments. Let's see wether or not this will be complicated to get this to work.  Typical Gcode is as here : 

( Plan XY )
G17
( SET LENGTH UNITS )
G21
( SET DISTANCE MODE )
G91
( SET CUTTER COMPENSATION )
G40
( SET TOOL LENGTH OFFSET )
G49
( SET PATH CONTROL MODE )
G64
( SET FEED RATE MODE )
G94

( Turn on hotwire )
M3
( Set wire heat using PWM spindle speed [Value relative in %] )
S10
( Move max speed )
F100
G1 X0 Y20 U0 V20
( Pause [millisec] )
G4 P2000
( Move cut speed )
F10
G1 X10 Y0 U10 V0
( Move cut speed )
G1 X0,482 Y0,694 U0,482 V0,694
G1 X0,911 Y0,792 U0,911 V0,792
.......

I will check wether or not the header commands are all needed... Let's see. but looking at the youtube where we see the A axis turning, ( with a wrong speed), I would hope I am not signing for 10 years effort. I know very well the alfawise board, and Marlin on it.

hobiseven, 2020-02-28:
@DerAndere1  I just saw that you have an other branch called marlin2forpipetbot where apparently you use E0 as an axis with an endstop. 
When you make a g28 do you get x y z e homing? That could be enough for me ....
 i am half way importing your changes for the 6 axis in marlin 2.0.4.2 . I might need a bit of help to change a few macros that I have a hard time to read...I will open my repo as soon as it compiles.

DerAndere1, 2020-02-28:
hi @hobiseven. I am happy to help with the translation of the macros to "human readable". Some are really cryptic, indeed :D  Marlin2ForPipetbot branch should work as intended (endstops + G28 for xyze). I probably removed some temperature code because it is used as lab robot firmware. I also added some artificial(?) feature restrictions using sanity checks to reduce code I had to change. I thought about pointing you to that branch but then thought it wouldn't be the a clean solution. You would need a post processor to translate G1 Xxxx Yyyy Uuuu Vvvv Fnnn syntax into the G1 Xxxx Yyyy Zzzz Eeee Fnnn syntax understood by Marlin2ForPipetBot.

DerAndere1, 2020-02-28:
@hobiseven: If you need full support for RS-274 standard compliant G-code I still recommend add support for your board to this HAL-ified grbl fork: https://github.com/terjeio/grblHAL

hobiseven, 2020-02-29:
@DerAndere1 Actually I believe G40 and G49 are not needed . The machine is a hot wire cutter and the only requirement is to position the wire at both ends , nothing more. And I do not even need a gcode post processor as the axis names are selectable in jedicut. I have a long rainy week end in front of me to look at this more in depth.

DerAndere1, 2020-02-29:
to get a better overview of the changes that enable e axis homing, see commits https://github.com/MarlinFirmware/Marlin/pull/13076/commits from my rejected PR . There might be one or two bugs in the cleaned up code but those idease are the basis of my working Marlin2ForPipetBot branch

hobiseven, 2020-02-29:
Thank you! I will check the pr and your code

hobiseven, 2020-02-29:
Indeed, the code changes in your PR https://github.com/MarlinFirmware/Marlin/pull/13076/commits 
 are  lot simpler than on the full blown 6 axis compared the last  up to date Marlin! Serial.cpp started to be a problem, and there was quite some changes in the Marlin code since you worked on the 6 axis, which makes thing even more complicated. 
I think E endstop will do the job for me! I will definitely restart on that one.

GabrielBeraldo, 2020-03-01:
Hello guys, some time ago i managed to fix all the issues, including eeprom storing and calculate the correct feedrate for all axes, in Cartesian kinematics. I'm using a 12864 lcd display and all the functions for this display was implemented too! 
Due a personal project and lack of experience in github i changed to my personal repository, now its time to share!

I need some help to port this to the current version of marlin, and this way contribute to the community! 

https://github.com/GabrielBeraldo/Marlin/tree/bf2_6axis_dev

hobiseven, 2020-03-01:
@GabrielBeraldo  hehe this looks a great news to me! So you say that you got this to work including g a 12864 lcd... this is nice as our alfawise code uses an 2x zoom of the regular 128x64 marlin screen... I think they call it dogm. 
Well seems I have many options now. G28 and g1 working fine for your code??

hobiseven, 2020-03-01:
I hope I do check for differences correctly....
The Alfawise maintainned repo is the one of Tpruvot... 

https://github.com/tpruvot/Marlin/compare/2.0.4.4-longer3D...GabrielBeraldo:bf2_6axis_dev

@tpruvot is this the right comparaison method. 52 files...

GabrielBeraldo, 2020-03-01:
@hobiseven Yes! fully functional G28 and G1. I assume, but not ensure, that all the other commands will work too! 
This development took me a LOT of debugging with the logic analyzer to figure out where the problem was, not complicate to fix but really hard to find.

GabrielBeraldo, 2020-03-01:
I made several mods that will not affect the movimentation factor, like graphical ones.
The main differences should be checked in the files that receive, process and execute the moving actions, as well as the endstops and peripheral files linked to G28 execution.

GabrielBeraldo, 2020-03-01:
> I hope I do check for differences correctly....
> The Alfawise maintainned repo is the one of Tpruvot...
> 
> [tpruvot/Marlin@2.0.4.4-longer3D...GabrielBeraldo:bf2_6axis_dev](https://github.com/tpruvot/Marlin/compare/2.0.4.4-longer3D...GabrielBeraldo:bf2_6axis_dev)
> 
> @tpruvot is this the right comparaison method. 52 files...

I was not used to work with git, so the commit logic might be messed, but i suggest you to look commit by commit and find the relevant modifications for you. 
Anyway, i'll be available to help you guys out!

DerAndere1, 2020-03-01:
@GabrielBeraldo Thanks a lot for sharing your fixes. The changes look good! I don't even have a logic analyzer, so thanks for bringing in your engineering expertise.

hobiseven, 2020-03-02:
@GabrielBeraldo hi. I have one question regarding axis acceleration management . Have you replicated xyz acceleration to ijk? Or xyz drives the acceleration for the secondary axis?

GabrielBeraldo, 2020-03-02:
> @GabrielBeraldo hi. I have one question regarding axis acceleration management . Have you replicated xyz acceleration to ijk? Or xyz drives the acceleration for the secondary axis?

All the non extruder axes (XYZ and IJK) has the same privileges and features, like independent Min and Max positions, Min or Max endstops, Feedrates, Accelerations, Jerks and dynamic feedrate calculation when moving alongside with other non extruder axis.

All the parameters can be found in Configuration.h and Configurations_adv.h, in your case:

>#define DEFAULT_MAX_ACCELERATION      { 500, 500, 500, 9, 9, 500 } //X, Y, Z, I, J, K, E

NOTE: this example are only using I and J as additional axes (9 and 9) ,following the extruder acceleration (500), if you wanna use K axis or only I axis, you should add or remove one more parameter before the extruder value. 
~~I forgot some traces of my personal development, like the parametric definition of these parameters defined as MODULE_50 and MODULE_20, related to my syringe pump modules of 50mL and 20mL.~~

hobiseven, 2020-03-02:
Ok clear... I was thinking marlin was computing something sort of integral acceleration of the head but likely not! Great job anyhow @GabrielBeraldo

DerAndere1, 2020-03-02:
@hobiseven: This branch adds 6 axis support for the basic kinematics models (cartesian, maybe CORE_XY) where optional steppers I,J,K do not influence positioning. For those setups it manages positioning of the toolhead including look-ahead planning of accelleration (junction deviation or jerk). 
I have no experience but I think that if more than 3 axes are affecting tool-head positioning, you have to take account of the kinematics. I suggest to keep the code for new kinematics out of Marlin and instead develop a post processor that takes the output of your CAM software and caluculates the corrected G-code by applying the kinematics for your machine.

DerAndere1, 2020-03-04:
I have a new branch that was rebased on current upstream MarlinFirmware/Marlin: During the rebase there were so manny conflicts that it was almost a complete rewrite of the 6 axis code. Please test and give feedback: https://github.com/DerAndere1/Marlin/tree/bf2_6axis_dev5

hobiseven, 2020-03-04:
Ouaaaa already a version 2? Need to give it a try...

DerAndere1, 2020-03-04:
Dont have time to test compile. I expect some typos etc., but we will get there now that Gabriels branch can be used as a reference.

DerAndere1, 2020-03-04:
especially have a look at the parts that have a comment // TODO Test ... or // TODO Add support for ... 
Im AFK, clubbing for the rest of the night.

hobiseven, 2020-03-05:
Have fun! I introduced 3/4 of the code in the working tag 2.0.4.2 for my board . The rebase you did has some small issues. 2/3 typos so far. Will do the rest tonight and try to compile. Overall that is quite a lot of changes!

DerAndere1, 2020-03-06:
I fixed some compilation errors in my bf2_6axis_dev5 branch.  I even have problems compiling bugfix-2.0.x right now. Also there is still a problem with definition of  `MSG_DWELL` and similar MSG_... strings.

hobiseven, 2020-03-06:
Well, this is exactly why I use the code with tag 2.0.4.4, as I know this one compiles for my machine, with the custom touch screen and FSMC STM32 LCD. 
In the rebase you did, there is a mix of changes due to the 6 axis, as well as ongoing changes they do in the live branch. 

I am currently stuck there : 

Compiling .pio\build\alfawise_U30\src\src\gcode\geometry\G53-G59.cpp.o
In file included from Marlin\src\gcode\../inc/../core/boards.h:24:0,
                 from Marlin\src\gcode\../inc/MarlinConfigPre.h:35,
                 from Marlin\src\gcode\../inc/MarlinConfig.h:28,
                 from Marlin\src\gcode\gcode.h:286,
                 from Marlin\src\gcode\gcode.cpp:28:
Marlin\src\gcode\../inc/../core/macros.h:249:25: error: 'LIST_NUM_AXIS' was not declared in this scope
 #define _LIST_N(N,V...) LIST_##N(V)
                         ^
Marlin\src\gcode\../inc/../core/macros.h:251:27: note: in expansion of macro '_LIST_N'
 #define ARRAY_N(N,V...) { _LIST_N(N,V) }

                   ^~~~~~~
Marlin\src\gcode\../inc/../../Configuration_adv.h:732:29: note: in expansion of macro 'ARRAY_N'
 #define AXIS_RELATIVE_MODES ARRAY_N(NUM_AXIS, false, false, false, false, false, false, false)
                             ^~~~~~~
Marlin\src\gcode\gcode.cpp:61:40: note: in expansion of macro 'AXIS_RELATIVE_MODES'
 static constexpr xyze_bool_t ar_init = AXIS_RELATIVE_MODES;
                                        ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\../inc/../core/macros.h:249:25: note: suggested alternative: 'LOOP_NUM_AXIS'
 #define _LIST_N(N,V...) LIST_##N(V)
                         ^
Marlin\src\gcode\../inc/../core/macros.h:251:27: note: in expansion of macro '_LIST_N'
 #define ARRAY_N(N,V...) { _LIST_N(N,V) }
                           ^~~~~~~
Marlin\src\gcode\../inc/../../Configuration_adv.h:732:29: note: in expansion of macro 'ARRAY_N'
 #define AXIS_RELATIVE_MODES ARRAY_N(NUM_AXIS, false, false, false, false, false, false, false)
                             ^~~~~~~
Marlin\src\gcode\gcode.cpp:61:40: note: in expansion of macro 'AXIS_RELATIVE_MODES'
 static constexpr xyze_bool_t ar_init = AXIS_RELATIVE_MODES;
                                        ^~~~~~~~~~~~~~~~~~~
*** [.pio\build\alfawise_U30\src\src\gcode\gcode.cpp.o] Error 1

GabrielBeraldo, 2020-03-06:
> I fixed some compilation errors in my bf2_6axis_dev5 branch. I even have problems compiling bugfix-2.0.x right now. Also there is still a problem with definition of `MSG_DWELL` and similar MSG_... strings.

is these messages properly defined on language files?

hobiseven, 2020-03-06:
I have posted a new git, with the 3 axis compile OK. KO for more axis, as we still have some mistakes / non declarations...

https://github.com/hobiseven/Marlin2.0.4.4_6axis_Longer3D

hobiseven, 2020-03-06:
@DerAndere1  I see you are pushing quite some branches... now 6 and 7... Fixing a bit more code each time!?

hobiseven, 2020-03-06:
@DerAndere1  now entering in DOGM, and I think that one was never touched so far... 

Marlin\src\lcd\dogm\status_screen_DOGM.cpp:393:55:   required from here
Marlin\src\lcd\dogm\../../inc/../core/types.h:864:126: error: could not convert '{((float)((const XYZEval<float>*)this)->XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::x), ((float)((const XYZEval<float>*)this)->XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::y), ((float)((const XYZEval<float>*)this)->XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::z), ((float)((const XYZEval<float>*)this)->XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::i), ((float)((const XYZEval<float>*)this)->XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::e)}' from '<brace-enclosed initializer list>' to 'XYZval<float>'
C ompiling .pio\build\alfawise_U30\src\src\lcd\dogm\u8g_fontutf8.cpp.o
  FI XYZval<float>   asFloat()                   const { return {   float(x),   float(y),   float(z),   float(i),   float(e) }; }
Compiling .pio\build\alfawise_U30\src\src\lcd\dogm\ultralcd_DOGM.cpp.o

hobiseven, 2020-03-06:
@DerAndere1  I sent you an invite as collaborator in my git so you can also touch the code...

DerAndere1, 2020-03-07:
OK. I saw some minor issues, I will have a closer look at it. Arduino IDE was not reliable anymore for compiling. So I switched from Eclipse + Arduino IDE to VS code + PlatformIO. My next branch bf2_6axis_dev7 now also compiles for NON_E_AXES 3 and most other errors are fixed too. Now I came to the point where the `template<typename T>
struct XYZval` in types.h is the causing errors for NON_E_AXES 4 (cannot convert to float). I expected issues with types.h because I absolutely do not know how it works and what it does. My changes there were pure guesswork.

hobiseven, 2020-03-07:
Ok. I actually corrected typos in type.h....I am now stuck with an error I do not yet understand . Let s see tomorrow! I also use platformio / vscode

hobiseven, 2020-03-07:
I have the test setup ready, steppers working in the 3 axis more. I am still stuck in the 4 axis mode with an error in the configutation_storage.cpp.. Related to macros, and so far did not figure out :
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::report(bool)':
Marlin\src\module\../inc/../core/serial.h:195:59: error: no matching function for call to 'serial_echopair_PGM(float, float)'
 #define _SELP_2_P(PRE,V)     do{ serial_echopair_PGM(PRE,V); SERIAL_EOL(); }while(0)
                                                           ^
Marlin\src\module\../inc/../core/serial.h:197:49: note: in expansion of macro '_SELP_2_P'
 #define _SELP_4_P(a,b,V...)  do{ _SEP_2_P(a,b); _SELP_2_P(V); }while(0)

hobiseven, 2020-03-07:
@DerAndere1  Code is compiling with 4 axis, but does not moove. To get to compile, I located the error in configuration_store.cpp, , line 3059. Commenting it out gets the compiler to go through, with no warnings / errors . ... I still try to understand what is wrong there... 
Going to 5 axis generates more errors , linked to types.h, as you suspected. So far, X axis mooves, but not I... Might be for other reasons though.

     #if HAS_CLASSIC_JERK
        , SP_X_STR, LINEAR_UNIT(planner.max_jerk.x)
        , SP_Y_STR, LINEAR_UNIT(planner.max_jerk.y)
        , SP_Z_STR, LINEAR_UNIT(planner.max_jerk.z)
          #if NON_E_AXES > 3
           // , SP_I_STR, LINEAR_UNIT(planner.max_jerk.i)
            #if NON_E_AXES > 4
              , SP_J_STR, LINEAR_UNIT(planner.max_jerk.j)
              #if NON_E_AXES > 5
              , SP_K_STR, LINEAR_UNIT(planner.max_jerk.k)
              #endif
            #endif
          #endif

hobiseven, 2020-03-08:
@DerAndere1  @GabrielBeraldo All menus look fine with proper preload values on I j and k. EPROM has a problem. Moving x fine but I moves about 1/4 of X and j moves 1/4 of I....
Need to focus now on the TODO that are not done....

DerAndere1, 2020-03-08:
@hobiseven: Great progress. You may want to make the dev branch your new master and work on that branch, see https://github.com/hobiseven/Marlin2.0.4.4_6axis_Longer3D/issues/#1. In the dev branch you find a fix for the K stepper. If you add those changes into other places, please first find out how to amend the correct original author to commits (https://stackoverflow.com/questions/3042437/how-to-change-the-commit-author-for-one-specific-commit) or use git to rebase or cherry-pick commits. Otherwise the copyright status is not clear.

hobiseven, 2020-03-08:
Ah ok i will look into this as I am a rookie in github ( same for sw coding...) !

GabrielBeraldo, 2020-03-08:
> @DerAndere1 @GabrielBeraldo All menus look fine with proper preload values on I j and k. EPROM has a problem. Moving x fine but I moves about 1/4 of X and j moves 1/4 of I....
> Need to focus now on the TODO that are not done....

In the case of EEPROM, check if you are iterating through all the additional axes (take a look: https://github.com/MarlinFirmware/Marlin/commit/0fd2a0db4d9d98a6be2b00705e191a8ac56b317c)

hobiseven, 2020-03-08:
@DerAndere1 
I made a new clone git from tpruvot , created a 6axis Branch and I will put the initial commit as coming from you. I wanted to keep the base as tpruvot as there are quite some changes done for alfawise touchscreen and lcd
Hope this will suits you better.

DerAndere1, 2020-03-08:
> 
> 
> Ah ok i will look into this as I am a rookie in github ( same for sw coding...) !

You made great contributions and you were kind enough to share your results. Dont worry, my comments were meant as tips, not as critique. I often struggle with git myself.

[7 more replies]

closed 2020-03-12 by DerAndere1