DerAndere1/Marlin2ForPipetBot #5: I,J,k not shown on the 12864 lcd

nguyennhattam268, 2019-12-27:
I have configured marlin-bf2_6axis_dev
#define NON_E_AXES 6
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
on Ramps1.4 and ok but when I open it, I don't see the axes I, J, k on lcd please help me to change more where

DerAndere1, 2020-01-07:
Hi, thanks for your interest in this project. The branch bf2_6axis_dev is still early WIP and I never managed to get the additional axes I,J,K working. See issue https://github.com/DerAndere1/Marlin/issues/1 . Keep in mind that this was a toy project for me to learn programming, so I did not research if 8 bit AVRs are to slow to process more than 4 Axes. Maybe the whole Idea to add additional axes to Marlin is not worth the effort. If you need more than 3 axes for a serious project , currently the best is to base your project on Smoothieware + their supported hardware.  Smoothieware has support for 6 axes. 
If you still want to contribute to marlin-bf2_6axis_dev, please start with a minimal setup consisting of only the required steppers and min-endstops. Disable all Displays and first fix issue https://github.com/DerAndere1/Marlin/issues/1 . For debugging, use e.g. Pronterface to manually send G-code. Only if you get G28 and G1 working it is worth to add support for graphical displays. The code to make LCD display working with more axes is quite scattered I guess. Never had a loo at it, but at least https://github.com/DerAndere1/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/menu/menu_motion.cpp would need changes

nguyennhattam268, 2020-01-08:
Hello, thank you  I have seen the axes I, J, K On the LCD
I still want to use marlin-bf2_6axis_dev and the next issue I can't control the I, J, and K axes on the LCD
I want to know how to control marlin X, Y, Z axes 
I have looked through the files motion.cpp, planner.cpp, stepper.cpp .. how can I find out how to find it?
Do you have a suggestion for me?
Thank you

DerAndere1, 2020-01-08:
> Hello, thank you I have seen the axes I, J, K On the LCD

great!

> I still want to use marlin-bf2_6axis_dev and the next issue I can't control the I, J, and K axes on the LCD

This is because of issue #1 . You will have to fix that first. Disable the LCD in the config files and reflash. Debug and fix until you can move XYZ (and IJK) from the host software via USB. Only if that works, enable a display and again debug + fix until it works. 

> I want to know how to control marlin X, Y, Z axes

you can control X, Y, Z axes by sending G-code, e.g. [G1](http://marlinfw.org/docs/gcode/G000-G001.html) via a host software such as printrun/pronterface. If it does not work, use original bugfix-Marlin2.0.x and introduce my changes one by one and repeatedly retesting until you find the offending code. You can find what I have chaged by analyzing all commits made by me. For each commit you can view the changes using a diff tool. If you want to do that using github's integrated interfaces: 
1. go to https://github.com/DerAndere1/Marlin/
2. click on "commits".
3. Select the branch "bf2_6axis_dev" to open [its commit history](https://github.com/DerAndere1/Marlin/commits/bf2_6axis_dev).
4. Search for all commits made by DerAndere1 or DerAndere and click on the commit ID. The oldest one (and only one) I found is https://github.com/DerAndere1/Marlin/commit/0258cc906eda008cd1863be44b05ab6c3a15fdc9  . 

> I have looked through the files motion.cpp, planner.cpp, stepper.cpp .. 

Those plus their corresponding header (*.h) files are the most important ones. Almost all relevant code that affects movement can be found if you check where I made changes. See https://github.com/DerAndere1/Marlin/commit/0258cc906eda008cd1863be44b05ab6c3a15fdc9 as described above.

nguyennhattam268, 2020-01-12:
Thank you I will do it right away

nguyennhattam268, 2020-02-10:
Thank you, I control the I, J, and K axes but I am having problems running very slowly

closed 2020-03-16 by DerAndere1