DerAndere1/Marlin2ForPipetBot #6: [FR] Merge six axis support into current upstream MarlinFirmware/Marlin bugfix-2.0.x

DerAndere1, 2020-03-13:
### Description

With my 6 axis code and Gabriels fixes, there is a proof of principle that support for 6 axes + extruders  can be added to Marlin. It would be nice to rebase Gabriels branch onto current upstream www.github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x . My first attemts (my branch bf2_6_axis_dev8) did not introduce regressions in G28 and G1 for XYZE but there are issues with the additional axes I,J and K.

hobiseven, 2020-03-15:
Hi. Looks like I accepted the pr and got your code in my repo. However it also appears that I merged code in your repo and that I do not know really why. Was it intended? The code compiles with non_e_axes=3 but not with 5 .. there is obviously somewhere some additional parameters not passed on. With corona I have a bit more time to look at that. Greetings from paris

DerAndere1, 2020-03-15:
It was intended, I chose the option to give you the rights to modify the pull request and with that you I accepted that you modify my remote PR branch. My local branch does not get affected by changes in the remote branch as long as I do not pull from origin. I now had several options (none of those will affect your september_base branch):  
a) `git push --force` my local branch to the PR branch to overwrite you changes in my remote PR branch
b) `git pull --rebase` your changes from the remote PR branch into my local branch
c) `push origin/bf2_6axis` pushes my local (original) branch to a new remote branch called bf2_6axis
d) delete the PR branch as it is merged and I have no use for it. 

I chose c) and then d). I also fixed compilation with Junction deviation enabled and NON_AXES 4 in your 6axis_september_base branch. You can get the changes like so:
1. checkout your 6axis_september_base branch 
2. `git fetch` 
3. `git pull --rebase`

hobiseven, 2020-03-15:
Ok, so nothing I should not have done! Good. 

I did get your changes regarding the junction deviation.

Now, something I face is the code does not compile for anything above 3 axis, and I get the first errors in G26, which as far as I know was not touched in the other builds. There is an inconsistency in the destination vaciable declaration, and I did not found yet where. I looked in motion.cpp, where the destination varaible is declared, but so far did not found the discrepancy.

hobiseven, 2020-03-15:
Well, for some reason I cannot see the basic include : 
{
	"resource": "/c:/Users/frq03719/Documents/GitHub/Marlin/Marlin/src/gcode/bedlevel/G26.cpp",
	"owner": "C/C++",
	"code": "1696",
	"severity": 8,
	"message": "cannot open source file \"stdint.h\" (dependency of \"../../inc/MarlinConfig.h\")",
	"source": "C/C++",
	"startLineNumber": 27,
	"startColumn": 1,
	"endLineNumber": 27,
	"endColumn": 36
}

hobiseven, 2020-03-16:
I try to understand that specific error message... 
{
	"resource": "/c:/Users/frq03719/Documents/GitHub/Marlin/Marlin/Configuration.h",
	"owner": "C/C++",
	"code": "1696",
	"severity": 8,
	"message": "#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\\Users\\frq03719\\Documents\\GitHub\\Marlin\\Marlin\\src\\gcode\\bedlevel\\G26.cpp).",
	"source": "C/C++",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 1
}

DerAndere1, 2020-03-16:
Just wanted to tell that I am finally able to test 6 axis code updated to bugfix-2.0.x from some days ago on my machine. Same issue as you had with the dev_2_rebased branch, @hobiseven : slower speed for axis I in a NON_E_AXES 4 config (XYZI+E) but at least no wrong direction. G28 auto-home works, too. Here is the branch: https://github.com/DerAndere1/Marlin/tree/bf2_6axis_dev13
Regarding your issue: it seems that your intellisense is confused: https://github.com/microsoft/vscode-cpptools/issues/1863

hobiseven, 2020-03-16:
Great! 
About the wrong direction, it is probably one of my steppers. I hacked them toget more power out of them, and likely made one mistake. So do not pay attention to that, until I do recheck my wiring, and put 2 non hacked steppers, on on X and one on I, then redo for J in 5 axis. 

For the compile issue, that is what I assumed, as the compile was going on fine.

So : dev13 branch is the one with the problematic type changes. 

And if I am correct, you have one with the non problematic changes, ie stopping the upstream end September. I am lost on that one. Which one is the one that compiles fine in 4 axis please? I still cannot compile G26...

DerAndere1, 2020-03-16:
@hobiseven: I never touched arc support, and I have no Idea how arc support is supposed to work with 6 axes. so you will have to deal with that yourself. either disable arc support in the configs or add 6 axis support to G26.cpp. My status is:  https://github.com/DerAndere1/Marlin/tree/bf2_6axis_dev13 is up-to date including the position-specific types but suffers from the feedrate issue for I (and possibly also JK). https://github.com/DerAndere1/Marlin/tree/bf2_6axis_dev12 is Garbriels branch slightly updated (no position-specific types) and was merged into your 6axis_september_base branch.

hobiseven, 2020-03-16:
@DerAndere1 Hmmm, ok. There might be a typo in my config.h, as simple as that. 
I do not need arc support, and will comment out, that should solve the issues. I hope. 
Sorry to ask so many basic questions. As I said, I am not a pro SW guy. I am just a pro HW guy...

DerAndere1, 2020-03-16:
Oh well, I am the one who introduced most typos so far I guess :)  I just fixed some in your 6axis_september_base a minute ago. You may want to `git pull --rebase` that commit into your local branch first. As long as you give feedback on your issues and how you expect it to work I might be able to help. Thanks a LOT for doing the tedious testing. It was is a huge help in the whole process.

hobiseven, 2020-03-16:
I checked and indeed, I made the assumptions that in September branch, and in my 2.0.4.4 branch the config.h and config_adv.h were the same, but NO!!! that is why I had troubles. Gosh... I will check later tonight! It will work I am sure. 
BTW, if you need anything to test on NXP CPU, give me a buzz at Olivier.briand at NXP.com ....

DerAndere1, 2020-03-16:
Unfortunatelly the Configs from different Marlin versions often are not compatible. The layout has to match.  The best is to use a diff tool to do the file compare. If you temporarily add a renamed config from the 2.0.4.4 branch into the current workspace you can use the IDEs file compare function (in VS code+ platformIO, hold Ctrl while selecting the default config and the custom config file and right-click -> "Compare selected". Then you can pick the parameter changes one by one into the new default config, keeping the layout of the default config.

DerAndere1, 2020-03-16:
@hobiseven  : I hope you don't mind the cleanup in this issue. please open new specific issue report if required. If it is about your fork, enable issue reports in your github repo (see https://help.github.com/en/github/managing-your-work-on-github/disabling-issues), open a new issue there and ping me by adding @DerAndere1 to the issue desription over at your fork . Thanks.

DerAndere1, 2020-03-16:
@hobiseven: Example configs that should work with your bf2_september branch may be found here: https://github.com/tpruvot/Marlin/tree/ee7558a6228747a7502ab50e2817234fb8a6feb4

DerAndere1, 2021-07-21:
6 axis support is now merged into upstream MarlinFirmware/Marlin

closed 2021-07-21 by DerAndere1