DerAndere1/Marlin2ForPipetBot #7: [bf2_6axis_dev13] [BUG] Additional axes IJK move too slow.

DerAndere1, 2020-03-16:
### Description

With the branch https:/github.com/DerAndere1/Marlin/tree/bf2_6axis_dev13 , axes IJK move slower than expected. When the DEFAULT_MAX_FEEDRATE and STEPS_PER_UNIT for the axes IJK in Configuration.h is set to the same value as for the Z axis, IJK move slower than Z.

### Steps to Reproduce

Set DEFAULT_MAX_FEEDRATE and STEPS_PER_UNIT if the I axis to the same value as for the Z axis

**Expected behavior:** Same speed for I and Z

**Actual behavior:** I moves slower than Z

#### Additional Information

Regression (bug is not present in branch https:/github.com/DerAndere1/Marlin/tree/bf2_6axis_dev12 which is based on pre-release Marlin-bugfix2.0.x)

hobiseven, 2020-03-29:
This is no longer an issue for the Gcodes. 
Still and issue for move menus. See other issue opened.

DerAndere1, 2020-04-07:
This issue is about the updated branch bf2_6axis_dev13 as  noted in the description and I think it is still not fixed there.

DerAndere1, 2020-08-18:
The last commit on bf2_6axis_dev14 contains a work-around. A proper fix of the underlying issue still should replace the work-around in the future. Pull requests are welcome.

DerAndere1, 2020-09-18:
Finally fixed thanks to bilsefs PR https://github.com/DerAndere1/Marlin/pull/31

closed 2020-09-18 by DerAndere1