DerAndere1/Marlin2ForPipetBot #8: [FR] Merge six axis support into  MarlinFirmware/Marlin bugfix-2.0.x : Move axis through menu gives different speeds depending on the axis

hobiseven, 2020-03-16:
Current implementation of 4 non E axis extension with X Y Z I gives proper response to G1 X10 Y10 Z10 I10, with all axis mooving at the same speed, and with the same angular displacements. 

However, displacing each individual axis using Marlin LCD menu gives different speeds and angles depending on the axis.

DerAndere1, 2020-03-18:
What display config?

hobiseven, 2020-03-18:
See below, in the independent move axis menu 
Motion > Move Axis > Move I
For example. 
I is way slower than Y. 
J is actually faster the I, but displacement is super small. H
[image]

DerAndere1, 2020-03-19:
I have no clue about the display-related code. Never had a serious look at it (I don't use a dedicated display but connect to a laptop). Maybe someone else knows how to fix it?

hobiseven, 2020-03-29:
@DerAndere1  ; After fixing the IJK steps calculation, this is still broken. I will fix this now.

hobiseven, 2020-03-29:
Ok, found the issue, at least something obviously wrong : Acceleration is correct for X, but incorrect for J.  Note, this is also possibly impacting all movements. I did set all steppers with same parameters.

To reproduce, simply request an axis moove through the moove menu, and request 10mm. We have 80 steps per mm > 800 steps requested overall. The steps requested is correctly calculated in stepper.cpp for X,I and J.
Result : 
X mooves by 1/4 turn, and my X stepper is 200 steps per turn > 50 steps done.
Y mooves by 1/2 turn, and my X stepper is 100 steps per turn > 50 steps done.
I mooves by 1/4 turn, and my I stepper is 200 steps per turn > 50 steps done but at very slow speed
J mooves by a very small amount very fast. Trying to go back to original position only works 1 time out of 2. the second time, axis direction is inverted. 

Issue can be seen in stepper.cpp, line 1877

// Compute the acceleration and deceleration points
      accelerate_until = current_block->accelerate_until << oversampling;
      decelerate_after = current_block->decelerate_after << oversampling;

For X  
Accelerate_until = 400
decelerate_after = 400
For I
Accelerate_until = 0
decelerate_after = 800
For J  
Accelerate_until = 0
decelerate_after = 800

hobiseven, 2020-03-30:
Update : For short Gcode files, all works perfectly. The attached file, much longer, still shows difference on J axis. I think this is the same root cause as the difference seen in the displacement issued by the menus. Gcode attached as a reference. Same Naca 43012 profile cut at the same time on both XY and IJ. 
Would be great is @DerAndere1 you could play this file in your AVR board, even on Marlin 119, to check wether the issue is a porting issue, or a missing code issue. 

[43012.zip](https://github.com/DerAndere1/Marlin/files/4402677/43012.zip)
Output on ARM  , M114 at the end : 

Send: M114
Recv: echo:M114
Recv: X:0.35 Y:0.00 Z:0.00 I:0.35 J:-1.42 E:0.00 Count X:28 Y:0 Z:0 I:14 J:-57
Recv: ok

The output should be exactly the same X /I and Y/J. 

Let me check the acceleration and deceleration calculation. There seems to be quite a lot of 6 axis code missing from motion.cpp... more to come

hobiseven, 2020-03-30:
This is fixed. Definition of manual feedrates ware missing for I, J and K axes. See commit https://github.com/hobiseven/Marlin/commit/efde6ca2a6abb3e05f3f909bc1d65dbbe0a2ece8

closed 2020-03-30 by hobiseven