DerAndere1/Marlin2ForPipetBot #10: 6 axis update for Marlin code : Zero extruder non supported

hobiseven, 2020-03-17:
Issue : Setting extruder number to zero results in compilation errors

Wrong parameter definition and use in the code utility.h :  

onstexpr char axis_codes[NUM_AXIS] = { 'X', 'Y', 'Z'
  #if NON_E_AXES > 3
    , 'I'
    #if NON_E_AXES > 4
      , 'J'
      #if NON_E_AXES > 5
        , 'K'
      #endif
    #endif
  #endif
  , 'E' };

In case of Extruder = 0 , the above code is wrong. 
There should be a different definition depending on the below variable from Configuration.h

// This defines the number of extruders
// :[1, 2, 3, 4, 5, 6]
#define EXTRUDERS 0

DerAndere1, 2020-03-19:
this seems to be a bug in upstream Marlin then. you can define EXTRUDERS 1 and define all E stepper pins as the same pin number.

DerAndere1, 2020-05-14:
I think this is not a bug, just a configuration error. The compiler error messages will show you what parts of the Configuration files are wrong. The reason for the compiler errors is typically that elements for the E-axis (extruder E0) are not deleted from the relevant arrays. To solve this issue, delete the last element from all relevant arrays in the configuration files.

closed 2020-05-14 by DerAndere1

DerAndere1, 2020-05-14:
I could compile with EXTRUDERS 0 after I changed in macros.h 

`#define NUM_AXIS (NON_E_AXES + 1)`
to 
```
#if EXTRUDERS == 0
  #define NUM_AXIS (NON_E_AXES)
#else
  #define NUM_AXIS (NON_E_AXES + 1)
#endif
```
Fixed in bf2_6axis_dev14

hobiseven, 2020-05-14:
Thank you!! 
I will cherry pick this to put it in my git.