DerAndere1/Marlin2ForPipetBot #9: [FR] Merge six axis support into MarlinFirmware/Marlin bugfix-2.0.x : EEprom version readback failed

hobiseven, 2020-03-17:
Even after initializing the Eeprom through the LCD UI, messages at boot are as follows :

start
echo:PowerUp
Marlin 2.0.x for Alfawise

echo: Last Updated: 2019-09-25 | Author: Hobi, tpruvot
echo:Compiled: Mar 16 2020
echo: Free Memory: 44983  PlannerBufferBytes: 2944
echo:SD card ok
echo:EEPROM version mismatch (EEPROM=? Marlin=V70)
echo:Hardcoded Default Settings Loaded
echo:SD card ok

DerAndere1, 2020-03-18:
After flashing Marlin, does it help to send M502 followed by M500 to the machine or do an EEPROM init?

hobiseven, 2020-03-19:
I did actually initialize the Eeprom through the menus already. That did not help. 
M502 followed by M500 gives the following : 

Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Send: N0 M110 N0*125
Recv: echo:PowerUp
Recv: Marlin 2.0.x for Hotwire Foam Cutter
Recv: 
Recv: echo: Last Updated: 2019-09-25 | Author: Hobi, tpruvot, DerAndere
Recv: echo:Compiled: Mar 17 2020
Recv: echo: Free Memory: 44983  PlannerBufferBytes: 3072
Recv: echo:SD init fail
Recv: echo:EEPROM version mismatch (EEPROM=? Marlin=V70)
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:SD init fail
Recv: echo:SD init fail
Recv: echo:SD init fail
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: M155 S2
Recv: ok
[...]
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: Error:Field probe_offset[Z_AXIS] mismatch.
Changing monitoring state from "Operational" to "Error: Field probe_offset[Z_AXIS] mismatch."
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Send: N3 M140 S0*102
Changing monitoring state from "Error: Field probe_offset[Z_AXIS] mismatch." to "Offline (Error: Field probe_offset[Z_AXIS] mismatch.)"
Connection closed, closing down monitor

Erinell, 2020-03-23:
I have the same problem with this :

echo:EEPROM version mismatch (EEPROM=ERR Marlin=V76)
echo:Hardcoded Default Settings Loaded
Error:Field planner_leveling_active mismatch.
echo:EEPROM Initialized

hobiseven, 2020-03-23:
Working on it... but macros are hard to follow

hobiseven, 2020-03-24:
@DerAndere1 
I could not get the latest version of the code, even reverting to "normal" 3 axes,running M500 and that is not normal, as all code modification you made are based on #ifdefs, and unless there are typos in those #ifdef , going back to 3 non E axes must work.

I therefore checked all EEprom related code where there was #ifdef changes, or where NON_E_AXES is appearing, and there was 2 typos to correct. At least I saw those. But still no go, and the EEprom write code is still failing in M500. It fails at Zprobe offset, line 626 

In the latest code it is : 
_FIELD_TEST(probe_offset[Z_AXIS]);
      EEPROM_WRITE(probe_offset);

> I checked the initial September branch, recompiled it, and M500 works > I therefore checked the code, and found that the M851 commit,(  Add M851 X Y probe offsets (#15202) ) changed the code at that location > 

I tried to revert it, manage to get to compile in 3 axis, and M500 works now without an error. 
I think I incompletely revert the code. 

In 5 axis, code hangs, but that I can understand, and I have to fix.

Can you please check my latest rever commit, and check if I did all properly please? If necessary, create a new branch, and make needed changes.? I am struggling with Github again..

hobiseven, 2020-03-26:
@DerAndere1 @Erinell 

The Eeprom storage and readback works now. It was indeed a mix of small code missing parts + M851 breaking some functions. We had to revert M851 to get the code to work. Check my branch 6_Axis_September_base. 

I will need to make further code cleanup as I still get some warnings, but all works so far, storing / retreiving I,J axes settings, as well as G1/G90/G91...

closed 2020-03-26 by hobiseven

DerAndere1, 2020-03-27:
@Erinell we continue to fix issues at https://github.com/hobiseven/Marlin/tree/6_Axis_September_base first, then I will pull the fixes into my bf2_6axis_dev12 branch etc.