DerAndere1/Marlin2ForPipetBot #11: 6axis Marlin code : M114 / G1 bug

hobiseven, 2020-03-19:
See below commands :

Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:-59.32 E:0.00 Count X:0 Y:0 Z:0 I:-12 J:-2379
Recv: ok
[...]
Send: G1 J 59.32
Recv: ok
[...]
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:59.32 E:0.00 Count X:0 Y:0 Z:0 I:-12 J:2367
Recv: ok

Expected answer :
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:0.00 E:0.00 Count X:0 Y:0 Z:0 I:-12 J:0

DerAndere1, 2020-03-19:
This is not a bug but expected behaviour. Marlin is in absolute mode by default. Send G91 to switch to relative mode and G90 to switch back to absolute

hobiseven, 2020-03-19:
@DerAndere1 hmmm, in that case, shouldn't  we get : 
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:-59.32 E:0.00 Count X:0 Y:0 Z:0 I:-12 J:-2379
Recv: ok
[...]
Send: G1 J 59.32
Recv: ok
[...]
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:0.00 E:0.00 Count X:0 Y:0 Z:0 I:-12 J:0
Recv: ok

BUT, we get : 
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:59.32 E:0.00 Count X:0 Y:0 Z:0 I:-12 J:2367... 

indtead of J 0, we get 59.32, ie twice what we should have as a displacement in absolute mode. 

I did try G91/G90, and the behavior is exactly the same.

DerAndere1, 2020-03-21:
If M114 initially reports X:0.00 Y:0.00 Z:0.00 I:0.00 J:-59.32 E:0.00 and you send G91 followed by G1 J59.32 followed by M114 you should indeed get Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:0.00 E:0.00

hobiseven, 2020-03-21:
I will debug this as I have a stlink with hardware debug capabilities. Let s see if I can find the root cause. Something like this can be the killing bug to run gcode properly. I have one 30 lines gcode using xy and the same code using ij and the results are totally different ...

hobiseven, 2020-03-22:
@DerAndere1 
Well, G91 is fixed now. Actually one small piece of code was needed for IJK, . IJK actually were never going in relative mode. This is fixing the issue, and I changed a few things for IJK in G1 too. 
Now, a Gcode mooving XY and IJ the same way gives almost the same result. There is somewhere a rounding error in I, J step calculations. 

If we start my simple gcode with that position : 
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:0.00 E:0.00 Count X:0 Y:0 Z:0 I:0 J:0
Recv: ok

ending the Gcode, returning to 0,0,0,0 gives the following : 
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:0.00 E:0.00 Count X:0 Y:0 Z:0 I:-35 J:-35
Recv: ok

We have made 35 steps too much on I and J. 

LCD DOGM working fine also. So I have now to fix
1 - minor steppers error secondary axes
2 - EEprom
3- Feedrate for foam curret, but I already made the changes, just need to test, but I want the base to be running fine before changing feedrates. 
[image]

closed 2020-03-22 by hobiseven

hobiseven, 2020-03-22:
Here is my test Gcode : 
[Jedi1XYIJ.zip](https://github.com/DerAndere1/Marlin/files/4365822/Jedi1XYIJ.zip)

DerAndere1, 2020-03-22:
Looks great! Regarding the -35 steps: Start from the homed state (all 0) and see if there is any pattern to the shown step counts. Does this happen for XYZ if you move only XYZ back and forth? is it a wrong fix offset? or is the error bigger after large IJK moves?