DerAndere1/Marlin2ForPipetBot #12: miss-calculated mooved / steps for I and J

hobiseven, 2020-03-26:
A small mistake in steps calculation on secondary axes : 

How to reproduce : a single G1 command like

Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:0.00 J:0.00 E:0.00 Count X:0 Y:0 Z:0 I:0 J:0
Recv: ok
[...]
Send: G1 I 50
Recv: ok
[...]
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 I:50.00 J:0.00 E:0.00 Count X:0 Y:0 Z:0 I:2100 J:1
Recv: ok

You can already see the issue.

Why J mooved by 1 step?

Calculation for I steps is correct, as we have the following parameters 👍  Recv: echo:Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z80.00 I42.00 J39.00 E80.00

42*50 = 2100

Hunting tonight this last bug for regular use of 4 axes machine!

Olivier.

hobiseven, 2020-03-26:
I checked G1 code, and then planner.cpp, and the block which is queued is correct after a G1 command. 
The error generating those extra steps looks to be in the interrupt driven code, which is a bit harder to debug. Let me see how we can debug this with the STLINK

DerAndere1, 2020-03-27:
First double-check that M114 correctly reports the actual steps. Is the step count for J increased exactly by one for each `G1 Iiii` ? Does the length of the I-axis move (parameter `Iiii`  of the G code) influence the steps added to J?

hobiseven, 2020-03-29:
II still try to nail this issue.. I navigate between checking the reporting, and also in stepper.cpp.planner.cpp.
A single G1 X0.1 is enough to get a strange 1 additional step reporte on I and J. 
500 G1 X1 result in I1 J1 steps
50 G1 X1; G1 X-1 still gives I 1 step and J 1 step...

I try to nail down where I and J are touched by the code while going a single G1 X0.1.... That is 8 steps in X direction. Should of course leave I and J untouched.

hobiseven, 2020-03-29:
@DerAndere1  it is fixed! 
There was an missing code to initialize 
 // Initialize Bresenham delta errors to 1/2
for I, J and K. 
Now, I run a typical foam cut Gcode, and results are the same for XY and IJ. I did cut a rectangular wing, with same geometries on XY and IJ. Steps are reporte correctly. 
See my latest commit :  https://github.com/hobiseven/Marlin/commit/61425665330adf7d05aabde523088e86783751c2

closed 2020-03-29 by hobiseven

DerAndere1, 2020-03-29:
@hobiseven : Good debuging job! Thanks a lot for sharing!