DerAndere1/Marlin2ForPipetBot #13: [FR] [bf2_6axis_dev12] 6 axes broken with certain configs and G-codes. Add Sanity checks!

DerAndere1, 2020-03-29:
### Description

Marlin support for 6 axes (XYZIJK + E) works only for certain configs and only a subset of G-codes is supported at the moment. I suggest to collect a whitelist of working features 

**Expected behavior:** [What you expect to happen]
add sanity checks against any non-working config settings. 

**Actual behavior:** [What actually happens]
some config combination may not compile with NON_E_AXES > 3, others may compile but do not work in practive. 

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2020-03-30:
Currently the configs by hobiseven are the only ones that were tested. Below I list Individual features that may work (although not all combinations). Alll other config settings currently should be treated as not supported yet. 

- A4988 drivers
- NON_E_AXES 3
- NON_E_AXES 4
- NON_E_AXES 5
- EXTRUDERS 1
- [DEBUG_LEVELING;], [DEBUG_PINS;]
- [dogm display (possibly not all menus work flawlessly with axes I,J,K)]
- One stepper driver per axis

The following G-codes were tested:

- G1 X10; G1 Y10; G1 Z10; G1 I10; G1 J10; G1 X10 Y10 Z10; 
- G1 X10 Y10 Z10 I10; G1 X10 Y10 Z10 J10;G1 X10 Y10 Z10 I10 J10; G1 I10 J10; 
- G28; G28 X; G28 Z; G28 I; G28 J; 
- G90; G91
- M43 E1
- M114; M114 D (?)
- M206 (?)
- M500; M502

hobiseven, 2020-03-31:
Added to the white list : 
M17 M18 M85
M501

closed 2021-07-21 by DerAndere1