DerAndere1/Marlin2ForPipetBot #14: DIOE_STOP_PIN_DDR and 'E_STOP_PIN' was not declared in this scope

igorlab, 2020-04-06:
### Description

when try to compile code for atmega2560 gets error 
### Steps to Reproduce

1. env_default = megaatmega2560

**Expected behavior:**
compile without error
**Actual behavior:**
errors:
DIOE_STOP_PIN_DDR and 'E_STOP_PIN' was not declared in this scope

`Processing megaatmega2560 (platform: atmelavr; framework: arduino; board: megaatmega2560)
-----------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES:
 - framework-arduino-avr 5.0.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.1.3
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <c1921b4>
|   |-- <SPI> 1.0
|-- <SailfishLCD> #6f53c19
|-- <SailfishRGB_LED> #4d0c888
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster> #3a18be5
|-- <Wire> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pioenvs\megaatmega2560\src\src\module\endstops.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\module\motion.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\sd\SdFatUtil.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\sd\SdFile.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\sd\SdVolume.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\sd\cardreader.cpp.o
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37:0,
Compiling .pioenvs\megaatmega2560\src\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp.o
                 from Marlin\src\module\../inc/MarlinConfig.h:33,
                 from Marlin\src\module\endstops.h:33,
Compiling .pioenvs\megaatmega2560\src\src\sd\usb_flashdrive\lib\Usb.cpp.o
                 from Marlin\src\module\endstops.cpp:32:
Marlin\src\module\endstops.cpp: In static member function 'static void Endstops::init()':
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:86:35: error: 'DIOE_STOP_PIN_DDR' was not declared in this scope
 #define _SET_INPUT(IO)        CBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                   ^
Marlin\src\module\../inc/../core/macros.h:80:19: note: in definition of macro 'CBI'
 #define CBI(n,b) (n &= ~_BV(b))
                   ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:98:35: note: in expansion of macro '_SET_INPUT'
 #define SET_INPUT_PULLUP(IO)  do{ _SET_INPUT(IO); _WRITE(IO, HIGH); }while(0)
                                   ^
Compiling .pioenvs\megaatmega2560\src\src\sd\usb_flashdrive\lib\masstorage.cpp.o
Marlin\src\module\endstops.cpp:248:7: note: in expansion of macro 'SET_INPUT_PULLUP'
       SET_INPUT_PULLUP(E_MIN_PIN);
       ^
In file included from c:\users\igor\.platformio\packages\toolchain-atmelavr\avr\include\avr\io.h:99:0,
                 from c:\users\igor\.platformio\packages\toolchain-atmelavr\avr\include\avr\pgmspace.h:90,
                 from C:\Users\Igor\.platformio\packages\framework-arduino-avr\cores\arduino/Arduino.h:28,
                 from Marlin\src\module\../inc/../HAL/HAL_AVR/HAL.h:27,
                 from Marlin\src\module\../inc/MarlinConfig.h:35,
                 from Marlin\src\module\endstops.h:33,
                 from Marlin\src\module\endstops.cpp:32:
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:86:54: error: 'DIOE_STOP_PIN_PIN' was not declared in this scope
 #define _SET_INPUT(IO)        CBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                                      ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:86:31: note: in expansion of macro 'CBI'
 #define _SET_INPUT(IO)        CBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                               ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:98:35: note: in expansion of macro '_SET_INPUT'
 #define SET_INPUT_PULLUP(IO)  do{ _SET_INPUT(IO); _WRITE(IO, HIGH); }while(0)
                                   ^
Marlin\src\module\endstops.cpp:248:7: note: in expansion of macro 'SET_INPUT_PULLUP'
       SET_INPUT_PULLUP(E_MIN_PIN);
       ^
In file included from Marlin\src\module\../inc/../HAL/HAL_AVR/HAL.h:36:0,
                 from Marlin\src\module\../inc/MarlinConfig.h:35,
                 from Marlin\src\module\endstops.h:33,
                 from Marlin\src\module\endstops.cpp:32:
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:82:41: error: 'DIOE_STOP_PIN_RPORT' was not declared in this scope
 #define _WRITE(IO,V)          do{ if (&(DIO ## IO ## _RPORT) < (uint8_t*)0x100) _WRITE_NC(IO,V); else _WRITE_C(IO,V); }while(0)
                                         ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:98:51: note: in expansion of macro '_WRITE'
 #define SET_INPUT_PULLUP(IO)  do{ _SET_INPUT(IO); _WRITE(IO, HIGH); }while(0)
                                                   ^
Marlin\src\module\endstops.cpp:248:7: note: in expansion of macro 'SET_INPUT_PULLUP'
       SET_INPUT_PULLUP(E_MIN_PIN);
       ^
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\module\../inc/MarlinConfig.h:33,
                 from Marlin\src\module\endstops.h:33,
                 from Marlin\src\module\endstops.cpp:32:
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:72:14: error: 'DIOE_STOP_PIN_WPORT' was not declared in this scope
   if (V) SBI(DIO ## IO ## _WPORT, DIO ## IO ## _PIN); \
              ^
Marlin\src\module\../inc/../core/macros.h:79:19: note: in definition of macro 'SBI'
 #define SBI(n,b) (n |= _BV(b))
                   ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:82:81: note: in expansion of macro '_WRITE_NC'
 #define _WRITE(IO,V)          do{ if (&(DIO ## IO ## _RPORT) < (uint8_t*)0x100) _WRITE_NC(IO,V); else _WRITE_C(IO,V); }while(0)
                                                                                 ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:98:51: note: in expansion of macro '_WRITE'
 #define SET_INPUT_PULLUP(IO)  do{ _SET_INPUT(IO); _WRITE(IO, HIGH); }while(0)
                                                   ^
Marlin\src\module\endstops.cpp:248:7: note: in expansion of macro 'SET_INPUT_PULLUP'
       SET_INPUT_PULLUP(E_MIN_PIN);
       ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:73:14: error: 'DIOE_STOP_PIN_WPORT' was not declared in this scope
   else   CBI(DIO ## IO ## _WPORT, DIO ## IO ## _PIN); \
              ^
Marlin\src\module\../inc/../core/macros.h:80:19: note: in definition of macro 'CBI'
 #define CBI(n,b) (n &= ~_BV(b))
                   ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:82:81: note: in expansion of macro '_WRITE_NC'
 #define _WRITE(IO,V)          do{ if (&(DIO ## IO ## _RPORT) < (uint8_t*)0x100) _WRITE_NC(IO,V); else _WRITE_C(IO,V); }while(0)
                                                                                 ^
Marlin\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:98:51: note: in expansion of macro '_WRITE'
 #define SET_INPUT_PULLUP(IO)  do{ _SET_INPUT(IO); _WRITE(IO, HIGH); }while(0)
                                                   ^
Marlin\src\module\endstops.cpp:248:7: note: in expansion of macro 'SET_INPUT_PULLUP'
       SET_INPUT_PULLUP(E_MIN_PIN);
       ^
In file included from Marlin\src\module\../inc/MarlinConfig.h:37:0,
                 from Marlin\src\module\motion.h:36,
                 from Marlin\src\module\motion.cpp:32:
Marlin\src\module\motion.cpp: In function 'void homeaxis(AxisEnum)':
Marlin\src\module\../inc/../pins/pins.h:812:25: error: 'E_STOP_PIN' was not declared in this scope
       #define E_MIN_PIN E_STOP_PIN
                         ^
Marlin\src\module\motion.cpp:1439:30: note: in expansion of macro 'E_MIN_PIN'
       (axis == _AXIS(A) && ((A##_MIN_PIN > -1 && A##_HOME_DIR < 0) || (A##_MAX_PIN > -1 && A##_HOME_DIR > 0)))
                              ^
Marlin\src\module\motion.cpp:1442:15: note: in expansion of macro 'CAN_HOME'
           && !CAN_HOME(E)
               ^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\megaatmega2560\src\src\module\endstops.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\src\module\motion.cpp.o] Error 1
============================================================ [FAILED] Took 15.25 seconds ============================================================

Environment           Status    Duration
--------------------  --------  ------------
megaatmega2560        FAILED    00:00:15.251

hobiseven, 2020-04-07:
Looks like you have a missing pin somewhere... like the E_MIN_PIN??? or badly named pin..

DerAndere1, 2020-04-07:
which branch was checked out when you compiled? If in doubt: how did you get the code? I guess you try to compile the branch Marlin2ForPipetBot-v1 for RAMPS with the default configs from https://github.com/DerAndere1/Marlin/blob/Marlin2ForPipetBot_v1/Marlin/src/config/default/Configuration.h ? Then indeed, you have to ´#define`the `E_STOP_PIN` in your pins_RAMPS.h file. Look for a pin number that is not used, maybe you can change the LED pin to `#define LED_PIN -1` and repurpose pin 13 (`#define E_STOP_PIN 13`)

igorlab, 2020-04-07:
Dear DerAndere1!
Thanks for your replay, it works!

closed 2020-04-07 by igorlab