DerAndere1/Marlin2ForPipetBot #15: could not compile as additional axis not defined??

Baltojikale, 2020-04-21:
Hello, 
kinda noob with good old c/c++ code syntax. Getting compiling errors as 4th axis pins arent defined..  Compiling for creality 2.4.1 motherbord based on ramps 1.4. For compiling used Marlin-Marlin2ForPipetBot-2.0.0. 

`In file included from Marlin\src\Marlin.cpp:259:0:
Marlin\src\Marlin.cpp: In function 'bool pin_is_protected(pin_t)':
Marlin\src\pins/sensitive_pins.h:158:17: error: 'I_STEP_PIN' was not declared in this scope
 #define _I_PINS I_STEP_PIN, I_DIR_PIN, I_ENABLE_PIN, _I_MIN _I_MAX _I_MS1 _I_MS2 _I_MS3 _I_CS
                 ^
Marlin\src\pins/sensitive_pins.h:642:3: note: in expansion of macro '_I_PINS'
   _I_PINS \
   ^
Marlin\src\Marlin.cpp:262:49: note: in expansion of macro 'SENSITIVE_PINS'
   static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;`

error repeats with I_STEP_PIN; I_DIR_PIN; I_ENABLE_PIN;
What Im missing??
thank you.

Baltojikale, 2020-04-22:
ok that problem overcome when understood which that 
Marlin-Marlin2ForPipetBot-2.0.0\Marlin\src\pins\ramps\pins_RAMPS.h was the targeted pin file notMarlin-Marlin2ForPipetBot-2.0.0\Marlin\src\pins\pins_RAMPS.h

also gcode\bedlevel\abl\ g29.cpp so i overwrited it with file from printer specific build.

DerAndere1, 2020-04-28:
> also gcode\bedlevel\abl\ g29.cpp so i overwrited it with file from printer specific build.

So there was a compile error in g29.cpp? Thanks for the info. I cannot test that feature in hardware, so I am happy to accept pull requests to fix the issue.

Baltojikale, 2020-04-28:
Few things to point out somewhere in your builds i saw that one should not us any Auto bed leveling, so that was first problem why it did not compile. After lots of struggling I've managed to adapt Pipetbot2 for cr10s pro printer with integrated touchscreen. Also with working 4th axis which currently configured to switch 5 channel peristaltic pump ( axis switches between active channels). i cant remember though which g29.cpp that worked for me, but basically one of InsanityAutomation releases, tried g29.cpp from either marlin 1.1.9 or 2.0 release .. ... Also current setup has auto bed leveling which i hoped to preserve... Most code editing was oriented to extensible ui particularly ui_api.cpp and printer specific Creality_DWIN.cpp.
If You are interested ill leave link to addapted fw:
https://drive.google.com/file/d/1O3AdhODX0gkJyrVSRTeOIhtUjTXDBayg/view?usp=sharing

And thank you for developing this multi axis support.  In my opinion it will be very useful fw for multi tooled or additional functions enabled machines

closed 2021-07-21 by DerAndere1