DerAndere1/Marlin2ForPipetBot #16: Homing E axis kills printer

touwen75, 2020-04-22:
Hello,

I have designed and build a SLS printer. It is a corexy system with a laser diode. The Z-axis is used to move the wiper, the E0 axis controls the buildplate en the E1 axis controls the powder buffer.

The mainboard is an bigtreetech SKR1.3, with a TFT3.5 v3 display. All 5 axis are TMC2130 SPI
At the moment only the wiper (z axis) has sensorless homing, the other axis use a mechanical end switch.

I want to use this Marlin 2.0 E-axis homing branch because it look like it is exactly what I need. So thank you guys for developing it!

However I did need some tinkering to get it to work. Mind you I am not a programmer, so I am don't know what the implication are of these changes, but this is what I did:

In types.h line42 : 
changed "E_AXIS   = 3," to " E_AXIS   = 3, E_HEAD = 3,"

In language.h
[Configuration.zip](https://github.com/DerAndere1/Marlin/files/4517160/Configuration.zip)
 : added two line.
#define STR_E_MIN                           "E_min" 
#define STR_E_MAX                           "E_max" 

I had to comment out: //#define EMERGENCY_PARSER  Because it gave a hole bunch of errors that ended with:   ......fatal error: ../inc/MarlinConfigPre.h: No such file or directory

Now it does compile, and I can home and control the the XY and Z axis via the UBS and the TFT display.
M119 shows the switches for X Y Z and E work as they should.

And now for my real problem:

When I give a G28 E command the printer halts and the display shows "printer halted, killed"

Also giving a "G0 E10 F20" or "G0 E-10 F20" command, and manually pressing the end switch does not stop the movement.

Does anybody know how to resolve this, or it it a bug?

-->

G28 Kills the printed, and needs to reset.

### Steps to Reproduce

1. Start with Marlin 2.0 E-axis homing branch
2. configuring to the setup as described above
3. Change the 3 lines of code as described above
4. Build the Firmware.bin and upload it to the printer
5. Give a G28 E command

**Expected behavior:** 

That the E-axis moves down until it hits a switch.

**Actual behavior:**

Printer stops working and shows a error message"printer halted , killed"

#### Additional Information

I really hope some some one get help me with this issue

Best Regards,

Erwin
[Configuration.zip](https://github.com/DerAndere1/Marlin/files/4517186/Configuration.zip)

touwen75, 2020-04-24:
After some trial and error I have solved my problem partially:
I defined the single nozzle extruder feature, but I had to comment it out: 
//#define SINGLENOZZLE
Now the G28 command does not halt the printer anymore!

But still it does not home the E-axis.....

I have read that this branch is not for corexy printers, and only for 4 axis homing.
but does this mean it is not tested, or is there a known problem if I do configure it for a 5 axis corexy setup?

Best regards,

Erwin

DerAndere1, 2020-04-28:
You can try the branch https://github.com/DerAndere1/Marlin/tree/2.0.x_E_homing. Make sure you do not use any of the example configs. instead, use the default configs that are located in the Marlin/ folder  and change every parameter to match your hardware. Also, in your pins_YOURMOTHERBOARD.h file, add a definition of the `E_STOP_PIN`  (use a free pin number)

DerAndere1, 2020-04-28:
It is not tested for Core kinematics. I checked the diff (https://github.com/DerAndere1/Marlin/commit/bec35b3feaa83ec3805b8ec851ed883a1a042567) and it looks as if E_AXIS_HOMING is compatible with corexy. It is just not tested.

DerAndere1, 2020-04-28:
In Configuration_adv.h line 596 you should have (I forgot to add that)
```
#if ENABLED(E_AXIS_HOMING)
#define E_HOME_BUMP_MM 2
#define HOMING_BUMP_DIVISOR { 2, 2, 4, 4 }
#else
  #define HOMING_BUMP_DIVISOR { 2, 2, 4 }  // Re-Bump Speed Divisor
#endif
```

I guess this is a bug, but maybe test what happens with different combinations of ENDSTOPPULLUP_EMIN, ENDSTOPPULLDOWN_EMIN and E_MIN_ENDSTOP_INVERTING. You can enable DEBUG_LEVELING_FEATURE (i think it is in configuration_adv.h) and PINS_DEBUGGING . Then you can continuously monitor the endstops with M43 E1 and debug the homing with M111 S38.

DerAndere1, 2020-04-28:
Also, first make it work with #define EXTRUDERS 1 before adding complexity with EXTRUDERS 2

DerAndere1, 2020-04-29:
I fixed some of your problems with commit https://github.com/DerAndere1/Marlin/commit/33a7d78f4de5cc628ba96e2471ac9de8770afb45

touwen75, 2020-04-30:
Thank you very much!
I will test it and report back...

touwen75, 2020-05-02:
I have just tried commit 33a7d78 with one extruder on my CoreXY, and it works! 
G28 E moved to the end switch and set the e-axis to zero.

Adding #define DEBUG_LEVELING_FEATURE did not compile.
Adding #define PINS_DEBUGGING compiled okay.

Then I changed the config to 2 extruders but that gave a whole bunch of errors and it did not compile:

Linking .pio\build\LPC1768\firmware.elf
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/gcode/calibrate/G28.cpp.o: in function `GcodeSuite::G28()':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\gcode\calibrate/G28.cpp:335: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/gcode/control/T.cpp.o: 
in function `GcodeSuite::T(unsigned char)':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\gcode\control/T.cpp:69: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/gcode/gcode.cpp.o: in function `GcodeSuite::get_target_extruder_from_command()':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\gcode/gcode.cpp:104: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/gcode/motion/G2_G3.cpp.o: in function `ABS<float>':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\gcode\motion\../../inc/../core/macros.h:271: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/gcode/temp/M106_M107.cpp.o: in function `GcodeSuite::M106()':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\gcode\temp/M106_M107.cpp:61: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/lcd/menu/menu_advanced.cpp.o:D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\lcd\menu/menu_advanced.cpp:162: more undefined references to `active_extruder' follow
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/module/motion.cpp.o: in function `CDCSerial::write(unsigned char)':
C:\Users\Gebruiker\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:173: undefined reference to `extruder_position'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Gebruiker\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:173: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/module/motion.cpp.o: in function `line_to_current_position(float const&)':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\module/motion.cpp:398: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/module/motion.cpp.o: in function `update_software_endstops(AxisEnum, unsigned char, unsigned char)':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\module/motion.cpp:816: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/module/motion.cpp.o: in function `prepare_line_to_destination()':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\module/motion.cpp:1308: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/module/motion.cpp.o: in function `do_homing_move(AxisEnum, float, float)':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\module/motion.cpp:1580: undefined reference to `active_extruder'
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/module/temperature.cpp.o:D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\module/temperature.cpp:436: more undefined references to `active_extruder' follow
c:/users/gebruiker/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/src/src/module/tool_change.cpp.o: in function `tool_change(unsigned char, bool)':
D:\tools\Marlin-2.0.x_E_homing 33a7d78/Marlin\src\module/tool_change.cpp:1082: undefined reference to `extruder_position'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1768\firmware.elf] Error 1

hobiseven, 2020-05-02:
I have tried to go to zero extrudées and that failed. I never tried 2 extruders. All the modifications in the code do have a similar pattern. 
First thing you have to check is wether or not all the Config.h and config_adv have all and enough parameters ( speed, jerk, etc)

DerAndere1, 2020-05-03:
Hi, Thanks for the feedback. I added a fix to be able to compile EXTRUDERS > 1 as well as DEBUG_LEVELING. Hope that goes into the right direction. Here the commit: https://github.com/DerAndere1/Marlin/commits/2.0.x_E_homing . @all be very clear about which branch you are talking. I update the bug description. This issue is about the 2.0.x_E_homing branch.

DerAndere1, 2020-05-14:
@touwen75 you can close this issue if your machine moves as expected.

touwen75, 2020-05-18:
@ DerAndere1: Thank you for your help!

closed 2020-05-18 by touwen75