DerAndere1/Marlin2ForPipetBot #17: Problems with compiling for SKR Pro v1.1

heppstar, 2020-05-12:
Hi,

First of all, thank you for adding 6axis support to Marlin! Best ever! I really need it for my CNC.

I have no problem at all to compile "Marlin-2.0.x_E_homing" for the "SKR Pro v1.1", but it will not compile "Marlin-bf2_6axis_dev14" or "Marlin-Marlin2ForPipetBot" with the same settings. Could you PLEASE help me!?

I only made the following changes:

**Configuration.h**
#define SERIAL_PORT -1

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_1
#endif

**Platformio.ini**
[platformio]
src_dir = Marlin
boards_dir = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_SKR_PRO

**Fault Marlin-df2_6axis_dev14**
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\SoftwareSerial.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:37,
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\eeprom_flash.cpp.o
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:25:
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:884: error: unterminated #if
  884 | #if ENABLED(CLASSIC_JERK)
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:813: error: unterminated #if
  813 | #if NON_E_AXES == 5
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:811: error: unterminated #if
  811 | #if NON_E_AXES == 6
      | 
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\Sd2Card_sdio_stm32duino.cpp:23:
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:884: error: unterminated #if
  884 | #if ENABLED(CLASSIC_JERK)
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:813: error: unterminated #if
  813 | #if NON_E_AXES == 5
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:811: error: unterminated #if
  811 | #if NON_E_AXES == 6
      | 
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\eeprom_flash.cpp:25:
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:884: error: unterminated #if
  884 | #if ENABLED(CLASSIC_JERK)
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:813: error: unterminated #if
  813 | #if NON_E_AXES == 5
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:811: error: unterminated #if
  811 | #if NON_E_AXES == 6
      | 
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\Servo.cpp:25:
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:884: error: unterminated #if
  884 | #if ENABLED(CLASSIC_JERK)
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:813: error: unterminated #if
  813 | #if NON_E_AXES == 5
      | 
Marlin\src\HAL\STM32\../../inc/../../Configuration.h:811: error: unterminated #if
[... 224 lines cut ...]
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\Servo.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/HAL.h:99,
                 from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32\persistent_store_sdcard.cpp:29:
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/timers.h:122:26: error: variable or field 'Step_Handler' declared void
  122 | extern void Step_Handler(stimer_t *htim);
      |                          ^~~~~~~~
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/timers.h:122:26: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
  122 | extern void Step_Handler(stimer_t *htim);
      |                          ^~~~~~~~
      |                          timer_t
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/timers.h:122:36: error: 'htim' was not declared in this scope; did you mean 'tm'?
  122 | extern void Step_Handler(stimer_t *htim);
      |                                    ^~~~
      |                                    tm
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/timers.h:123:26: error: variable or field 'Temp_Handler' declared void
  123 | extern void Temp_Handler(stimer_t *htim);
      |                          ^~~~~~~~
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/timers.h:123:26: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
  123 | extern void Temp_Handler(stimer_t *htim);
      |                          ^~~~~~~~
      |                          timer_t
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\HAL_STM32\persistent_store_sdcard.cpp.o] Error 1

DerAndere1, 2020-05-14:
Hi, general compil errors are fixed, at least for AVR, with https://github.com/DerAndere1/Marlin/commit/345335330560ec4f2d21208cb24ab2bd8dbd58b9. You can attach your Config so I can see which features you need. Currently, not every feature of Marlin is ready for 6 axis. Also you could help me by giving me an update if issue https://github.com/DerAndere1/Marlin/issues/7 is still occuring. Depending on your CAM software, you may need different axis names. To tackle that, I am currently working on branch https://github.com/DerAndere1/Marlin/tree/bf2_6Xdev_axis_rename (untested). Feel free to test (carefully) and report issues.

heppstar, 2020-05-14:
Hi,

Thank you. Right now I am writing my G-code by hand to test the robot that I built. Very much like you pipett robot. I have attached my "Configuration", "Configuration_adv" and my "pins_BTT_SKR_PRO_V1_1". And I will help you with as much as I can when my robot starts moving! :)

I think one of the problem that I have with "Marlin-Marlin2ForPipetBot" is that it is built on Marin 2.0 and not Marlin "2.0 Bugfix" The Bugfix version solved som big problems with the new SKR Pro v1.1 card. But the  "Marlin-bf2_6axis_dev14" should work fine...If I get the compile error so work.

[Modified files for SKR Pro v1.1.zip](https://github.com/DerAndere1/Marlin/files/4628768/Modified.files.for.SKR.Pro.v1.1.zip)

DerAndere1, 2020-05-14:
Marlin2ForPipetBot is based on Marlin branch bugfix-2.0.x commit https://github.com/MarlinFirmware/Marlin/commit/ee7558a6228747a7502ab50e2817234fb8a6feb4 which is pre-Marlin2.0.0. So indeed, that branch will be a bad choice for the BTT Pro V1.1. After commit https://github.com/DerAndere1/Marlin/commit/ae2850f7e9777c034970b9f8dce35fbf6cf33048 and setting `default_envs BIGTREE_SKR_PRO` in platformio.ini I was able to compile the bf2_6axis_dev14 with your config ( I had to update your Configuration_adv.h as attached  below:

[Configuration_adv.zip](https://github.com/DerAndere1/Marlin/files/4629439/Configuration_adv.zip)

DerAndere1, 2020-05-14:
Are you sure you need 6 homing axes XYZIJK PLUS 3 extruders? If you want a setup with independent X cariages and 3 extruders, that should be possible with default Marlin 2.0.x (see features like DUAL_X_CARRIAGE  (for independent extruders (IDEX setup)) or MIXING_EXTRUDER  or PARKING_EXTRUDER) or SINGLENOZZLE

heppstar, 2020-05-14:
Hi,

Thank you very much. I will try to compile it. No I got that wrong in my conf. The SKR Pro v1.1 board only have 6 TMC2209 so I need XYZIJK and no extruders. So I think my setup should be very much like your robot!

DerAndere1, 2020-05-14:
I found some more code related to TMC, see latest commit in the bf_6axis_dev14 branch.

heppstar, 2020-05-14:
I will. I am still having trouble compiling the code...

heppstar, 2020-05-14:
So, your latest commit with my "configuration" and the "configuration_adv" that you send me gives me compile error. If I comment out the DRIVER_TYPE TMC TMC2209 it works. There seems to be some problem with _DRIVER_TYPE TMC2209...

With:
#define X_DRIVER_TYPE TMC2209
#define Y_DRIVER_TYPE TMC2209
#define Z_DRIVER_TYPE TMC2209
#define I_DRIVER_TYPE  TMC2209
#define J_DRIVER_TYPE  TMC2209
#define K_DRIVER_TYPE  TMC2209

**Compile Error:**
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\module\planner_bezier.cpp.o
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::report(bool)':
Marlin\src\module\configuration_store.cpp:3736:25: error: 'SERIAL_ECHO_P' was not declared in this scope; did you mean 'SERIAL_ECHO_F'?
 3736 |             if (chop_i) SERIAL_ECHO_P(SP_I_STR);
      |                         ^~~~~~~~~~~~~
      |                         SERIAL_ECHO_F
Marlin\src\module\configuration_store.cpp:3738:27: error: 'SERIAL_ECHO_P' was not declared in this scope; did you mean 'SERIAL_ECHO_F'?
 3738 |               if (chop_j) SERIAL_ECHO_P(SP_J_STR);
      |                           ^~~~~~~~~~~~~
      |                           SERIAL_ECHO_F
Marlin\src\module\configuration_store.cpp:3740:29: error: 'SERIAL_ECHO_P' was not declared in this scope; did you mean 'SERIAL_ECHO_F'?
 3740 |                 if (chop_k) SERIAL_ECHO_P(SP_K_STR);
      |                             ^~~~~~~~~~~~~
      |                             SERIAL_ECHO_F
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\module\printcounter.cpp.o
*** [.pio\build\BIGTREE_SKR_PRO\src\src\module\configuration_store.cpp.o] Error 1

DerAndere1, 2020-05-14:
I forgot to push some changes related to TMC, fixed with latest commit in the bf_6axis_dev14 branch (commit https://github.com/DerAndere1/Marlin/commit/5b5cea83a6977ff5853dd139bf1f3d85c6db307a). Sorry that I bothered you with non-compiling commits, I am not a professional programmer.

 If you have no E-axis (no conventional extruders), then set EXTRUDERS 1 (or even try to set EXTRUDERS 0) in your Configuration.h. Also, you may have to change `J_MIN_ENDSTOP_INVERTING false` and `K_MIN_ENDSTOP_INVERTING false` to true, depending on if your limit switches are normally open or closed.  If you need support for Trinamic-specific G-codes, look at folder src/feature/ and go through all its files and subfolders with "trinamic" or "tmc" in the filename and verify they have code for NON_E_AXES > 3

heppstar, 2020-05-14:
Thank you VERY much! You solved the compile problem for TMC! As soon as I have the SKR card I will test to run all the axis!

What is the man differences between Marlin2ForPipetBot and Marlin-bf2_6axis_dev14?

DerAndere1, 2020-05-14:
Marlin2ForPipetBot is an older but more stable branch based on Marlin branch bugfix-2.0.x commit MarlinFirmware@ee7558a which is pre-Marlin2.0.0 with added 6axis support. So far it works as expected for my PipetBot-A8. It is better documented but suffers from all issues that were fixed in upstream MarlinFirmware/Marlin since september 2018 and maybe more.

Marlin-bf2_6axis_dev14 is one of the development branches that is aimed at preparing a pull request to bring 6axis support to upstream MarlinFirmware/Marlin. It is based on recent MarlinFirmware/Marlin bugfix-2.0.x (post Marlin 2.0.5 release). Marlin-bf2_6axis_dev14 is not tested, I guess it suffers from issue 7.

closed 2020-08-19 by DerAndere1