DerAndere1/Marlin2ForPipetBot #19: [Bug] [bf2_6axis_dev14] regression: IJK endstop validation during homing kills printer

DerAndere1, 2020-05-22:
### Description

Regression: axis I moves with G1 but does not move during G28 auto home. Therefore, IJK endstop validation during homing kills the machine.  

### Steps to Reproduce

1. Upload bf2_6axis_dev14
2. send G28
3. After XYZ are homed, when axis I should move, 

**Expected behavior:** [What you expect to happen]

G28 should sucessfully home axes IJK

**Actual behavior:** [What actually happens]

Axis I it does not move during G28. Because I endstops are not triggered, `endstop_validation()` in motion.cpp results in the kill error message 

#### Additional Information

This bug was not presennt in bf2_6axis_dev13 or bf2_6axis_dev12

heavylight-repo, 2020-06-25:
Did you find any solution to this problem?

DerAndere1, 2020-08-18:
Should be fixed with today's commit. Reopen if you still encounter this bug

closed 2020-08-18 by DerAndere1