DerAndere1/Marlin2ForPipetBot #20: I-axis only moves in one direction regardless of coordinate

heppstar, 2020-07-01:
### Description

I-axis only moves in one direction regardless of coordinate.

Hardware: Prototype Robot with SKR PRO v1.1 and six axis (TMC2209)
Marlin Dist: BF2_6axis_dev14 0f8be77

### Steps to Reproduce

1. Send "G1 I100.0" from Pronterface
2. Send "G1 I10.0" from Pronterface
3. Send "G1 I100.0" from Pronterface

**Expected behavior:** [What you expect to happen]

1. "I" move right
2. "I" move left
3. "I" move right

**Actual behavior:** [What actually happens]

1. "I" move right
2. "I" move right
3. "I" move right

#### Additional Information

It in the expected direction for XYZ and JK.

I would be glad help you test all the new commits that are coming! This is the best project ever on Github! :)

#### My config
[Config_files.zip](https://github.com/DerAndere1/Marlin/files/4859386/Config_files.zip)

DerAndere1, 2020-08-18:
Can you test again with the recent commit? does behaviour of the I axis change if you change to NON_E_AXES 4 or NON_E_AXES 5

DerAndere1, 2020-09-18:
Should be fixed. Use 6axis_PR1 branch with commit https://github.com/DerAndere1/Marlin/commit/449f369f35c0ba54515cd5819d3de2e5ccf53a58 or later

closed 2020-09-18 by DerAndere1