DerAndere1/Marlin2ForPipetBot #24: Supported boards

rampa069, 2020-07-15:
Hi, what boards are supported?  I need it with a 6 axis board (i have some STM32 based boards) and want to port to them.

Thanks in advance.

DerAndere1, 2020-08-19:
Try it with any of the bords you have that are supported by upstream Marlin, make it work and report back. The newer the hardware, the less likely it will work well out of the box. Just do not buy a new board explicitely to test my code. Personally, I can only verify that options listed in https://github.com/DerAndere1/Marlin/issues/13 work. 
In addition, the bf2_6axis_dev14 branch compiles with Trinamic TMC drivers but you have to test if it works in practice.

Aeronut, 2020-08-24:
Thanks for the quick lesson on environment/pins dependency and the Jedicut link!
I was able to achieve the above and compile successfully, but once I changed environments and with the specific amendments to pins_BTT_SKR_V1_3.h, I get the following after setting #define NON_E_AXES 4 in configuration.h; 
types.h - could not convert 'o' from 'XYZval<float>' to 'XYZOnlyval<float>' [1283, 112] [1284, 112]
I've raised this as a separate issue, rightly or wrongly.
Almost there, I will upload my config as requested once successful.

DerAndere1, 2020-09-18:
@Aeronut : make sure you use the 6axis_PR1 branch as of commit https://github.com/DerAndere1/Marlin/commit/449f369f35c0ba54515cd5819d3de2e5ccf53a58 or later.

hobiseven, 2020-09-19:
@rampa069  I did made the very first tests on an STM32F103, and the basis was working with the foamcutter. You may have to make minor text editions in the Jedicut generated Gcode, but really small ones in the header file. Syntax is not exactly the one of Marlin for feedrates as far as I remember. 

@DerAndere1 Apparently, there are a massive set of contributions and yourself. What is the status / upstream marlin. I received a few notifications from Github, and apparently the redesign of the variable structure is overcomed, right? 

Great your you did anyway.!

DerAndere1, 2020-09-20:
@hobiseven: Yes, after a long coding-pause I revived the project, made the axis names configurable and opened a pull request https://github.com/MarlinFirmware/Marlin/pull/19112 . That move atracted some other skilled contributors so we could properly fix some bugs. Best regards!

closed 2021-07-21 by DerAndere1