DerAndere1/Marlin2ForPipetBot #25: [bug] [bf2_6axis_dev14] M504 does NOT print "EEPROM OK" to stdout.

DerAndere1, 2020-08-19:
### Description

With branch bf2_6axis_dev14, EEPROM cannot be validated. M504 does NOT print "EEPROM OK" to stdout.

### Steps to Reproduce

1. enable (uncomment) option `#define EEPROM_SETTINGS` in configuration.h
2. reflash
3. Open Printrun pronsole and enter
```
connect
M502
M500
M504
```

**Expected behavior:** 
"EEPROM OK" printed to the console of pronsole

**Actual behavior:**
Observe that no "EEPROM OK" printed to the console of pronsole 

#### Additional Information

None

DerAndere1, 2020-08-19:
Error: Field planner_leveling_active mismatch

DerAndere1, 2020-09-18:
I guess we need to add dummy values somewhere between the last passing `_FIELD_TEST` and  `_FIELD_TEST(planner_leveling_active);`. See settings.cpp lines 702 - 755 and lines 1602 - 1551 .

bilsef, 2020-09-19:
#36 fixes this issue.

DerAndere1, 2020-09-19:
Thanks, I did not test in hardware but the code changes from PR https://github.com/DerAndere1/Marlin/pull/36 look absolutely fine, so I close this as fixed. Will reopen if this is still an issue. Thanks once again, @bilsef

DerAndere1, 2020-09-19:
Not fixed for me on a Melzi clone

bilsef, 2020-09-19:
What output do you get from M500 and M504?  I tried a couple of different boards and all axis settings and it worked.  There are a lot of settings that can change what is stored, and I have not tried them all, so maybe it is some particular combination that is causing the problem

>>> m500
SENDING:M500
echo:Settings Stored (733 bytes; crc 20487)
>>> m504
SENDING:M504
echo:EEPROM OK

DerAndere1, 2020-09-19:
Hah, it was indeed fixed by your PR. I missed one changed line when backporting it to my testing branch. Sorry for the noise. This is awesome. Closing as fixed again.

closed 2020-09-19 by DerAndere1