DerAndere1/Marlin2ForPipetBot #27: 6axis_PR1

Aeronut, 2020-08-24:
Description

<#define NON_E_AXES 3 - Line 170 in configuration.h>

### Steps to Reproduce

<Enter a config other than 3, (4 - 5 - 6) generates errors>

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [Types.h - could not convert 'o' from 'XYZval<float>' to 'XYZOnlyval<float>' [1283, 112] [1284, 112] - Sanitycheck.h - #error "Enable USE_JMIN_PLUG when homing J to MIN." [1996, 10]]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2020-08-26:
Can you retry with the current 6axis_PR1 branch? Also, you have to enable USE_JMIN_PLUG as the error message indicates.

Aeronut, 2020-08-26:
Hi, I used the latest PR1 branch at the time of reporting. Both 6axis-PR1 & dev suffer the same issue from memory. The USE_JMIN_PLUG is somewhat of a red herring, apologies, as I was testing all the options for NON_E_AXES, 5 in this instance hence J, and only X, Y, Z & I were configured in config.h.
Your build that I successfully replicated was void of this issue. 
I will happily re-run any test, but it won't be until Monday.

DerAndere1, 2020-08-27:
Ah ok. this can be closed then, as not a bug. The initial pull request contained a config file that had preconfigured values for all required parameters. The maintainer of upstream MarlinFirmware/Marlin did some cleanup, now you have to define the required values manually as described in the comment above the LINEAR_AXES option (that was previously called NON_E_AXES). Be careful about https://github.com/DerAndere1/Marlin/issues/30 . Right now it might be easier to make Z axis name configurable and change the FOAMCUTTER_XYUV-specific code in the file planner.cpp than making LINEAR_AXES 5 work.

closed 2020-08-27 by DerAndere1