DerAndere1/Marlin2ForPipetBot #28: 6axis_PR1

Aeronut, 2020-08-24:
### Description

<#define FOAMCUTTER_XYUV creates errors>

### Steps to Reproduce

<Un-comment the above in configuration.h Line 86 >

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [ Error - too few arguments to function 'float sqrtf(float)']

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2020-08-27:
FOAMCUTTER_XYUV currently requires at least LINEAR_AXES 5. Since   LINEAR_AXES 5 is currently broken (see https://github.com/DerAndere1/Marlin/issues/30). The fastest strategy to get out of this frustrating situation is to make the Z axsis name configurable and change the FOAMCUTTER_XYUV logic in the file planner.cpp to make it work with LINEAR_AXES 4.

DerAndere1, 2020-09-18:
Should be fixed. FOAMCUTTER_XYUV currently requires at least LINEAR_AXES 5. A sanity check is in place. You can use a dummy value for the J_STEP_PIN, J_DIR_PIN, J_ENABLE_PIN, J_STOP_PIN. You can open a feature request if you want compilation FOAMCUTTER_XYUV with LINEAR_AXES 4

closed 2020-09-18 by DerAndere1