DerAndere1/Marlin2ForPipetBot #29: 6axis_PR1 - 240820

Aeronut, 2020-08-24:
### Description

<TMC driver selection in configuration.h Lines 768 onwards>

### Steps to Reproduce

<If selecting any TMC driver from 2130 onwards in the list, produces error - All other drivers are fine>

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [drivers.h - missing binary operator before token "(" [89, 51]]

#### Additional Information. This didn't occur with 230820 release

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2020-08-27:
should be fixed in 6axis_PR1 branch now.

Aeronut, 2020-08-31:
Tested with the latest build, TMC selection is still an issue but the error has changed;
**macros.h - pasting "LIST_" and "(" does not give a valid preprocessing token [285, 25**]
As other stipulations exist, TMC drivers that dont support STEALTHCHOP are sometimes flagged.

DerAndere1, 2020-09-18:
Attach a ZIP file containing your Configuration.h and Configuration_adv.h files and your pins_YOURMOTHERBOARD.h file

DerAndere1, 2020-09-18:
Never mind, I could reproduce. 
Fixed with commit https://github.com/DerAndere1/Marlin/commit/4a43d74b1ad0219f725ab659fb27a2b64759153f

closed 2020-09-18 by DerAndere1