DerAndere1/Marlin2ForPipetBot #30: [bug] [6axis_dev] Safety risk: With LINEAR_AXES 5, axes move way too fast and too far!

DerAndere1, 2020-08-27:
### Description

With the branch https://github.com/DerAndere1/Marlin/tree/6axis_dev and https://github.com/DerAndere1/Marlin/tree/6axis_PR1, when I set NON_E_AXES 5 (LINEAR_AXES 5), all axes move about 4x faster and further than expected. This is a safety risk. I have to investigate further, but I am not sure if endstops work with LINEAR_AXES 5. At least the internally used steps per mm do not match the configured values for DEFAULT_STEPS_PER_UNIT. Be prepared to hit the emergency button to kill the printer movement. 

### Steps to Reproduce

Set NON_E_AXES 5 (or LINEAR_AXES 5), Set DEFAULT_MAX_FEEDRATE and STEPS_PER_UNIT of the I and J axis to the same value as for the Z axis. Send G28

**Expected behavior:** Normal  I, J and Z movement

**Actual behavior:** Axes (at least Z and I) move faster and further than expected, probably due to wrong internal steps per mm

#### Additional Information

Regression (bug is not present in branch https:/github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot which is based on pre-release Marlin-bugfix2.0.x)

DerAndere1, 2020-08-27:
In the branch 6axis_PR1 I now added a sanity check to prevent compilation with LINEAR_AXES 5. To test LINEAR_AXES >= 5, install an emergency kill switch to be able to instantly abort movement of the machine and delete the following lines from src/inc/SanityCheck.h:

#if LINEAR_AXES >= 5
  #error "LINEAR AXES >= 5 not yet supported, it causes unsafe movements. Developers should have a kill switch installed to stop movement before removing this check."
#endif

The isse still needs a fix

bilsef, 2020-09-16:
I did not encounter this issue in my testing.  Was it resolved?

DerAndere1, 2020-09-18:
thanks a lot @bilsef, fixed thanks to https://github.com/DerAndere1/Marlin/pull/31

closed 2020-09-18 by DerAndere1