DerAndere1/Marlin2ForPipetBot #39: #define EXTRUDERS 0

Aeronut, 2020-10-07:
[6AxisPR1.zip](https://github.com/DerAndere1/Marlin/files/5343459/6AxisPR1.zip)

### Description

### Steps to Reproduce

1. [Set EXTRUDERS to 0]
2. [Comment out #define DEFAULT_NOMINAL_FILAMENT_DIA 1.75]
3. [and so on...]

**Expected behavior:** [This to be allowed as per the latest bugfix-2.0.x release, to allow FoamCutter config to work on a five axis board - Linear Axis 5 being minimum at present, no room for extruder driver that will be expected, albeit unused]

**Actual behavior:** [Compiles with warnings, all relating to 'array subscript 5 is above array bounds of 'float [5]' [-Warray-bounds]' in planner.cpp, settings.cpp, menu_advanced.cpp & serial.h. 
Setting EXTRUDERS to 0 will compile, but 9 warnings are generated - Setting to one is warning free ]

#### Additional Information
types.h approx line 448 -  #if LINEAR_AXES == 5     //SG Changed from >=  (also config for 4 & 6) for another compile error

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2020-10-07:
Can you compile upstream MarlinFirmware/Marlin 2.0.6 with EXTRUDERS 0 while keeping the arrays in Configuration.h unmodified? Did you try deleting the last element of the following arrays in Configuration.h / Configuration_adv.h ?

```
DEFAULT_AXIS_STEPS_PER_UNIT
DEFAULT_MAX_FEEDRATE
DEFAULT_MAX_ACCELERATION
AXIS_RELATIVE_MODES
MICROSTEP_MODES
MANUAL_FEEDRATE
```

... Regarding the bug mentioned in the additional information: that might be related to https://github.com/MarlinFirmware/Marlin/pull/19112#issuecomment-704411620

DerAndere1, 2020-10-07:
As a workaround you can leave EXTRUDERS 1 and in the file pins_YOURMOTHERBOARD.h assign a dummy value to the E_STEP_PIN, E_DIR_PIN and E_ENABLE_PIN

Aeronut, 2020-10-07:
My fix ,==, is specific, but it works if I'm interpreting the fault correctly.
With regards pins, I had already done that as E0 step direction & enable are persistently required - It's a pity E0 can't be totally removed as part of #define FOAMCUTTER_XYUV

Aeronut, 2020-10-07:
DEFAULT_AXIS_STEPS_PER_UNIT     Prior to opening issue
DEFAULT_MAX_FEEDRATE               Prior to opening issue
DEFAULT_MAX_ACCELERATION      Prior to opening issue
AXIS_RELATIVE_MODES           Just tested - No difference
MICROSTEP_MODES                Just tested - No difference
MANUAL_FEEDRATE                 Just tested - No difference

Aeronut, 2020-10-07:
Compiled bugfix-2.0.x - No errors, no warnings, EXTRUDERS 0 as previously stated - I currently don't have 2.0.6 to try.
[bugfix-2.0.x 061020.zip](https://github.com/DerAndere1/Marlin/files/5344200/bugfix-2.0.x.061020.zip)

Aeronut, 2020-10-07:
Compiled Marlin 2.0.6 No errors, no warnings, EXTRUDERS 0 , arrays unmodified
[Marlin 2.0.6.zip](https://github.com/DerAndere1/Marlin/files/5344400/Marlin.2.0.6.zip)

DerAndere1, 2020-10-30:
Closing as fixed in branch 6axis_PR1 with commit https://github.com/DerAndere1/Marlin/commit/8e8e2bf9d26e8175e8da419c8f642bd6c5a9c2d8 .

closed 2020-10-30 by DerAndere1