DerAndere1/Marlin2ForPipetBot #40: [FR] 6axis_dev: Customized Gcode for tool change

GenaUser, 2020-10-18:
Could you add possibility to create a custom gcode executed when activating a tool and another when deactivating (identical that the Repetier firmware offers)
This allows all axes to be controlled depending on the tools state. For example, to manage a tools changer using a stepper motor through a fourth axis

Thanks in advance for studying this feature.

DerAndere1, 2020-10-26:
I am not sure I understand.  I am not familiar with Repetier firmware. Can you point me to information about the feature of Repetier  firmware you are talking about? Marlin already has the option EVENT_GCODE_AFTER_TOOLCHANGE in Configuration_adv.h, see https://github.com/MarlinFirmware/Marlin/blob/88a2ac92d5792de3de005b3b97d7bb561946c148/Marlin/Configuration_adv.h#L2020

GenaUser, 2020-10-26:
The problem with Marlin's "#define EVENT_GCODE_AFTER_TOOLCHANGE" is that the custom Gcode is the same for all tools. But maybe I misunderstood
The Repetier firmware allows the possibility to create 2 gcode for each tool: during selection and deselection.
In the following example, activating T0 tool moves the « A » axis to its origins (G28 A).
Activating T1 tool moves the « A » axis to 12 mm (G1 A12).

Repetier V2
// TOOL_EXTRUDER(name, offx, offy, offz, heater, stepper, resolution, yank, maxSpeed, acceleration, advance, startScript, endScript)
TOOL_EXTRUDER(ToolExtruder1, 0, 0, 0, HeaterExtruder1, E1Motor, 1.75, 220.0, 20, 30, 1000, 40, "G28 A", "", &Fan1PWM)
TOOL_EXTRUDER(ToolExtruder2, 12, 0, 0, HeaterExtruder2, E2Motor, 1.75, 220.0, 20, 30, 1000, 40, "G1 A12", "", &Fan1PWM)

It is not possible to do the same with a single Gcode, as it seems to be with Marlin.
But maybe I'm wrong.

Repetier V1 had specific variables:
EXT0_SELECT_COMMANDS = ""
EXT0_DESELECT_COMMANDS = ""
EXT1_SELECT_COMMANDS = ""
EXT1_DESELECT_COMMANDS = ""

Thank you for your interest in my request.

DerAndere1, 2020-10-26:
Lets assume you use my branch 6axis_PR1 (which currently has more bugfixes than 6axis_dev) and your 4th axis drives the tool changer and you have in Configuration.h `#define AXIS4_NAME 'A'`. As I understand:

- tool change command `T0` should move the 4th axis (A axis) to its origin (A0)
- tool change command `T1` should move the 4th axis to e.g. 12mm from origin (A12). 

To make it work, I propose the following change (untested): The following lines in the file Marlin/src/module/tool_change.cpp:

https://github.com/DerAndere1/Marlin/blob/c7b99451e77bcbda696049e244f51dce210552b2/Marlin/src/module/tool_change.cpp#L1192-L1195

should be changed to 

```
    #ifdef EVENT_GCODE_AFTER_TOOLCHANGE
      if (!no_move && TERN1(DUAL_X_CARRIAGE, dual_x_carriage_mode == DXC_AUTO_PARK_MODE))
        #if EVENT_GCODE_AFTER_TOOLCHANGE_T1
          if (tool_index == 1) {
            gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_AFTER_TOOLCHANGE_T1));
          }
         else {
           gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_AFTER_TOOLCHANGE));
         }
        #else
          gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_AFTER_TOOLCHANGE));
        #endif
    #endif

```

## Usage

- Your start script includes a `G28` to home all axes (XYZA) and leaves the 4th axis at the origin (A0). 
- Your slicer produces Gcode in absolute position coordinates (absolute mode, G90).
- You `#define AXIS4_NAME 'A'` in the file Configuration.h.
- You `#define DEFAULT_STEPS_PER_UNIT { 100, 100, 10, 20, 30, 40 }` in the file Configuration.h, which means that your A axis has 20 steps/mm (as an example).
- You do the following in the file Configuration_adv.h:
    `#define EVENT_GCODE_AFTER_TOOLCHANGE "G28 A\nG1 A0" // G-code script after tool change command T0`
- You insert the following line in the file Configuration_adv.h:
    b) `#define EVENT_GCODE_AFTER_TOOLCHANGE_T1 "G1 A12" // G-code script after tool change command T1`
- You change tools by sending the following tool change commands: `T0` or `T1`

GenaUser, 2020-10-27:
Thanks for your proposition. In theory, this should work.
I will try to configure your 6axis_PR1 and then test it on my test bench.
It may take a little time while as I am not used to using Marlin.
Thank you for your work

DerAndere1, 2020-10-27:
I am actually not sure under what circumstances the EVENT_GCODE_AFTER_TOOLCHANGE is processed in original Marlin since it is not documented well. Maybe it is better to do the following change in Marlin/Marlin/src/module/tool_change.cpp:

https://github.com/DerAndere1/Marlin/blob/c7b99451e77bcbda696049e244f51dce210552b2/Marlin/src/module/tool_change.cpp#L1192-L1195

should be changed to
```
    #ifdef EVENT_GCODE_AFTER_TOOLCHANGE_T0
      if ((!no_move) && (new_tool == 0)) {
        gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_AFTER_TOOLCHANGE_T0));
      }
    #endif
    
    #ifdef EVENT_GCODE_AFTER_TOOLCHANGE_T1
      if ((!no_move) && (new_tool == 1)) {
        gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_AFTER_TOOLCHANGE_T1));
      }
    #endif  

    #ifdef EVENT_GCODE_AFTER_TOOLCHANGE  
      if (!no_move && TERN1(DUAL_X_CARRIAGE, dual_x_carriage_mode == DXC_AUTO_PARK_MODE))
            gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_AFTER_TOOLCHANGE));
    #endif
```

Then, in Confguration_adv.h, you **don't** have to define EVENT_GCODE_AFTER_TOOLCHANGE.
You rather add the following two lines of code to Confguration_adv.h, e.g. 

https://github.com/DerAndere1/Marlin/blob/620a17e18759dd92ffa39f36116e4bb03e18c200/Marlin/Configuration_adv.h#L2078-L2080

should be changed to 
```
  #if ENABLED(TOOLCHANGE_NO_RETURN)
    //#define EVENT_GCODE_AFTER_TOOLCHANGE "G12X"   // Extra G-code to run after tool-change
  #endif  
  #define EVENT_GCODE_AFTER_TOOLCHANGE_T0 "G28 A\nG1 A0" // Extra G-code to run while executing tool change command T0
  #define EVENT_GCODE_AFTER_TOOLCHANGE_T1 "G1 A12" // Extra G-code to run while executing tool change command T1
 ```

GenaUser, 2020-10-27:
Noted. 
Thank you

GenaUser, 2020-10-30:
Hello,
I had compilation errors: "gcode" was not declared in this scope.
[image]

I worked around the problem by moving #define EVENT_GCODE_AFTER_TOOLCHANGE with a neutral gcode, as a result of
#define EVENT_GCODE_TOOLCHANGE_T0 and
#define EVENT_GCODE_TOOLCHANGE_T1

Which gives this:

#if HAS_MULTI_EXTRUDER
  // Z raise distance for tool-change, as needed for some extruders
  #define TOOLCHANGE_ZRAISE                 0 // (mm)
  //#define TOOLCHANGE_ZRAISE_BEFORE_RETRACT  // Apply raise before swap retraction (if enabled)
  //#define TOOLCHANGE_NO_RETURN              // Never return to previous position on tool-change
  #if ENABLED(TOOLCHANGE_NO_RETURN)
    //#define EVENT_GCODE_AFTER_TOOLCHANGE "G12X"   // Extra G-code to run after tool-change
  #endif  
  #define EVENT_GCODE_TOOLCHANGE_T0 "G28 A\nG1 A0" // Extra G-code to run while executing tool change command T0
  #define EVENT_GCODE_TOOLCHANGE_T1 "G1 A12" // Extra G-code to run while executing tool change command T1
  #define EVENT_GCODE_AFTER_TOOLCHANGE "G21"   // Extra G-code to run after tool-change

It's not a clean solution, but it allowed me to do some testing.
Your modification works.
Custom gcodes are executed correctly when tools change. However, you must first have homing all axes.
To ensure the change of head even if the origins are not made, I add "G28 A" and "G28 A12" to the "T0" and T1 "commands of my screen. This can create redundancy, but it does not should not create a problem.
I continue my tests.
Thank you for this modification which will undoubtedly allow me to use Marlin.

DerAndere1, 2020-10-30:
> Custom gcodes are executed correctly when tools change. However, you must first have homing all axes.

With your  

```
#define EVENT_GCODE_TOOLCHANGE_T0 "G28 A\nG1 A0" // Extra G-code to run while executing tool change command T0
#define EVENT_GCODE_TOOLCHANGE_T1 "G1 A12" // Extra G-code to run while executing tool change command T1
#define EVENT_GCODE_AFTER_TOOLCHANGE "G21" // Extra G-code to run after tool-change
```

if you have a G28 (home all axes) in the start-script of your print , can you have as many `T0` and `T1` in your print's G-code script as you like? (I am not talking about performing a toolchange from a screen menu) ? Or do you have to send `G28 A` before each tool change command?

Could you test branch https://github.com/DerAndere1/Marlin/tree/6axis_EVENT_GCODE_TOOLCHANGE and report if it works the same  ? it has a cleaner solution (see commit https://github.com/DerAndere1/Marlin/commit/fd6efc3a8b5813c629d637ca84422820caf2b94e)

DerAndere1, 2020-10-31:
How exactly did you add the `G28 A` to the screen command for tool-change? 

was the change in file menu_motion.h? 

https://github.com/DerAndere1/Marlin/blob/fd6efc3a8b5813c629d637ca84422820caf2b94e/Marlin/src/lcd/menu/menu_motion.cpp#L292-L313 

Was that part changed to something like the following?

```
    #elif EXTRUDERS == 3
      if (active_extruder < 2) {
        #if defined(EVENT_GCODE_TOOLCHANGE_T0) 
          if (active_extruder)
            GCODES_ITEM_N(0, MSG_SELECT_E, PSTR("G28 A\nT0"));
          else
            GCODES_ITEM_N(1, MSG_SELECT_E, PSTR("T1"));
        #else 
          if (active_extruder)
            GCODES_ITEM_N(0, MSG_SELECT_E, PSTR("G28 A\nT0"));
          else
            GCODES_ITEM_N(1, MSG_SELECT_E, PSTR("T1"));
        #endif
      }
    #else
      #if defined(EVENT_GCODE_TOOLCHANGE_T0) 
        if (active_extruder)
          GCODES_ITEM_N(0, MSG_SELECT_E, PSTR("G28 A\nT0"));
        else
          GCODES_ITEM_N(1, MSG_SELECT_E, PSTR("T1"));
      #else
        if (active_extruder)
          GCODES_ITEM_N(0, MSG_SELECT_E, PSTR("T0"));
        else
          GCODES_ITEM_N(1, MSG_SELECT_E, PSTR("T1"));
      #endif
    #endif

  #elif ENABLED(DUAL_X_CARRIAGE)
    #if defined(EVENT_GCODE_TOOLCHANGE_T0)
      if (active_extruder)
        GCODES_ITEM_N(0, MSG_SELECT_E, PSTR("G28 A\nT0"));
      else
        GCODES_ITEM_N(1, MSG_SELECT_E, PSTR("T1"));
    #else
      if (active_extruder)
        GCODES_ITEM_N(0, MSG_SELECT_E, PSTR("T0"));
      else
        GCODES_ITEM_N(1, MSG_SELECT_E, PSTR("T1"));
    #endif
  #elif 
```

GenaUser, 2020-10-31:
The screen modification does not concern Marlin, but my TFT35 Bigtreetech touch screen which drives a Rumba32 MKS board.
I added a specific menu for the heads with an icon  "T0" and "T1" to which I associated my gcodes commands.

[image]

I also added a "T" button in the home menu to initialize the "A" axis.

[image]

My printer is currently using an 8-bit Rumba+ board with Repetier-Firmware V1.
For update board in 32 bits, I compare Repetier V2 and Marlin V2

DerAndere1, 2020-10-31:
Thanks for the insight, I will close this then. You can reopen if there are issues remaining. In case your stepper driver for the A axis is different from A4988, I would love to hear what stepper driver features already work, but only if you have time.

closed 2020-10-31 by DerAndere1

GenaUser, 2020-11-01:
At first I had a DRV8825.
Now I have installed a TMC5160 SPI.
Everything seems to be working fine on my test rig at the moment.
Good work!