DerAndere1/Marlin2ForPipetBot #42: [FR]  No extruders (motors) but hotends (PID control for hotend temperature)

DerAndere1, 2021-01-05:
### Description

@Daniele-Dondi (https://github.com/Daniele-Dondi) opened the following feature request in the upstream MarlinFirmware/Marlin issue tracker;
Hi, I'm quite new here. I'm trying to use Marlin on my robot for chemical reactions. Actually the robot has 6 axes for driving syringes plus several servos for valves. I wish to implement the possibility to heat (by using PID) but the only way I found is to enable one extruder (that I don't need and also I haven't enough pins to use on my board).
So, I'm asking if it is possible to have hotends without extruders (motor).
I tried to modify the code without success.
(I know that I could use the bed pins but actually I need more hotends)

### Steps to Reproduce

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 
[Dondi.zip](https://github.com/DerAndere1/Marlin/files/5771306/Dondi.zip)

Daniele-Dondi, 2021-01-09:
Thank you for the hint. By putting a dummy extruder (setting the same unused pin for the three pins needed) I was able to compile it correctly. I've also setup the sw for 12 servos. I'm waiting for some hardware stuff  for the final check (I got stuck for some missing connectors and linear bearings). After the test if you are interested I can post here the files I modified.

DerAndere1, 2021-01-10:
Yes, definitely, I'm still interested in your progress and updated configs. Also, you may want to get in touch with user @scientistnobee (https://github.com/scientistnobee) by writing a comment at https://github.com/DerAndere1/contact_DerAndere1/issues/3 . (S)he is currently building a liquid handling system (stationary, no positioning) with multiple syringe pumps.

scientistnobee, 2021-01-13:
> by writing a comment at DerAndere1/contact_DerAndere1#3

Hi DerAndere,

Thanks. Whom should I contact particularly?

Daniele-Dondi, 2021-01-17:
Hi, I'm still alive. Now that I received the hw I'm having some problems.. I added 12 servos but only the first 4 moves.
Other than
pins_MEGATRONICS_3.h
Configuration.h
I modified also the files:
servo.h
servo.cpp
pins.h
I'm making some checks to avoid some trivial hw or sw errors

Daniele-Dondi, 2021-01-18:
Finally I think to have solved all the issues. 
Since I manipulated many files I prefer to attach the whole marlin (I started from marlin for pipetbot Derandere's version)
This version should work with the following hw:
-board Megatronics 3.0 
-6 stepper motors
-PID with thermocouple in the first slot and power exit on the first extruder as BED
-12 servos

On most (if not all) lines I modified the //Dondi comment was added

[6 axes bed 12 servos.zip](https://github.com/DerAndere1/Marlin/files/5831852/6.axes.bed.12.servos.zip)

Daniele-Dondi, 2021-02-18:
@DerAndere1 Hi, I'm thinking to publish some stuff on my robot. How can address you and/or your work as a citation in the paper?

DerAndere1, 2021-02-18:
@Daniele-Dondi : Thanks for your feedback and good work! I am Christian Bartz ("DerAndere"), E-Mail: ch-bartz[at]web.de (replace the [at] with @ ) . I did not publish any peer-reviewed article on the subject, but you can say that you used my firmware Marlin2ForPipetBot (https://github.com/DerAndere1/Marlin) which was developed as part of my PipetBot-A8 project. You could cite the PipetBot-A8 project page. The relevant information is 

Author First Name: Christian
Author Last Name: Bartz
Author Pseudonym: DerAndere
Last Modified Date: 5. Januar 2021
Website title: DerAndereOnline
Sub-page title: DIY lab robot Pipetbot-A8
URL: https://derandere.gitlab.io/pipetbot-a8/ 

If your journal requires you to choose between real name and pseudonym, prefer to cite my real name (Christian Bartz).
In [MLA citation style](https://owl.purdue.edu/owl/research_and_citation/mla_style/mla_formatting_and_style_guide/mla_works_cited_electronic_sources.html ), citation would be: 

DerAndere [Bartz, Christian]. "DIY lab robot Pipetbot-A8". _DerAndereOnline_, 5. Jan. 2021, https://derandere.gitlab.io/pipetbot-a8/. Accessed 18 February 2021. 

In case you publish your firmware, it is advisable to fork the repository on github (DIY lab robot Pipetbot-A8) and commit/push your modifications to your fork so that the git history remains intact and allows tracking of contributions. You can add a copyright statement in the following form to the beginning of modified source code files (do not delete any existing copyright statements):

```
/**
 * [Placeholder for your institute's copyright statement. Replace this line]
 * 
 * Based on
 * Marlin2ForPipetBot [https://github.com/DerAndere1/Marlin]
 * Copyright 2019 - 2020 Christian Bartz ("DerAndere") and other Marlin2ForPipetBot authors [https://github.com/DerAndere1/Marlin]
 * 
 * Based on 
 * Marlin 3D Printer Firmware
 */
```

Daniele-Dondi, 2021-02-20:
Thank you, you are very kind. I'll follow your suggestions

DerAndere1, 2021-02-21:
@Daniele-Dondi : I added copyright statements to the modified source code files in my Marlin2ForPipetBot branch. You should be able to update your fork with one of these workflows: https://docs.github.com/en/desktop/contributing-and-collaborating-using-github-desktop/syncing-your-branch#pulling-to-your-local-branch-from-the-remote or https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/syncing-a-fork

Daniele-Dondi, 2021-02-23:
Hi, I'm trying to merge the projects with some difficulties. Since I'm new I don't understood yet the netiquette here. But I'm asking also another thing: I created a python program to control my robot. Do you think it's better to include the code in my branch or I've to create a new project?

DerAndere1, 2021-02-23:
If you have difficulties with the merging. git is a powerful but can be daunting at times. Klick on the hyperlink below to see the diff of my recent change. It might be faster for you to just have a look at the changes and copy into your code manually whatever you like about it:
Commit https://github.com/DerAndere1/Marlin/commit/361c3169d02e1f637f2c861bb601432f362e175a did not change any functionality, I just tried to make it easier to  distinguish between my fork and upstream Marlin Firmware, which was initially not my focus.
Regarding the control software, I have no experience how publications about software usually proceed. First make sure what sequence of steps the journal expects. You may have to keep your software in a private repository an maybe give the editors and reviewers exclusive access to the code so the publication of the paper and the public release of the software can be coordinated. Google will tell you what to do. I would prefer a seperate repository in any case. If the journal gives you a choice, pick a good License and make your choice transparent, e.g. by adding a LICENSE file to the root of your control software repository as the first/second commit. I usually prefer the Apache-2.0 License in combination with a NOTICE file.

closed 2021-08-30 by DerAndere1

thebjtfellow, 2024-12-27:
I see this is old, but I thought I'd describe how I got around it. I too am using DerAndere's pipette-bot, but for a binder jetting machine. I needed to enable the heater channels for things like an overhead heater, a curing stand, a binder warmer, etc. I also am using all of the axes available, and have nothing left to assign to extruders. And I needed all four channels of heaters, so spoofing was just sort of out.

The board is the BTT Octopus Pro V1.1. 8 Axis, 4 Hotend channels, Heated Bed channel

In configuration.h, I set the TEMP_SENSOR_0 - 3 options, and the TEMP_SENSOR_BED option.
I added #define HOTENDS 4 after the #define EXTRUDERS 0
Comment out PREVENT_COLD_EXTRUDE, PREVENT_LENGTH_EXTRUDE, it makes it a little easier

At the top of Conditionals-1-axis.h, I inserted #if HOTENDS<1 ... #endif from lines 55-65 so the temp sensors wouldn't be erased

At line 175, I added #ifndef HOTENDS ... #endif at line 185 so the # is set by the configuration.h declaration

Then in planner.cpp. at 1385, I inserted #if EXTRUDERS>0 ... #endif at 1388 so it doesn't make as .se value for the move block

FInally, in gcode.cpp, at line 128 I changed it to: if( (e < EXTRUDERS) || (e < HOTENDS) ) return e;

Those changes let me use all of my hotend controllers while having no extruders defined. I hope that helps somebody, but I doubt this is a common need.

Daniele-Dondi, 2024-12-27:
For me it is still interesting. Are you able to post your modified files in
a zip folder?

DerAndere1, 2024-12-29:
I added the necessary changes to [https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot)](https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot). In the future I will prepare a pull request to add those changes to MarlinFirmware/Marlin also. Thanks @thebjtfellow