DerAndere1/Marlin2ForPipetBot #45: Limit switch problem

yunxiang521, 2021-04-15:
i. Whether J and K can be set independently and whether limit switches are needed is very inconvenient sometimes if they have to be used

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2021-04-16:
Is the process of configuring limit switches for axes I, J and K different from  configuring limit switches for XYZ ?

yunxiang521, 2021-04-16:
Yes, I, J limited switch function is not required, I and j are used for functions like e-axis.

DerAndere1, 2021-05-19:
I want I, J, K to behave exactly like X,Y,Z. If this is currently the case, your request won't get adressed. If axes  I,J,K currently behave different from X,Y,Z, please report.

closed 2021-06-03 by DerAndere1