DerAndere1/Marlin2ForPipetBot #46: Customizing for I and J to use E0 and E1. Robotic arm project.

Cliffhang3r, 2021-04-18:
Hi, I have been trying to configure marlin for my robotic arm project but keep getting stuck on adding homing the final two axis on the 5 axis robot. Then I found your modded version, and it seems to be what I need. I looked through everything I could find of what you have posted and cant get any of it (even the most up to date version) to work for me.

I need to use the two extruder driver slots to run the 4th and 5th axis motors. I am using ramps 1.4. After modifying and building, I got the following errors:

```
Marlin\src\pins/sensitive_pins.h:160:19: error: 'I_STEP_PIN' was not declared in this scope
Marlin\src\pins/sensitive_pins.h:160:31: error: 'I_DIR_PIN' was not declared in this scope
Marlin\src\pins/sensitive_pins.h:160:42: error: 'I_ENABLE_PIN' was not declared in this scope
```
and the same errors for "J".
I specified 5 non extruders, to try to tell it that I wanted to use I and J instead of E0 and E1.

Basically, I want to use I and J as E0 and E1. Can I just substitute all of the e0 pins for I and E1 for J, or how would you define the pins as required?

I can use the two E axis just fine right now using  t0 and t1, but I need to be able to home them. Any suggestions for how to modify the code? I'm pretty good at digging deep into marlin by now, but some of this is a little above me. This will be an open source project that I will post in a few weeks for anyone to use, but I need to get the code right first.

Photo of the robotic arm here:
[](https://photos.google.com/share/AF1QipN7WLjTOSKIV5JZOX2RPzMeGEj-_Eg7QLZPvST-krkJmBXiFYLtGRSrx26LW2dZnA/photo/AF1QipPDmPXj-uZmWzHB3JXrR57uKqLkeQkKRCTJGgaY?key=UzFOQlJvdUYxQmVRZnQ2N3dBREg5SjJaRnJiY2R3)

Cliffhang3r, 2021-04-18:
I have also tried your version named "Marlin 2.0.x_E_homing Branch" at https://github.com/DerAndere1/Marlin/tree/2.0.x_E_homing.

After modifying and building I get the following errors:
`Marlin\src\module\motion.cpp:1717:36: error: 'E_MIN_PIN' was not declared in this scope
Marlin\src\module\motion.cpp:1717:36: error: 'E_MAX_PIN' was not declared in this scope
Marlin\src\module\../inc/../../Configuration_adv.h:604:46: error: too many initializers for 'const xyz_float_t {aka const XYZval<float>}'`

I figured this version might be least modified for your pipet bot, but after getting the above errors, I cant get rid of them. I think this one might work but I need help with these errors. Where or how do I declare the e min and e max pins? Using ramps 1.4. 

My assumption would be that I wire the two limit switches together and use T1 and T0 to switch between extruders and use G28 E to home each in turn. Is that correct?

Thanks in advance.

Cliffhang3r, 2021-04-18:
I thought that I'd update this and mention that I also tried the most recently updated version PR1 at https://github.com/DerAndere1/Marlin/tree/6axis_PR1.

I got it modified and got the following errors, just like my first example, which used the main files at https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot.

Errors:
`
Marlin\src\pins/sensitive_pins.h:160:19: error: 'I_STEP_PIN' was not declared in this scope
Marlin\src\pins/sensitive_pins.h:160:31: error: 'I_DIR_PIN' was not declared in this scope
Marlin\src\pins/sensitive_pins.h:160:42: error: 'I_ENABLE_PIN' was not declared in this scope
Marlin\src\pins/sensitive_pins.h:201:19: error: 'J_STEP_PIN' was not declared in this scope
Marlin\src\pins/sensitive_pins.h:201:31: error: 'J_DIR_PIN' was not declared in this scope`

So I think, to do what I want it to do, I need to define I and J as the original E0 and E1 pins, does that sound right? Any suggestions?

FlorianFritz, 2021-04-20:
Hi Clifford, I ran into the same problem (on the 6axis_PR1 branch). 

Besides the I had set the I/J_*_PINS in Marlin/src/pins/ramps/pins_RAMPS.h (Change all E0 to I and all E1 to J) but even if you set EXTRUDERS to 0 some files will request the E_*_PINS to be defined. I somewhere read, that you can set them all to a free/unused PIN and they may share the same PIN. I assume Marlin will react on all G-code commands which address the E axis and switch this PIN, since it is mentioned in all G-code outputs (like M92, M114, M201, M203, etc.). So you basically have a 6 axis (X, Y, Z, A, B, E) machine with 5 motors connected. Therefore the arrays DEFAULT_AXIS_STEPS_PER_UNIT, DEFAULT_MAX_FEEDRATE, DEFAULT_MAX_ACCELERATION in Configuration.h must contain 6 numbers.

Cliffhang3r, 2021-04-20:
Thanks so much, that worked and allowed me to compile. I also had to add the following code to language.en:
` PROGMEM Language_Str MSG_AUTO_HOME_I                     = _UxGT("Home I");
  PROGMEM Language_Str MSG_AUTO_HOME_J                     = _UxGT("Home J");`

Now the two  axis work, I have to use A and B to move and not I or J. G1 A10 moves axis 4 and G1 B10 moves axis 5. Is that normally how its supposed to work?

The Homing function is still not working for those two axis. G28 I just homes XYZ, same with G28 J. I think maybe its a pins or endstop thing maybe. I cant find where the I_MAX and J_MAX pins are defined. I expected an error but none came. Do I just add the lines to Pins.h or am I missing something?

Cliffhang3r, 2021-04-20:
Ok so I added the endstops to Pins_ramps.h and no errors. I used pins from the endstops that weren't defined/used. This is what I added:
`#ifndef I_STOP_PIN
  #ifndef I_MIN_PIN
    #define I_MIN_PIN                         18
  #endif
  #ifndef I_MAX_PIN
    #define I_MAX_PIN                         19
  #endif
#endif

#ifndef J_STOP_PIN
  #ifndef J_MIN_PIN
    #define J_MIN_PIN                         18
  #endif
  #ifndef J_MAX_PIN
    #define J_MAX_PIN                         3
  #endif
#endif`

But it still wont home the last two axis.

M119 shows that it appears to be reading the pin states correctly:
`09:27:10.527 : Reporting endstop status
09:27:10.527 : x_max: open
09:27:10.528 : y_max: open
09:27:10.532 : z_min: open
09:27:10.532 : a_max: open
09:27:10.532 : b_max: open
`
If I manually toggle one of the endstops and resend M119, it shows that it was read correctly. G28 A and G28 B don't do anything, and G28 I and G28 J just home XYZ.

What am I missing?
Thanks so much for your help, sorry if I am asking too much. This is a project for school so I really would like to get this figured out before I have to present in a week.

Cliffhang3r, 2021-04-20:
Ok, one more update.

The motion controls on the lcd don't work for I and J. Axis I might be moving, but its too slow to see the change, and doesn't appear to change J.

As for the homing issue, looking in menu_motion.h, it looks like it is is trying to use the letters I and J instead of A and B:

```
  //
  // Auto Home
  //
  GCODES_ITEM(MSG_AUTO_HOME, G28_STR);
  #if ENABLED(INDIVIDUAL_AXIS_HOMING_MENU)
    GCODES_ITEM(MSG_AUTO_HOME_X, PSTR("G28X"));
    GCODES_ITEM(MSG_AUTO_HOME_Y, PSTR("G28Y"));
    GCODES_ITEM(MSG_AUTO_HOME_Z, PSTR("G28Z"));
    #if LINEAR_AXES >= 4
      GCODES_ITEM(MSG_AUTO_HOME_I, PSTR("G28I"));
    #endif
    #if LINEAR_AXES >= 5
      GCODES_ITEM(MSG_AUTO_HOME_J, PSTR("G28J"));
    #endif
    #if LINEAR_AXES >= 6
      GCODES_ITEM(MSG_AUTO_HOME_K, PSTR("G28K"));
    #endif
  #endif

  //
```

It seems like the code is trying to use I and J, and from my experimentation, it looks like only A and B move the machine. As I said above, though, G28 A and G28 B don't do anything, so I'm not sure if it should be changed.

It looks like there is confusion in the code between using I/J and A/B. Any suggestions?

Cliffhang3r, 2021-04-25:
Ok so I renamed the axis I and J in the first bit of Customize.h. Not sure why it was named A and B there. Also fixed the sanity check so it wouldn't freak out. Now G1 I and G1 J work for movement commands and the endstop check looks like this:
```
10:25:06.281 : Reporting endstop status
10:25:06.281 : x_max: open
10:25:06.281 : y_max: open
10:25:06.281 : z_min: open
10:25:06.284 : i_max: open
10:25:06.284 : j_max: open
```
Go now it is starting to look right but homing I and J, and motion via the LCD menu still doesn't work. Ill keep looking but any suggestions?

FlorianFritz, 2021-04-29:
Hi Clifford, 

I think naming the additional axis internally 'X', 'Y', 'Z', 'I', 'J', 'K' is misleading, they should be named by number 1 to 6.

Externally (in G-Code), from what I know, you should use 'A', 'B', 'C' or 'U', 'V', 'W' for the naming of the axis 4 to 6 because 
1) it is a standard or at least a common convention in robotics. A-B-C is mostly used for rotational axis and U-V-W for a second set of linear/cartesian axis.
2) the g-code interpreter should not get messed up - you might run into problems if (at least one of) your axis names conflicts with another command (or part of it). Sticking with  'A', 'B', 'C' / 'U', 'V', 'W' assures that it will not lead to conflicts (assuming correct implementation)

Yes, the modification of language_en.h was necessary for me too when using the LCD. There are three lines missing like I mentioned in my comment on the [pull request](https://github.com/MarlinFirmware/Marlin/pull/19112):
```
PROGMEM Language_Str MSG_AUTO_HOME_I = _UxGT("Home ") AXIS4_STR;
PROGMEM Language_Str MSG_AUTO_HOME_J = _UxGT("Home ") AXIS5_STR;
PROGMEM Language_Str MSG_AUTO_HOME_K = _UxGT("Home ") AXIS6_STR;
```
Adding `AXIS(4|5|6)_STR` at the end of the MSG strings will give you the axis names defined in Configuration.h (`AXIS(4|5|6)_NAME`) on the LCD display.  

PS: I cloud not test the homeing of the additional axis and your modifications to the menu_motion.h, because my hardware is still not complete.

DerAndere1, 2021-05-14:
> Yes, the modification of language_en.h was necessary for me too when using the LCD. There are three lines missing like I mentioned in my comment on the pull request:
>
> PROGMEM Language_Str [...]

Thanks for the patch. I will look at this. Your explanation of user-facing axis names (XYZUVW or XYZABC) vs internal axis names (XYZIJK) is correct. I agree that internal axis names should change to AXIS1 ... AXIS6 in the long run, but original Marlin started with XYZ, so I used IJK to have a consistant naming scheme. regardless of internal axis names, I recommend standard-conforming axis names (U/V/W for linear axes or A/B/C for rotational axes). Your config
#if LINEAR_AXES >= 4
  #define AXIS4_NAME 'I' 
#endif
#if LINEAR_AXES >= 5
  #define AXIS5_NAME 'J' 
#endif

indeed results in conflicts, so you cannot use G-commands commands like G2 (arc movement).
I recommend to use the following config:

```
#if LINEAR_AXES >= 4
  #define AXIS4_NAME 'A' // :['A', 'B', 'C', 'U', 'V', 'W']
#endif
#if LINEAR_AXES >= 5
  #define AXIS5_NAME 'B' // :['A', 'B', 'C', 'U', 'V', 'W']
#endif
```` 

Then you can use standard-conforming G-code like
```
G28 A; home 4th axis
G28 B; home 5th axis
G1 A10; move 4th axis (10 mm)
G1 B20; move 5th axis (10 mm) 
```

DerAndere1, 2021-05-17:
> The motion controls on the lcd don't work for I and J. Axis I might be moving, but its too slow to see the change, and doesn't appear to change J.
>
> As for the homing issue, looking in menu_motion.h, [...]

This is indeed a bug. For people that have in their configuration.h 
```
#define AXIS4_NAME 'A'
#define AXIS5_NAME 'B'
```
currently manually have to change  
```
GCODES_ITEM(MSG_AUTO_HOME_I, PSTR("G28I"));
```
and 
GCODES_ITEM(MSG_AUTO_HOME_J, PSTR("G28J"));
```
to
```
GCODES_ITEM(MSG_AUTO_HOME_I, PSTR("G28A"));
``` 
and
``` 
GCODES_ITEM(MSG_AUTO_HOME_J, PSTR("G28B"));
```
should work in your case. So if I understand correctly, your main  problem remains: that G28 A and G28 B do nothing, even when sent as G-code via USB or from SD-card. What happens if you send `G28` without any axis name? If nothing happens, then I suspect a wrong config of endstops or a regression

>

DerAndere1, 2021-05-25:
The current 6axis_PR1 is probably broken again but I provide a link to the 6axis_PR1 branch at the time of commit https://github.com/DerAndere1/Marlin/tree/2c232728ae4e80d34c7a88695f38248a8c2d0dc8 where G28 was supposed to work

closed 2021-05-25 by DerAndere1