DerAndere1/Marlin2ForPipetBot #47: utility/spi_com.h   Not found on PR1 branch

paloky, 2021-05-13:
Hi,
I download the PR1 brach, and I don't be able to compile.

The compiler don't find the "utitily/spi_com.h" file.
Where can I find this file?
```
In file included from Marlin\src\HAL\STM32\MarlinSPI.cpp:24:
Marlin\src\HAL\STM32\MarlinSPI.h:28:12: fatal error: utility/spi_com.h: No such file or directory
   28 |   #include <utility/spi_com.h>
      |            ^~~~~~~~~~~~~~~~~~~
```

For other hand, I need to change in platformio.io the board "BigTree_GTR_v1" for the "marlin_BigTree_GTR_v1" because platformio doen't recognize the board.
```
#
# Bigtreetech GTR V1.0 (STM32F407IGT6 ARM Cortex-M4)
#
[env:BIGTREE_GTR_V1_0]
platform          = ststm32@>=5.7.0,<6.2.0
extends           = common_stm32
board             = marlin_BigTree_GTR_v1    #BigTree_GTR_v1
extra_scripts     = ${common.extra_scripts}
  pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags       = ${common_stm32.build_flags}
  -DUSB_PRODUCT=\"STM32F407IG\"
  -DTARGET_STM32F4 -DSTM32F407IX -DVECT_TAB_OFFSET=0x8000
```

My goal is use this repositori to use with a BigTreeth_GTR_V1 + M5 board (Up to 11 axis), but at moment I only need 8 axis.
It's is possible expand up to 8 axis ??

Thank's a lot.

DerAndere1, 2021-05-14:
I will have a look how to fix the missing spi_com.h. When upstream Marlin allows to use, BigTreeth_GTR_V1 + M5, there is a chance that PR1 could work. At the moment only 6 axes are possible. Extension to 9 axes is high on the priority list, But it is a lot of work, I will try to start extending to 9 axes soon. If you have experience with c++, you will certainly be quicker then I.  Have a look at  commit https://github.com/DerAndere1/Marlin/commit/363f654f988287b420eea67626280080b60ad597 There you see how axes IJK were added. You can do the same for AXIS7, AXIS8, and AXIS9. as you can see, it is a lot of work, though

DerAndere1, 2021-05-15:
Have you verified that spi_com.h is found in current upstream Marlin bugfix-2.x? Because that file is part of the Arduino core library. If you configured everything correctly for your board, it should compile with the correct core library. the file comes from the STM32 core (https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/SPI/src/utility ). Please verify, that your board uses the STM32 core.

paloky, 2021-05-17:
Hi DerAndere1.
Finally, I am working on the "6axis-dev" branch.

This weekend, I have successfully managed to add a new Axis.  
I have tested it with the DRV8825 and with the TMC2209 in UART Mode.
I have found some bugs in this branch, for the use of the TMC2209 in UART Mode on axis "I, J and K", that I have managed to fix.
For the time being, I have internally numbered it as "M".

I will finish implementing the rest of axes, to get all the BTT GTR V1+M5 (11 axis in total).

Since I'm going to add the new axes, I'm going to change the actual internal name (I, J and K) to "A, B, C" and continue with the rest "U, V, W".

In total, we will get (X, Y, Z, A, B, C, U, V, W, E, ?? ) Axis.
Any suggestions for the name of axis number 11?  "D" maybe?

Thank's.
Best Regards.
Paloky.

DerAndere1, 2021-05-17:
The recommended branch depends on where you want your code to land. Historically I used the 6axis_dev branch to develop the PR branch. Some time after I had opened pull request 19112 at https://github.com/MarlinFirmware/Marlin to get the 6axis support merged into upstream MarlinFirmware/Marlin, I stopped synchronizing with the PR1 branch. Currently, the 6axis_PR1 branch gets often updated by MarlinFirmware developer thinkyhead to prepare the merge. If you want to get your changes merged into upstream Marlin, I recommend you to prepare a pull request targeted at the PR1 branch and fx all issues there. 

Because upstream Marlin development is so fast, it is very difficult to get a working snapshot from the PR1 branch. so if the 6axis_dev branch works for you and you only want a working firmware for yourself, just go ahead with the 6axis_dev branch and dont hesitate to share a link to your fork.  

In any case, you should be aware, that thinkyhead made a seperate pull request (21932 at https://github.com/MarlinFirmware/Marlin) that prepares for addition of more than 8 axes. The problem is that for more than 8 axes (or 7 + extruder ?), the size of some 8bit variables is not enough to store data for the 8 axes. Axis Name D sounds fine. If you want your code to be useful for others, you should make user-facing axis names (those used in G-code) configurable.

DerAndere1, 2021-05-26:
I cherry-picked your 9 axis code and I am now updating it to latest state of multi-axis Marlin. I made sure that the git history including your authorship will remain intact. As soon as I publish the updated code, I will mention your contribution in my documentation, too. Thanks for doing the hard work.

paloky, 2021-05-26:
Hi DerAndere1.
Thank you.
I haven't had time to do the PR yet.
I have put a lot of "/**SG**/" comments, in all the lines I have modified. You can erase this comments...

I'm not been able to implement the 2 missing axes, for my BTT GTR1.0 + M5 Board (11 axis total)
I think it's a problem with the arrays that Marlin uses and I had compilation problems.

Best Regars.

DerAndere1, 2021-05-26:
No worries, I was just too curious to test your code. I finished rebasing it onto the 6axis_PR1 branch and will test-compile after some clean-up. current 6axis_PR1 branch is probably already prepared for more than 8 non-extruder axes plus extruder. We will see during hardware testing.

DerAndere1, 2021-06-03:
I close this as fixed. My branch [9axis_pull](https://github.com/DerAndere1/Marlin/tree/9axis_pull) now compiles, including utility/spi_com.h using PlatformIO for your board. Configs for 7 axes from JanHendrik-5D can be compiled. The configs can be found as part of the attachment of the following post: 
https://github.com/DerAndere1/Marlin/issues/44#issuecomment-850885727

closed 2021-06-03 by DerAndere1