DerAndere1/Marlin2ForPipetBot #50: The MKS_SGEN_L motherboard is stuck on the startup screen under the 2.0.9.1 firmware.

gaifeng8864, 2021-07-08:
The MKS_SGEN_L motherboard is stuck on the startup screen under the 2.0.9.1 firmware.

I use MKS_MINI12864 screen, TMC2209 driver, enable SENSORLESS_HOMING, the motherboard is used for 3D printer.
When I used VScode to compile the latest 2.0.9.1 firmware for my MKS_SGEN_L motherboard, there was no error in the compilation process. When I put the compiled bin file on the micSD card for upgrade, my motherboard stuck on the startup page and cannot enter the system. The indicator lights D1 to D4 are always on, and the buzzer always sounds. My key configuration parameter settings are the same as those on the 2.0.8.1 firmware that was used normally before.The attachment is the configuration file I used when compiling.
[Marlin.zip](https://github.com/DerAndere1/Marlin/files/6783661/Marlin.zip)

gaifeng8864, 2021-07-08:
I used the same key parameter configuration and tried to compile different versions. I found that as long as the version number is higher than or equal to 2.0.9, the firmware I compiled cannot start normally. I guess it may be caused by the difference between some key parameters and 2.0.8, but I tried a lot of combinations, and I still haven't found out what caused this problem.

DerAndere1, 2021-08-16:
You have reported in the issue tracker of my Marlin fork. was that intentional? Try again with latest upstream MarlinFirmware/Marlin bugfix-2.0.x from https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x and configs from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x

closed 2021-08-30 by DerAndere1