DerAndere1/Marlin2ForPipetBot #52: Small error in motion_motion.cpp regarding axis names

hobiseven, 2021-07-30:
Hi,

great that you managed to get an official merge in Marlin for 6 axis support. 

I took the core there and checked / redid the compilation for 6 axis, and there is a tiny error in one of the macros / includes definition  👍 

Compiling .pio\build\alfawise_U30\src\src\lcd\menu\menu_tune.cpp.o
In file included from Marlin\src\lcd\menu\menu_motion.cpp:33:0:
Marlin\src\lcd\menu\menu_motion.cpp: In function 'void menu_motion()':
Marlin\src\lcd\menu\menu_motion.cpp:363:47: error: expected ')' before 'I_STR'
       GCODES_ITEM(MSG_AUTO_HOME_I, PSTR("G28" I_STR));
                                               ^
Marlin\src\lcd\menu\menu_item.h:263:39: note: in definition of macro '_MENU_INNER_P'
     MenuItem_##TYPE::action(plabel, ##V);                      \

The actual line of code is : 

   #if LINEAR_AXES >= 4
      GCODES_ITEM(MSG_AUTO_HOME_I, PSTR("G28" I_STR));
      //GCODES_ITEM(MSG_AUTO_HOME_I, PSTR("G28U"));
    #endif

I_STR is not properly defined, or the macro expansion does not result in the proper character appended to "G28" string... Beside that, all compiles fine, and I will test this later in August. 

Thanks again for your superb work. 

Olivier.

DerAndere1, 2021-08-01:
Thanks. Can you test if it copiles if you use the following code menu_motion.cpp line 363 ?
```
GCODES_ITEM(MSG_AUTO_HOME_I, PSTR("G28" AXIS4_STR));
```

hobiseven, 2021-08-04:
Hi,

Of course this works.

Thank you.

closed 2021-08-04 by hobiseven