DerAndere1/Marlin2ForPipetBot #54: I axis only works in absolute positioning

alana12345, 2021-08-24:
Hello,

I am new to Marlin and having an issue with the I axis (4th axis). The stepper motor only runs in absolute coordinates. For example, even if I send a G91 command, the I axis will treat any following commands as absolute coordinates rather than relative. This is not true of the X,Y, and Z axes--they all work perfectly. Any help would be greatly appreciated! Thank you. 

### Steps to Reproduce

1. Send G91 command
2. Send G1 I10 command (or other command)
3. I axis treats command as absolute positioning rather than relative

**Expected behavior:** 
I axis moves in relative when G91 command is given preceding G1 commands 

**Actual behavior:** 
I axis moves in absolute positioning always

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin-Marlin2ForPipetBot.zip](https://github.com/DerAndere1/Marlin/files/7040660/Marlin-Marlin2ForPipetBot.zip)

DerAndere1, 2021-08-30:
I released a new version of Marlin2ForPipetBot: https://github.com/DerAndere1/Marlin/releases/tag/Marlin2ForPipetBot-2.0.9.1 .
It is up to date with branch https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot as of today and I have no problems with relative/ absolute mode (G90 or G91).

alana12345, 2021-08-31:
Great, will test it out soon. Thank you!

DerAndere1, 2021-09-01:
Sorry, this isprobably still not fixed for configurations with `EXTRUDERS 2` or higher.  See upstream issue https://github.com/MarlinFirmware/Marlin/issues/22681 .

DerAndere1, 2021-09-24:
This should now be fixed in https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

alana12345, 2021-09-26:
That works, thank you so much!

closed 2021-11-28 by DerAndere1