DerAndere1/Marlin2ForPipetBot #55: Marlin-9axis_pull  hardware test report Axes 8 and 9

HendrikJan-5D, 2021-08-25:
To support rebasing 9axis_pull branch onto current upstream MarlinFirmware/Marlin/bugfix-2.0.x

Have I configured Marlin-9axis_pull for 8 axes
And tested with a Dummy hardware shaft.
All functions work.

I configured Marlin-9axis_pull for 9 axes.
I am getting compiler errors

Marlin > src > feature > tmc_util.cpp 

445 #if AXIS_IS_TMC(P)
446   if (monitor_tmc_driver(stepperP, need_update_error_counters, need_debug_reporting))
447     step_current_down(stepperPM); <-------------> must be (stepperP )
448 #endif

Marlin > src > core > macros.h

LIST_INC_INC_16 was not declared in this scope [332,25]

Marlin > src > core > serial.cpp

identifier q is undefined  C/C++ (20) [117,3]
q was not declared in this scoop [118, 160]

Marlin > src > gcode > config > M92.ccp

Q_AXIS was not declared in this scope

It seems if you choose 9 axes , ( P ) the 10th axis is also chosen ( Q )
Would you take a look at this, I can't figure it out.

Send the complete configuration to you.

[Marlin-9axis_pull 9 Axis test.zip](https://github.com/DerAndere1/Marlin/files/7044654/Marlin-9axis_pull.9.Axis.test.zip)

HendrikJan-5D, 2021-08-28:
Your latest update did not resolve the issue.

G1 W20    + ------>  20 mm
G1 W-20   +------>  40 mm  totaal +----> 60 mm

The dir function of the TMC2209   P axis  is not controlled

When compiling I regularly see a problem message.
Which has to do with the P axis.

[image]

DerAndere1, 2021-08-28:
Memo to myself: I have to change uint8_t to linear_axis_bits_t in the following lines: 
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/stepper.h#L291
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/stepper.h#L545
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/gcode/gcode.h#L329
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/planner.cpp#L2333
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/planner.cpp#L2337
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/stepper.cpp#L152
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/stepper.cpp#L1662
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/stepper.h#L291
https://github.com/DerAndere1/Marlin/blob/2b18e920727343de8b49906a556763e121aeb889/Marlin/src/module/stepper.h#L545

and make sure the following macro from motion.h L425 is moved to types.h or elsewhere so that it is available in all files listed above:
```
typedef IF<(LINEAR_AXES > 8), uint16_t, uint8_t>::type linear_axis_bits_t;
```

DerAndere1, 2021-08-28:
above changes are now in 9axis_pull branch. Testing is much appreciated

HendrikJan-5D, 2021-08-28:
I downloaded your latest version.
And think you made a version mistake somewhere.
I run into compiler errors that have already been fixed in an earlier version. ( 3 days ago )
And now reappear.

[image]

DerAndere1, 2021-08-28:
yes my git repository was in a strange state. should be fixed now including an additional correction regarding relative/absolute mode

DerAndere1, 2021-08-28:
If the current solution is problematic with 8 or 9 axes, we will have to use `axis_bits_t` instead of `linear_axis_bits_t`in the above places.

HendrikJan-5D, 2021-08-28:
In the first configured version 3 days ago, the 8th axis worked without any problems.
Only the 9th axis had problems.

HendrikJan-5D, 2021-08-29:
Latest version downloaded

My pin files and configs copied over without changes.

The dir problem with the 9th axis is not solved yet.

The other 8 Axis work well

G1 X100 Y100 Z100 A90 B45 C45 U90 V90

I send you the current configuration

[Marlin-9axis_pull 9 Axis test 5.zip](https://github.com/DerAndere1/Marlin/files/7072079/Marlin-9axis_pull.9.Axis.test.5.zip)

HendrikJan-5D, 2021-08-29:
When looking for a solution to the ( dir ) problem of the P Axis
I came across the following things.

Marlin > src > core > types.h

376 template<typename T>
377 struct XYZval {
378   union {
379    struct { T LINEAR_AXIS_LIST(x, y, z, i, j, k, m, o, p, q); };
380    struct { T LINEAR_AXIS_LIST(a, b, c, u, v, w, ax7, ax8, a9, ax10); };
381    T pos[LINEAR_AXES];
382  };

Line 380 ----> a9 this must be ax9

Unfortunately this doesn't solve the ( dir ) problem of the P Axis.

I also came across a few things while searching that I don't know if it's wrong.

Marlin > src > inc > Conditionals_post.h

186 #if LINEAR_AXES >= 9
187   #define _P_HALF_PMAX ((P_BED_SIZE) / 2)
188 #endif

I'm missing the code for the 10th Axis .
While he is present at line 213

Marlin > src > feature > stepper_driver_safety.cpp

In this file I see everything from the X Y and Z Axis plus extruders E
I expect there should also be a code for I J K M O P Q Axes

DerAndere1, 2021-08-30:
I forgot to change the type of one occurance of `uint8_t dm` and did the changes proposed by you. To understand why it is so much more difficult to get 9 axes working compared to 8: to safe space. Marlin uses variables of type uint8_t whenever possible which is a variable with 8 bits, so up to 8 axes are possible with that code. For more axes, I have to change to a variable type with more than 8 bits. Changing all occurances of u8int_t would increase code size too much, so I have to find the right spots. I now use axis_bits_t in all places I listed so far, which hopefully it resolves the issue. Thanks for being patient with me.
Things that have to be tested for 8 axes and seperately for 9 axes: 
- change of direction
- change between absolute mode and relative mode (G90, G91)

HendrikJan-5D, 2021-08-30:
Just a quick feedback.
The 9th Axis works.
Come back to it later, test more first
We've fixed it.

DerAndere1, 2021-09-01:
This is great. If you like, you can prepare a list of Gcodes you need for printing and one list for G--codes you tested succssfullly. 

Alternatively, you could attach an example gcode file (preferrrably one with many different commands) that you tested, and report the Start-Gcode and End-Gcode (if any) you are using. That would be equally helpful - Would be nice to see where we stand.

HendrikJan-5D, 2021-09-01:
I did the tests you asked for.
See test report.
Can I do other tests ?
Otherwise I'm going to configure the 10th axis and test it out.

DerAnder1
I would like to get in touch with you as a member of Hcc!3d outside of github.
Is this possible

[Marlin-9axis_pull   9 Axis test report.txt](https://github.com/DerAndere1/Marlin/files/7089515/Marlin-9axis_pull.9.Axis.test.report.txt)

HendrikJan-5D, 2021-09-04:
I configured the 10th Axis.    (The 10th Axis works after corrections )
See attached test report.
And configured files
 
Question

For the cooling of drivers on the extension board M5.
I also want to control a second fan at the same time ---> FAN3_PIN

Configuration_adv.h

466 --> #define CONTROLLER_FAN_PIN FAN2_PIN   // Set a custom pin for the controller fan

Is that possible with the current configuration ?

[Test report Axis 10.txt](https://github.com/DerAndere1/Marlin/files/7109383/Test.report.Axis.10.txt)
[Marlin-9axis_pull 10 Axis test 1.zip](https://github.com/DerAndere1/Marlin/files/7109384/Marlin-9axis_pull.10.Axis.test.1.zip)
[image]

DerAndere1, 2021-09-04:
Thanks for the corrections. One of the more challenging things to get working correctly with a 5+ DoF machine is probably bed leveling. For your setup, we could change the  AUTO_BED_LEVELING_3POINT code so that it automatically adjusts the rotational axes of the table. Then the bed can be assumed to be level after a G29. Otherwise (or with mesh-based bed leveling procedures) you would probably have to implement inverse kinematics for your machine within Marlin.

DerAndere1, 2021-09-04:
I will contact Hcc!3d via their website and see if we can get in contact as soon as  I have time.

HendrikJan-5D, 2021-09-05:
((The AUTO_BED_LEVELING_3POINT code so that it automatically adjusts the rotational axes of the table.))

That would be nice .
Because I now have to make sure that all axis are in the 0 position before I apply G29

1193--> #define NOZZLE_TO_PROBE_OFFSET { 10,-20, -2.0, 0, 0, 0, 0, 0, 0, 0 }

In this line you must be able to indicate how all axes are in the G29 function.
That doesn't work yet.

Before I bring my printer back to 7 Axes I send you a video of the test setup for 10 Axes

DerAndere1, 2021-09-05:
> 1193--> #define NOZZLE_TO_PROBE_OFFSET { 10,-20, -2.0, 0, 0, 0, 0, 0, 0, 0 }
> In this line you must be able to indicate how all axes are in the G29 function. 

Not so sure about that. I think you allways should home all axes before sending G29 and in that case, only offsets in X, Y and Z are meaningful. NOZZLE_TO_PROBE_OFFSET should become distances in the cartesian 3D space, so only X, Y and Z element is meaningful. The additional elements of value 0 in that array are only a workaround that is needed until Marlin has a dedicated variable type for 3D coordinates.
To make the firmware respect those 3D offsets with more than 3 axes, you need inverse kinematics. For task of inverse kinematics is exactly what you describe: translating those 3D coordinates into the required joint positions according the kinematic parameters (dimensions of your specific machine) and that is outside of the scope of my work.

Quote from the [wiki](https://github.com/DerAndere1/Marlin/wiki):

> If you have a different machine where additional axes contribute to tool positioning, you currently have to implement your own kinematic model or develop a post-processor that translates desired tool positions to compatible G-code.

What I meant with my proposal was to eliminate the need for bed leveling compensation in software during the print by doing it in hardware once during the G29 procedure. Here is some pseudo-code for that implementation of G29:
- probe 3 points
- Get probing data
- Calculate the bed tilt angle in YZ plane (for A axis) and bed tilt angle in XZ plane (for B axis) using linear algebra:
```
  vector_3 planeNormal = vector_3::cross(points[0] - points[1], points[2] - points[1]).get_normal(); 
  matrix_3x3 bed_level_matrix;
  bed_level_matrix = matrix_3x3::create_look_at(planeNormal);
```
(inspired from
https://github.com/DerAndere1/Marlin/blob/de061ba5d1d1013ba3b118bf66c48ffc03399e8a/Marlin/src/gcode/bedlevel/abl/G29.cpp#L702
and
https://github.com/DerAndere1/Marlin/blob/de061ba5d1d1013ba3b118bf66c48ffc03399e8a/Marlin/src/gcode/bedlevel/abl/G29.cpp#L704)

- turn A axis and B axis stepper to compensate for the bed tilt in hardware 
- set A and B axis position to 0 (get inspiration from `G92` command)
- throw away bed level data and turn of software bed level correction because the bed can be assumed to be really level (get inspiration from `G29 J` command)
--> G29 finished

HendrikJan-5D, 2021-09-05:
I think I misunderstood you.
What I had in mind, that when starting the G29 code from arbitrary axes position
All axes move first to their 0 position .
And only then start the bed leveling function

HendrikJan-5D, 2021-09-09:
Have you already implemented my changes to, the 10 Axis test report, in the official version?
Then I can mechanically test it again with 10 Axis.
Before I build my printer back to 7 Axes.

HendrikJan-5D, 2021-09-15:
G29 AUTO BED LEVELING > LINEAR < also works

HendrikJan-5D, 2021-09-22:
When I download Marlin-9axis_pull as zip file.
I get Marlin-Marlin2ForPipetBot.zip

HendrikJan-5D, 2021-09-23:
Also when I download zip file
10 axis_PR1    
9axis-pull_b
bugfix-2.0.x

I get Marlin-Marlin2ForPipetBot.zip

DerAndere1, 2021-09-25:
This is how github.com works. When you go to "Code" -> "Code" -> "Download as .zip", it will always download the current state of the default branch, which is Marlin2ForPipetBot in my fork. For older releases, you can go to "Code" -> "tags".  To obtain other versions, clone the repository and use git to checkout the desired branch or commit. The easiest way to get it is with Git + MS VSCode + PlaformIO IDE extension. By the way, I brought 9 axis code up to date and opened a pull request at MarlinFirmware/Marlin . You can test the current state in branch [10axis_PR1](https://github.com/DerAndere1/Marlin/tree/10axis_PR1). We had to remove the 10th axis though, because it is non-standard. I have updated your configs to be compatible with that branch. You can find the download link in section "Configurations" at https://github.com/MarlinFirmware/Marlin/pull/22822

HendrikJan-5D, 2021-09-29:
First homen G28 and then bed leveling G29 will not always go well .
If an axis has a negative offset . ( I_MIN_POS -90 ) (I_MAX_POS 90 )
Then after homen the I axis is at -90
And must first be brought to 0 before I can bed level G29

Does it help to set the following in Configuration.h ?

#define MANUAL_I_HOME_POS -90
----------------------------------------------------------------------

I don't think you can solve it with this.
Even then, after homing, the I axis is at -90
And it must be set to 0 before you want to perform bed leveling

--------------------------------------------------------------------------------

I configured Marlin-10axis_PR1 with
9axis_PR1_configs.zip (from Support for up to LINEAR_AXES 9 )

Tested
Compilation ok
Problems 2
Hardware printer test ok

See test report
And completely configured zip file

[Test report Marlin-10axis_PR1.txt](https://github.com/DerAndere1/Marlin/files/7250145/Test.report.Marlin-10axis_PR1.txt)
[Marlin-10axis_PR1_9axis test.zip](https://github.com/DerAndere1/Marlin/files/7250147/Marlin-10axis_PR1_9axis.test.zip)

DerAndere1, 2021-09-29:
it would make sense to create a new parameter `#define SAFE_POSITIONS { 0, 0, 0, 0, 0, 0 }`  for the i,j,k, m,o,q axes and move to those positions as part of G28 (with SAFE_HOMING) and as part of G29. I will see what i can do.

HendrikJan-5D, 2021-09-30:
That would be nice.

--------------------------------------------------
I have reduced the existing test configuration to 7 axes.
For my 5D printer
But keep getting the same 2 problem messages.
Does this have anything to do with limit switch names?

DerAndere1, 2021-10-30:
In your Configuration.h, the following
```
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
```
looks wrong. the value `X_MAX_POS` (and similarly `Y_MAX_POS`) should be high enough to allow moving the tool almost to the physikal x travel limit, but not more. You have bed center at 0,0, `X_MIN_POS -173` and `#define X_BED_SIZE 220`. Suppose your X axis frame is slightly more than 300 mm long. Your x endstop switch is at position -173mm, so at max, the axis could physically go to position 133. then you should set something between `#define X_MAX_POS 110` (half the bed size) and `#define X_MAX_POS 133`

HendrikJan-5D, 2021-10-31:
[Test report Safe Probing Position.txt](https://github.com/DerAndere1/Marlin/files/7448555/Test.report.Safe.Probing.Position.txt)

DerAndere1, 2021-10-31:
You should ensure that a sheet of paper fits between tool and bed after 
```
G28
G1 X0 X0 Z0 A0 B0 C0 U0
```
If you tool jams into the bed at position 0, you have to do one of the following:
- build a bed that sits on springs so that its hight can be adjusted manually with screws
- mount your physical z min endstop switches slightly higher
An alternative method woud be:

- change X_MIN_POS to a negative value (if your limit switch sits approx. 10 mm below the bed, start with `ZMIN_POS -20`), send 
```
M502
M500
G28
G1 X0 X0 Z0 A0 B0 C0 U0
```
, then determine the exact distance between tool (lets assume you measure 9 mm) and bed then adjust ZMIN_POS accordingly (in this example, change it to `ZMIN_POS -9`).

HendrikJan-5D, 2021-10-31:
I know the alignment procedure for the bed.
And it works fine with my well-aligned bed.
But is it necessary for the z axis to go all the way to 0 before leveling?
This also does not happen with normal bed leveling .
Z after homing doesn't work for me.
Because I home Z to the max position.
Similar to a delta printer.
This is also used in CNC milling.

DerAndere1, 2021-10-31:
Yeah, from the video I should have seen that you know what you are doing ;D - Wow. 5D print :-))))))))))))) Amazing work, congrats!

Was it a manually written gcode script, or did TU Delft come up with something? When you feel like you want to share information with the public, I can help documenting your progress and/or publishing info online. I have some experience with choosing a license and so on, just ask me. 

> But is it necessary for the z axis to go all the way to 0 before leveling?
> This also does not happen with normal bed leveling .
> Because I home Z to the max position.

Z home to max is a good decision for your setup, indeed and in that case Z_AFTER_HOMING is not necessary. 
In the 10axis_PR1 branch I have now implemented a solution that should work for all bed leveling methods. The relevant change set was this commit: https://github.com/DerAndere1/Marlin/commit/d7f8d076e1e3704267f16f867030e52606438745#diff-6a6786c59c62c9ee2983ef565565fcd779791a9006d2e8f2e348572f637e5cd2

With this, you can set the following in Configuration_adv.h 
/**
 * Safe bed leveling start coordinates. If enabled, the respective axis is moved to the specified position at the beginning of the bed leveling procedure.
 * Required e.g. with LINEAR_AXES >= 4, if Z probe is not perpendicular to the bed after homing.
 * Values must be chosen so that the bed is oriented horizontally and so that the Z-probe is oriented vertically.
 */
//#define SAFE_BED_LEVELING_START_X 0.0
//#define SAFE_BED_LEVELING_START_Y 0.0
//#define SAFE_BED_LEVELING_START_Z 0.0
#define SAFE_BED_LEVELING_START_I 0.0
#define SAFE_BED_LEVELING_START_J 0.0
#define SAFE_BED_LEVELING_START_K 0.0
#define SAFE_BED_LEVELING_START_U 0.0
//#define SAFE_BED_LEVELING_START_V 0.0
//#define SAFE_BED_LEVELING_START_W 0.0

With this, only the axes for which the feature is enabled should move to the specified safe start coordinate at the beginning of G29. Depending on the needs of your machine, you may have to use other values than 0.0 for some axes.

HendrikJan-5D, 2021-10-31:
Thank you for your nice comment.
I'm making a PowerPoint presentation with YouTube film about my 7 Axis 5D printer.
For our Hcc!3D group.
I also want to use this as an introduction to companies that are developing slicer software.
With the request if I can test their software with my printer.
It was also the intention to make it public at a fair and to invite you to this.
However, due to the covid situation, it was not possible to organize this.
At the moment I don't feel like promoting it public.
Takes me too much time, which I prefer to spend on new developments.
+++
The G code script was manually manipulated.

DerAndere1, 2021-10-31:
the presentation and reaching out to slicer developers is a good idea. I was just worried that your work would be a once-only effort. But if you make some information available to your group members, that's good news. In case you have no financial interest in the subject, let them know if/how they can get access to the bill of materials or 3D models for the required parts to build the robot. Those can be a time saver for them when they want to join your efforts and push the limits of 3D printing. have a nice evening.

HendrikJan-5D, 2021-11-04:
I have completely reconfigured the latest version of Marlin-10axis_PR1.
Because my old config files are out of date.
I run into a compilation problem.
See screen copy,
Would you like to see what the cause is

[image]

[Marlin-10axis_PR1_7axisconfig 5.zip](https://github.com/DerAndere1/Marlin/files/7478514/Marlin-10axis_PR1_7axisconfig.5.zip)

DerAndere1, 2021-11-08:
One general warning: Marlin is still not officially supporting rotational axes. Thus, the speed of each stepper is calculated from the specified feedrate as if all axes were linear (I believe the method for interpreting feedrate is called "multidimensional linear interpolation"). Thus, when the tool is in contact with the workpiece and X, Y and/or Z are different from 0, you have to set a lower feedrate before moving the rotational axes to account for the radius of the workpiece. Otherwise, the actual surface speed of the tool along the workpiece surface is too high. To properly support rotational axes, I guess the feedrate has to be calculated differently, depending on your inverse kinematic model. The most relevant section in this regard is probably  https://github.com/MarlinFirmware/Marlin/blob/fa5bfca3678ef6e695e8e5d4a51af2bcd02e8e85/Marlin/src/module/planner.cpp#L2080-L2111. For more info regarding reverse kinematics, see
https://xyzdims.com/2021/02/08/3d-printing-penta-axis-pax-5-axis-printing-option/

HendrikJan-5D, 2021-11-08:
Thanks for the info.

Still, the safe probing is not going well.

I have entered the following test values

[image]

At G29 goes

Axis A ---> 90
Axis B ----> 0
Axis C ----> 0
Axis U ----> 100

Then the Axes A and U go to 0
And start the actual bed leveling

See the log report.

[Test.report.Safe.Probing.Position 2 .txt](https://github.com/DerAndere1/Marlin/files/7499726/Test.report.Safe.Probing.Position.2.txt)

---------------------------------------------------------------

I also want to report an older problem that has I not yet been reported to you.
From the Z axis, the HOMING_BUMP_MM function does not work.
The Z axis stops immediately at the limit switch.

[image]

DerAndere1, 2021-11-08:
Thanks for testing other values than 0. Otherwise I wouldnt have seen that bug. I hope it does not move the IJKUVW axes to 0 anymore when you try this change:
https://github.com/DerAndere1/Marlin/blob/a8eae7a080f925da941f03e15d7df23740c1f57f/Marlin/src/module/probe.cpp#L768
change that to
```cpp
  xyz_pos_t npos = LINEAR_AXIS_ARRAY(
    rx, ry, TERN(DELTA, _MIN(delta_clip_start_height, current_position.z), current_position.z),
    current_position.i, current_position.j, current_position.k,
    current_position.u, current_position.v, current_position.w
  );
```
For the issue with HOMING_BUMP_MM, i will try to reproduce on my machine. Maybe it is an isssue with one of your z limit switches (see MarlinFirmware/Marlin issue 20406 "[BUG] Homing bump doesn't work if both Z_MIN and Z_MAX enabled"). You can find out if one of your z endstops is triggering wrongly by enabling `PINS_DEBUGGING` in Configuration_adv.h and sending M43 E before homing. (https://marlinfw.org/docs/gcode/M043.html)

HendrikJan-5D, 2021-11-10:
I copied your change to probe.cpp.
And enter the following values for test.

//#define SAFE_BED_LEVELING_START_X 0.0
//#define SAFE_BED_LEVELING_START_Y 0.0
//#define SAFE_BED_LEVELING_START_Z 0.0
#define SAFE_BED_LEVELING_START_I 45.0
#define SAFE_BED_LEVELING_START_J 45.0
#define SAFE_BED_LEVELING_START_K 45.0
#define SAFE_BED_LEVELING_START_U 45.0
//#define SAFE_BED_LEVELING_START_V 0.0
//#define SAFE_BED_LEVELING_START_W 0.0

Under these circumstances, the actual probing cannot take place.
But everything now works as expected.
The problem is solved.
If I set the values for I-J-K-U to 0, the bed leveling goes well.

See picture:
All four axes are rotated 45 degrees here and the probe is activated.

[image]

The HOMING_BUMP_MM issue has also been resolved.
I had activated the Z min plug.
This one was not connected

//#define USE_ZMIN_PLUG

Another general question.
This question keeps popping up in VisualStudio.

Do you want to install the recommendation extensions for C++ ?

Do I need to install this??

DerAndere1, 2021-11-10:
That's great. I will apply the same fix in the 10axis_PR1 branch.

> Do you want to install the recommendation extensions for C++ ?

No, those exensions are meant to make developing software in C++ a bit easier, but if you use VisualStudio only for configuration and compiling, you don't need them.

HendrikJan-5D, 2021-11-11:
On your wiki page you can add the following tested parts.
Two stepper drivers per Axis (applied for the Y Axis)
#define AUTO_BED_LEVELING_LINEAR ( this is what I'm using now )

To cool the stepper drivers on the motherboard I connected a fan to the FAN2_PIN.
The extension board BTT M5 also has step drivers and connectors for extra fans.
Is it possible to control 2 PIN addresses with
#define CONTROLLER_FAN_PIN.
e.g. FAN2_PIN and FAN3_PIN

see print screen Configuration.adv.h

[image]

DerAndere1, 2021-11-14:
I've added those to the wiki. I only plan to list features that have to do with axes in the wiki, so I won't mention the fans. It's never the less nice that you manage to improve your machine with additional features. In the meantime opened a feature request: Marlin firmware issue "[FR] rotational axes with feedrate calculation like in NIST RS274 NGC Interpreter - version 3". Do you think that  the proposed interpretation of feedrate is in accordance with industry standard multi axis CNC machines?

HendrikJan-5D, 2021-11-14:
I have no knowledge of this complex matter.
So far I have configured the axes I - J - K - U in degrees.
And also controlled in degrees via G codes, which worked well until now.
Because all axes arrived at the same time at the end coordinate.

HendrikJan-5D, 2022-01-07:
DerAndere1.

This for your information.

About multi Axis slicer software.
I received this from Guoxin Fang / TU Delft.

I also send you a poster which has been published by Hcc!3d.
However, this information is in Dutch.

[Info Tu + Poster.zip](https://github.com/DerAndere1/Marlin/files/7828413/Info.Tu.%2B.Poster.zip)

DerAndere1, 2022-01-11:
That is very very good news. and the poster looks great. Thanks for contacting TU Delft! I spread the word about Guoxin Fang's slicer toolchain in the Marlin developer's communication channel on discord. 
In the mean time I fixed another bug related to INCH_MODE_SUPPORT. Also, I verified your observation: the DEFAULT_MAX_FEEDRATE correctly limits the feedrate of multi-axis moves. The latest version can be obtained from https://github.com/DerAndere1/Marlin/tree/9axis_PR2. If you also need CONTROLLER_FAN2_PIN, I am preparing a new release of Marlin2ForPipetBot at https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_edge . Would be great if you could test that branch with the attached updated configs for your printer. Here is a g-code script to test things:
```
G1 X50 Y50 Z50 A0 B0 C0; safe position
G91; incremental movement mode
G20; inch mode
G1 A4; rotates 4 degrees
G1 B4
G1 C4
M114
G1 A41 F3600; rotates A axis another 41 degrees at a feedrate of 3600 degrees per minute
G1 X2 A-20 F10; moves X axis 2 inches at a feedrate of 10 inches per minute. Simultaneously rotates A axis 20 degrees back
M114
G21; back to sanity with metric mode
G1 A-2; rotates 2 degrees back
G1 B-2
G1 C-2
M114
G90; absolute movement mode
```
[Marlin2ForPipetBot_HendrikJan-5D.zip](https://github.com/DerAndere1/Marlin/files/7848707/Marlin2ForPipetBot_HendrikJan-5D.zip)

DerAndere1, 2022-03-09:
I have updated Marlin2ForPipetBot and will now maintain configs for your setup as an example for a multi axis printer at https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot/config/examples/multi_axis_3D_printer/HendrikJan-5D. Let me know whenever you have suggestions for changes.

HendrikJan-5D, 2022-03-11:
I have looked at the configs.
These are in order.
I use this version but then modified with the Fan 2 function.
Which I will send along as a complete zip file.

In the meantime I have done some hardware testing and manual slicer testing.
I ran into a possible problem of bed leveling.
If I have done a bed leveling.
And then turn the bed for example 90 degrees around.
Turns the bed 90 degrees.
But according to me, the measured bed level pattern remains still.
And this should also rotate.
I'm still not quite out.
What is your vision on this.

I also played a little with the joystick function in Marlin.
It is very difficult to set and poorly documented.
I have a hopefully better joystick on order at Aliexpress.
Would this be something to extend this function to 9 Axis.
To e.g. apply in a robot arm configuration.

[Marlin-9axis_PR2-7axisconfig Fan2 Pin.zip](https://github.com/DerAndere1/Marlin/files/8234929/Marlin-9axis_PR2-7axisconfig.Fan2.Pin.zip)

DerAndere1, 2022-03-12:
> Turns the bed 90 degrees.
> But according to me, the measured bed level pattern remains still.

Currently bed leveling only does the right thing if the rotational axes are in neutral position. IMHO, to do the right thing in case of rotated axes, one has to implement kinematics for your machine in the firmware. bed leveling has to be applied to the Z axis in cartesian space and then inverse kinematics have to applied to calculate the corrected joint (motor) positions. When the bed rotates around the rotation axis of the motors A and B, then the bed levelling data must be rotated when the A and B axes but not when other axes are moved. In other machines, the bed may not rotate around the same motor axes. This is handled by kinematics. 

> I have a hopefully better joystick on order at Aliexpress.
Would this be something to extend this function to 9 Axis.

Not sure how control of more than 2 axes with a joystick is typically handled. You might just write a host application that reads joystick inputs and sends G1 commands to your liking. Here is how to read joystick input into a wx.Python app: https://gist.github.com/robsears/3943374

DerAndere1, 2022-03-18:
A workaround for the missing kinematics in Marlin would be to perform the bed probing with G29, then disable bedleveling in firmware with `M420 S0` and send the bed leveling mesh data to the host software with `G29 S0` or `M420 V1`. Then use that data in the post processor / slicer. Not a good solution, IMHO.

DerAndere1, 2022-04-28:
9 axis support was merged into upstream MarlinFirmware/Marlin. I will maintain the Marlin2ForPipetBot fork which in addition has support for the second controller fan pin. I close this thread because it is no longer focussed on a single subject. whenever there is something to discuss, please open seperate issues (one per topic) at https://github.com/DerAndere1/Marlin/issues. Happy printing!

Eljanaliyev, 2022-05-04:
> To support rebasing 9axis_pull branch onto current upstream MarlinFirmware/Marlin/bugfix-2.0.x
>
> Have I configured Marlin-9axis_pull for 8 axes And tested with a Dummy hardware shaft. All functions [...]

hello dear. Sorry I didn't read all discussion. hopefully your problem has been solved. would you please tell me  the name of stepper motor library if you use one.
I am working on a project which has 6 stepper motor( one for each axis) I have to be able to drive them at the same time with different direction or speed. 
On arduino IDE accelstepper library is brilliant for such projects.
But I have to use Raspberry pi 3B+ for my project because I will need some other opportunities. and I will code in C/C++ programming language.  if I'm not mistaken you code in C++ in this project. and you have used stepper motors. Do you use any library for Stepper motors?
I have been started learning Python. Because i haven't find good stepper motor library for such project in C/C++.

HendrikJan-5D, 2022-05-07:
Hello,
Marlin does not use a stepper motor library like accelstepper.
Marlin is a uiniversal applicable program that can control stepper motors simultaneously using G codes.
To program this yourself is a very difficult job.
I would use Marlin with an arduino board e.g. Bigtreetech GTR V1.0 
and then control it through your Raspberry pi program.

Eljanaliyev, 2022-05-07:
Hello. Thank you to reply my email. I got it. But I have to create all program by myself.
There are some good stepper motor libraries but they are in Python language. I have no enough experience with python and  I am bad at it. I have to find a stepper motor library in C/C++ language.
Or I have to find out another solution for my project. Actually I have found one. To control stepper motors with 2 esp32. And create serial port between raspberry pi and they. But it is still expensive for me. it is my last choice.
Thank you dear. Have a good day.

DerAndere1, 2022-05-10:
the BTT Octopus boards are another example of a suitable board that Supports 6 Stepper drivers. 2x ESP32 sounds like over-Engineering. I would choose a Board with a Single 32 Bit microcontroller that has enough Pins to Control 6 Stepper Drivers. Preferrably, its main microcontroller is Supported be your preferred software framework (e.g. Zephyr, CMSIS, mbed, STcube HAL, or Arduino libraries). Have a Look  at the Library Collections and User Forums for these Frameworks. The PlatformIO 
IDE has a Built-in Library Manager with search Funktion

DerAndere1, 2022-05-12:
I would ask your supervisor, if it is ok if you put a microcontroller between Raspberry pi and stepper drivers. If it is not allowed, ask your supervisor if it is ok to use GPL licensed code and publish your code under the GPL license. In that case you can reuse Marlin code. Support for Raspberry pico is work in progress, see pull request 24042 "Support for RP2040 and BTT SKR Pico (WIP)" at https://github.com/MarlinFirmware/Marlin/pulls

Eljanaliyev, 2022-05-12:
Hello again. I am sorry to reply you late. I am a student on master degree. And I am  making 6 axis robot arm for dissertation. I haven’t planned all process. Because the dead line is in next year. But I am working on it. First I tried Arduino mega as a microcontroller and it worked enough for me. But when I was sending data from computer to Arduino via serial port I realised that Arduino stops the part of programming that control motors and goes serial communication part of program. I have mentioned this at last email. And I thought I need a controller that can do multitasking and there are some other tasks that I will need when time goes. Like camera control, perhaps Artificial intelligence, display, touchscreen ect. I want to control that robot without connecting to computer. That is why I want to use raspberry pi. But programming side is also hard for me right now. Because of I haven’t planned it. Right now exams are important for me. Therefore I will work on project after a month.
I also want to create a software that can help me to create G-code and a program that can create moving from G-code. There are to many problems to solve.
Thank you for your advices and help. Hopefully I will continue this project after a month. In my opinion I will contact you again. Or when I will finish my project I will inform you.
You don’t have to reply this email.
Thank you dear. Have a good day...

Eljanaliyev, 2022-05-12:
> A fast single core, 32 bit microcontroller is definively enough to control 6 steppers and perform serial communication as demonstrated by Marlin and other firmwares for 3D-Printers / CNC controllers. It should be possible to fix your existing program. Have a look at https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/stepper.cpp, and the files starting with serial* at https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/core and have a look at the developer documentation at https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/docs . But first, good luck with your exams!

Thank you

DerAndere1, 2022-10-09:
Reopened. In my [kinem2](https://github.com/DerAndere1/Marlin/blob/kinem2) branch, I have integrated inverse kinematics for 5 axis machines with a swivel head (B axis) and a rotating table ( C axis). This allowed me to add commands G43.4 and G43.9 to enable/disable rotational tool center point (TCP) control. If you are interested, you need to enable the new option `XYZBC_HEAD_TABLE`, and configure exactly 5 axes: XYZ, I (B axis) and J (C axis). Also, configure these new settings;
https://github.com/DerAndere1/Marlin/blob/88486498d2107cc54a38061fd038b2fd8730835a/Marlin/Configuration.h#L1017-L1027

For a printer with a single nozzle, you may set MRZP_OFFSET to a value that equals the distance between shaft of the B axis and the nozzle tip.

DerAndere1, 2022-10-19:
> Is it still possible to control Axis 6 and 7 with G codes.

Currently not. it will result in compile errors if you try to configure anything with more than X,Y,Z,I(B),J(C) axes. You will have to bring additional axes into neutral position manually before starting the newly configured machine. If you need it, I can add the ability to compile and auto-home additional axes 6 and 7. Note that while using TCP (G43.4), additional axes 6 and 7 must stay in neutral position. 

> If I turn the B Axis, for example by 45 degrees, is the Z Axis height also automatically adjusted.

When configured correctly and TCP enabled (G43.4), when you move rotational axes A or B, XYZ motors should compensate so that the tool tip stays at the programmed XYZ position relative to the bed.

> And also the automatic bed leveling function.

Bed leveling still does not work as intended when rotational axes are not in neutral position. In the long run, the plan is to make automatic bed leveling account for the kinematics, TCP is the first step towards that goal.

HendrikJan-5D, 2022-10-19:
Manually moving the bed into position is not a problem.
But my printhead is rotated 90 degrees wrong.
It is possible, but manually difficult to bring it into the correct test position.

[74 more replies]