DerAndere1/Marlin2ForPipetBot #59: Configure X2 Step/mm, Acceleration, Feedrate in IDEX mode.

mindfreak269, 2021-09-11:
How Configure X2 Step/mm, Acceleration, Feedrate in IDEX mode. since i,J,K can have independent setting, are this firmware can configure X2 on DUAL_X_CARRIAGE mode. in original marlin X2 will reference X1 setup.
My problem is i'm install Closed Loop stepper S42B at X1, and normal open loop stepper at X2. S42b have 102.4 step/mm and normal stepper 80 step/mm. X1 always give inacurrate move. any solution of this?

DerAndere1, 2021-09-17:
No idea how IDEX is supposed to work. Is there a specific reason why you use my fork instead of MarlinFirmware/Marlin bugfix-2.0.x?

DerAndere1, 2021-09-17:
Do I understand correctly that the original IDEX setups use MarlinFirmware/Marlin with `DUAL_X_CARRIAGE` enabled and for now that means it has two parallel axes in X direction which were were called X1 and X2 and that X1 and X2? Do I understand correctly that with that setup, you had to use the same steps/mm for X1 and X2? You mention axes I, J, K. If the axes I, J, K are the reason why you are looking at my fork, please use current MarlinFirmware/Marlin. It now contains my multi-axis support. You you can make use of the existing i-axis of MarlinFirmware/Marlin bugfix-2.0.x , but it is still not trivial. Citing my wiki: 

> If you have a different machine where additional axes contribute to tool positioning, you currently have to implement your own kinematic model or develop a post-processor that translates desired tool positions to compatible G-code.

mindfreak269, 2021-09-21:
> 
> 
> Do I understand correctly that the original IDEX setups use MarlinFirmware/Marlin with `DUAL_X_CARRIAGE` enabled and that means it has two parallel axes in X direction which are called X1 and X2? Do I understand correctly that with that setup, you had to use the same steps/mm for X1 and X2? You mention axes I, J, K. If the axes I, J, K are the reason why you are looking at my fork, please use current MarlinFirmware/Marlin. It now contains my multi-axis support. You can make use of the existing i-axis of MarlinFirmware/Marlin bugfix-2.0.x , but it is still not trivial. Citing my wiki:
> 
> > If you have a different machine where additional axes contribute to tool positioning, you currently have to implement your own kinematic model or develop a post-processor that translates desired tool positions to compatible G-code.

thanks for suggestion, and i have done same questin in marlin github, i forgot mechanism of IDEX is not only swap head for multi material, they also have duplication and mirroring feature use same signal from chip mainboard (only change direction pin for mirroring). and its not possible to have diferrent step/mm configuration for X1 and X2. only one solution, i'm buy another closed loop stepper for x2. 

thanks for advice,

closed 2021-09-21 by mindfreak269