DerAndere1/Marlin2ForPipetBot #60: setup 6 axis with dual y and b

rds11080, 2021-09-17:
hi, i try to use your marlin to setup 6 axis with x and y axis dual (x, y , z, a ,b, c, y2, b2) but i cant build the marlin firmware, i alway get 
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:605:26: error: static assertion failed: AXIS_RELATIVE_MODES must contain X Y Z I J K elements.

DerAndere1, 2021-09-17:
What do you mean by "x an y axis dual" setup. Does it mean 
A) two steppers connected connected in row (or in parallel) to one driver per axis
B) two steppers drivers per axis
C) IDEX setup / DUAL_X_CARRIAGE

Please follow https://github.com/DerAndere1/Marlin/wiki  
and if that did not help, attach the resulting configs here

rds11080, 2021-09-20:
hi, i got the dual y and dual z to work, i have another problem with sensorless homing, i cant get the A B C sensorless homing work, X Y Z sensorless working, when i run M501, i dont see A B C stallguard threshold, also when i set 
Marlin\src\module\motion.cpp:1739:23: error: 'I_MIN_PIN' was not declared in this scope; did you mean 'X_MIN_PIN'?
is there a setting for BTT OCTOPUS pin for  I_MIN_PIN???

`StallGuard threshold:
RECV: echo:  M914 X108 Y108 Z108 `

DerAndere1, 2021-09-22:
> is there a setting for BTT OCTOPUS pin for I_MIN_PIN???

by default, no. As described in the wiki, you have to define it and assign the pin number of the pin to which you connect the endstop limit switch for the I axis. SENSORLESS is not yet implemented for the I,J,K (A,B,C) axis.

rds11080, 2021-09-27:
https://github.com/MarlinFirmware/Marlin/issues/22649

closed 2021-09-27 by rds11080