DerAndere1/Marlin2ForPipetBot #65: Foam Cutter Display

rcKeith, 2022-01-31:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

rcKeith, 2022-02-01:
The error only happens when I define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER .  This is the LCD I have which does work with the default Marlin 2.0 code. It shows a error in  \Marlin-Marlin2ForPipetBot\Marlin\src\lcd\dogm\status_screen_DOGM.cppcannot open source file "U8glib-HAL.h" (dependency of "dogm_Statusscreen.h")
I wonder is there some dependency I need to add somewhere
Thanks
Keith

hobiseven, 2022-02-01:
@rcKeith 

Hi. I made the initial tests a long time ago on the XY UV foam cutter, and I did write in planner.cpp a speed rule which is MAX ( speed(X,Y), speed ( U,V) )  to be used as the speed of cut for the max wire speed displacement. Have your tried this? 

Regarding U8glib, this is not specific to that branch of Marlin with the multiaxis. Should not.... Unless nobody else has tried to compile a multi axis version on something else than STM32. 

the piece of code \u8g_dev_tft_upscale_from_128x64.cpp is used to upscale by doubing pixels size to upscale from the small DOGM screen, and place a picture on a TFT RGB display 320x240. This is what we have on the Alfawise u30  STM32 board. For regular / standard black and white display, that should not be called. 

There might be a problem in the options you selected in config.h.

rcKeith, 2022-02-01:
Hi Hobiseven
Thanks for your reply.  I've only been testing on a Mega2560 not connected to any motors yet. So not tried your planner.cpp.
The reason for this is that I already have a GRBL Mega 5X compiled for the foam cutter https://rckeith.co.uk/grbl-hotwire-mega-5x-for-cnc-foam-cutters/  but it needs a Windows PC with the windows software I've adapted based on GRB-Panel.  Several guys have asked can we run the foam cutter like a 3d printer so I started some investigations which have led me here. 
Yes, it looks like I may have some configuration issues.  I'll look into it
Thanks
Keith

hobiseven, 2022-02-01:
Hi,
Well, what you are looking for is exactly what this option in Marlin can do. So this is the right website you ended up at! 
This will work in a standalone mode with if needed firmware in an SDcard reader, and you can actually configure the Gcode reader to use the letters you want to designate the axis systems. 

Regarding configuration, you need to be able to compile and then display on the 3 axis systems. Once that works, then I suggest to go for a 5 axis system, and do not use Z. I assume Mega 250 uses the regular DOGM display if I am correct. 

If that does not compile anymore, it would mean that there is an error in the definition of the menus/includes of the additional axis. This is possible. As I only compiled for a TFT screen, using pixel doubling, I never tried you configuration , and I do not know wether or not @DerAndere1 did try that on a Mega. He might tell us! 

Regards.

DerAndere1, 2022-02-01:
Hi everyone. @hobiseven: for context: Keith asked for a feature to display axis positions on the DOGM display. The error reported here is due to the fact that I just added that feature to the Marlin2ForPipetBot_edge branch but made some typos. 

@rcKeith:  This shoud be fixed now in Marlin2ForPipetBot_edge. The reason why I added the LCD_SHOW_SECONDARY_AXES feature to the Marlin2ForPipetBot_edge branch is because we need to test both, the newest planner, as well as the position display, anyway. When testing the edge branch, please do not modify planner.cpp for now. just `#define FOAMCUTTER_XYUV` . That already does what you want: see the code in Marlin2ForPipetBot_edge :

https://github.com/DerAndere1/Marlin/blob/e9f3a607186fa593a16b3930710798db3472a611/Marlin/src/module/planner.cpp#L2122-L2125

I will prepare a seperate pull request for Marlin with just the LCD_SHOW_SECONDARY_AXES feature after everything in Marlin2ForPipetBot_edge works as expected.

DerAndere1, 2022-02-01:
@hobiseven nice to see that you are still active here. If you are interested, you can also test the https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_edge branch. It contains a mayor overhaul of the planner that now works similar to LinuxCNC for other multi-axis machines. I hope, FOAMCUTTER_XYUV still works as before. You will have to rename `LINEAR_AXES` to `NUM_AXES` and make sure the new options AXIS4_ROTATES and AXIS5_ROTATES are **not** defined. Of cause, also add `#define FOAMCUTTER_XYUV`. Don't feel pushed, though.

DerAndere1, 2022-02-01:
@rcKeith: When the other people asked "can we run the foamcutter like a 3d printer" I hope they dont expect a slicer to outpt multi-axis G-code. That does not exist. You still have to use a dedicated CAM software to generate the G-code. Also, I am not aware of any G-code sender (host software) that has controls for 4 axes (besides yours). So, using Marlin with the usual G-code senders like pronterface or octoprint will be the same experience as using grbl with G-code senders for 3 axis CNC machines (e.g. UGS https://winder.github.io/ugs_website/). It works, but only through G-code, not through buttons.

rcKeith, 2022-02-01:
Hi DerAndere1
No, they don't. It's usually because the foam cutter is quite a big machine in an outside workshop where they don't want to leave a laptop or a PC.  I have several posts and videos showing how to generate g-code for foam cutters.
Thanks for all your work on this.
Keith

DerAndere1, 2022-02-01:
Understood. customizing octoprint should be possible. I just saw this plugin for adding custom controls: https://community.octoprint.org/t/adding-my-own-buttons-to-the-control-tab-system-commands-not-gcode/4753

rcKeith, 2022-02-01:
Yes, I had thought about that.  I use it on my CR-10 V2 it works well on that.
Keith

DerAndere1, 2022-02-01:
btw. Keith, is it ok if I add a link to the download section of your home page in my wiki or blog? I like to gather links to software for multi-axis CNC machines.

rcKeith, 2022-02-01:
Yes that fine, thank you

DerAndere1, 2022-02-01:
Another combinatiion that might be worth investigating is [grblHAL](https://github.com/grblHAL/core) (the succesor of grbl that supports many boards and up to 6 axes) with https://pypi.org/project/bCNC/ (GUI for 3 or 6 axes). octoprint also has support for grbl: https://plugins.octoprint.org/plugins/bettergrblsupport/ .  feedrate interpretation might not be as good as with Marlin's FOAMCUTTER_XYUV feature, though.

rcKeith, 2022-02-02:
Hi 
Having these issues when trying to compile with status_screen_DOGM.cpp
I have REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER defined in Configuration.h
Tried my best to figure it out but no luck.  Any help is greatly appreciated.
[image]

Keith

DerAndere1, 2022-02-02:
My bad again. hope it compiles now. Sorry, I dont have the time to test compilation.

rcKeith, 2022-02-02:
Thanks, DerAndere1.  Compiles fine thank you. Will update you when I have a working system.
Keith

DerAndere1, 2022-02-03:
Can you Change that erroneos Line to 
```
#if ENABLED(LCD_SHOW_SECONDARY_AXES)
```
? Im not at my PC right now

rcKeith, 2022-02-03:
Hi DerAndere1
That did it, thank you very much
Keith

DerAndere1, 2022-02-06:
Can you attach a photo of how the status screen looks `like with `LCD_SHOW_SECONDARY_AXES` when you have your setup ready? When I have seen the result, I could then further develop this feature to get some of the icons back and squeeze the positions of secondary axes into one small corner of the screen,

rcKeith, 2022-02-06:
Hi 
Yes, no problem it's going quite well. I have been adding some menus and hiding the Z-axis.  Will get some pictures out today or tomorrow.
Keith

rcKeith, 2022-02-06:
Hi DerAndere1
Photo attached of the status screen
[image]
Keith
.

DerAndere1, 2022-02-07:
When you have time, Can you test the latest Marlin2ForPipetBot_edge ? With your existing configs (LCD_SHOW_SECONDARY_AXES enabled), it should now show the positions for the extra axes in the top right corner instead of the fan icon. This should now allow other icons to be displayed.

hobiseven, 2022-02-07:
@rcKeith 

So right now, it seems that you have all compiled and working fine on a Mega2560, is this correct? You also have this LCD display with a few buttons, correct? I will have to also recompile on the Mega, as my Foam cutter was initially using the mega + a ramps all running with the Jedicut plugin. It will be simpler for me to reuse the Mega, although I did debug the code initially only on an STM32... Do you have a git accessible with your config please? 
I have to cut new foam profiles ... 
Do you also manage the heat percentage?

rcKeith, 2022-02-07:
Hi
Yes compiled and working on the Mega2560 but haven't actually connected any
motors yet to confirm pinouts. Hopefully later this week. The LCD is
working with a few custom menus and I have updated the language file as
well.  I've created a GitHub repository here
https://github.com/rcKeith/MarlinFoamCutter but haven't added those changes
from your last email.  We've been to the vet's this morning with our dog.
Had a few problems with him but he's fine now,
The heat control is via the D8 connector on the RAMPS board which is the
same as the spindle control so I've set that as a percentage reading and
altered the language file to say Heat control instead of Spindle.
I haven't used GitHub a great deal so hopefully, the repository is OK. I'll
add those changes later today.
Thanks
Keith

https://rckeith.co.uk <http://www.rckeith.co.uk>

rcKeith, 2022-02-07:
Hi DerAdnere1
Have an error at line 741 in dogm_StatusScreen.h 
It looks like it's trying to reference a macro NEITHER which doesn't; seem to be defined.
Keith
[image]

DerAndere1, 2022-02-07:
Ah you have been too fast :D I corrected that error just after asking you to test. Should be fixed already

DerAndere1, 2022-02-07:
Regarding you repo: it is ok to upload a local copy of the source code to your repo, but you lose the old commit history, so it will become more difficult to track authorship and to update your code with changes from upstream. Therefore, it is usually recommended to 
1. fork my repo https://github.com/DerAndere1/Marlin on github: https://docs.github.com/en/get-started/quickstart/fork-a-repo
2. , then `git clone` your new remote repo to get a local copy under version control that preserves the whole commit history
3.  Create a new local branch (e.g. called master) from the Marlin2ForPipetBot_edge branch. 
4. Switch to the master branch (`git checkout master`). Make your modifications and commit your changes to that new branch. 
5. `git push` the changes to the online repository
6. You can then go to the settings of your github repo and change the default branch to master: https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository
7. You can delete your current repository on github and rename your new repository later: https://docs.github.com/en/repositories/creating-and-managing-repositories/deleting-a-repository . The warnings can be ignored if only you contributed changes after the forking.

rcKeith, 2022-02-08:
Hi
Followed the instruction you sent and just one repository forked from
yours. Thanks
Keith

https://rckeith.co.uk <http://www.rckeith.co.uk>

DerAndere1, 2022-02-08:
Your master branch right now was created from the Marlin2ForPipetBot branch. If you want to base your modifications on Marlin2ForPipetBot_edge , one strategy is to 
1. If needed, commit any work in progress on the current master branch
2. Make another branch the default branch in github
3. [rename](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository/renaming-a-branch) the current master branch to e.g. `oops`
4. Create a new master branch based on Marlin2ForPipetBot_edge (short: `git checkout -b master Marlin2ForPipetBot_edge`). Long version:
   1. Switch to the Marlin2ForPipetBot_edge branch (`git checkout Marlin2ForPipetBot_edge`) !!!
   2. Create a new master branch based on Marlin2ForPipetBot_edge (`git branch master`)
   3. Switch to the new master branch (`git checkout master`)
5. if needed, `git cherrypick` your own commits from the oops branch to the master branch
6. delete the `oops` branch

hobiseven, 2022-02-08:
Hi folks,

I will really need to recompile the foamcutter to go on a Mega + ramps + display, same as apparently rcKeith is working on. 
I made a fork out of his repo, but apparently I will have to wait a bit to get things stabilized....

DerAndere1, 2022-02-08:
@hobiseven: That will be fixed once the steps I recommended above are done.

rcKeith, 2022-02-08:
Hi DerAndere1
I think I have Github sorted now.  I've just committed Configuration.h and Configuration_adv.h so far. But I'm getting a strange error with 
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:34:0: error: unterminated #if
 #if HAS_MARLINUI_U8GLIB && DISABLED(LIGHTWEIGHT_UI)

Yet there is an #endif at the bottom of the file. Spent quite a bit of time today trying to fix it but no luck
Keith

DerAndere1, 2022-02-08:
this is fixed with the following change: https://github.com/DerAndere1/Marlin/commit/21f53706951d76b59b9130973a47a7163016624f

to update your master branch you may have to 
Switch to your master branch:
```
git checkout master
```
and add https://github.com/DerAndere1/Marlin as a remote (lets call it upstream).
`git remote add upstream https://github.com/DerAndere1/Marlin.git`

After that, you can update your repo:
Make sure you have checked out your master branch, then fetch changes from remote repositories: 
```
git fetch --all
```
Then merge (or rebase) upstream/Marlin2ForPipetBot_edge into the currently checked out master branch:
```
git merge upstream/Marlin2ForPipetBot_edge
```

Followed by a push to your online repo:
```
git push
```

How to do this in VSCode + Gitlens extension is described at https://derandere.gitlab.io/version-control#:~:text=rebase%20current%20branch%20onto%20a%20branch%20from%20a%20different%20repository

DerAndere1, 2022-02-08:
@hobiseven the [master branch of the Marlin fork by rckeith](https://github.com/rcKeith/Marlin/blob/master/Marlin) now is by default configured for a hot wire foamcutter based on the Arduino Mega with RAMPS v1.4 . Happy testing!

hobiseven, 2022-02-09:
Hi! Thanks. I will fork the git and start to play! Ready to eat jedicut gcode! 
@rcKeith did you had a look in any tuning of octoprint to have a basic vertical cutter “guillotine” and 4 axis jog buttons?

rcKeith, 2022-02-09:
> Hi! Thanks. I will fork the git and start to play! Ready to eat jedicut gcode! @rcKeith did you had a look in any tuning of octoprint to have a basic vertical cutter “guillotine” and 4 axis jog buttons?

Hi Olivier
I would hold off for a bit as I haven't tested it on any hardware yet.  Haven't looked at octoprint yet but I will later.

hobiseven, 2022-02-09:
Ok! If you have not modified marlin core and only made screen changes that s ok… if you made critical changes in the core itself then let me know. I stay tuned. I plan to restart cutting foam in 2/3 weeks .

rcKeith, 2022-02-09:
Hi DerAndere1
Almost there on the status screen. Photos attached.  I've tried adjusting a few things but made no difference
This one is with LCD_SHOW_SECONDARY_AXES_LINE
[image]
And this with LCD_SHOW_SECONDARY_AXES_LINE
[image]
.

DerAndere1, 2022-02-09:
I left out the axis letters on purpose to save space. I have no Idea why the position values / "??" for the V axis are trunkated. It looks like they are drunkated at the hight of the icon's bottom. I need a photo with `LCD_SHOW_SECONDARY_AXES` enabled where actual positions values above 100 mm are shown (after homing) to see how much more I can squeeze it together.

DerAndere1, 2022-02-09:
Also: I need a photo of the screen with none of LCD_SHOW_SECONDARY_AXES or LCD_SHOW_SECONDARY_AXES_LINE enabled to see if there is more space at the top left corner.

rcKeith, 2022-02-09:
Hi DerAndere
Yes, no problem I'll get one later today. Thanks for all your help
Keith

https://rckeith.co.uk <http://www.rckeith.co.uk>

rcKeith, 2022-02-09:
OK, I'll do that as well.
Keith

https://rckeith.co.uk <http://www.rckeith.co.uk>

rcKeith, 2022-02-09:
Hi DerAndere
The first photo is with axis over 100mm
The second is with both secondary commented out in configuration_adv.h
[image]
[image]

DerAndere1, 2022-02-09:
I tried to reduce whitespace even more and added an option to show the position in the top left.
Would be nice to get some feedback on the following combinations. no hurry.
- enable only LCD_SHOW_SECONDARY_AXES
- enable both, LCD_SHOW_SECONDARY_AXES, SHOW_SECONDARY_AXES_TOP_LEFT
- enable only LCD_SHOW_SECONDARY_AXES_LINE

rcKeith, 2022-02-10:
Hi DereAndere
Getting this error to do with the macro TERN
[image]
Thanks
Keith

hobiseven, 2022-02-12:
Hi folks 

Maybe this is something that was already done , but I ask as I am not sure. 

Can we create a default 4 axis xyuv + wire heat management configuration and get this merged in the marlin release? We could call it foamcutter_xyuv_ramps .... this way no more need to download forks from your repos.... 

Thanks!

DerAndere1, 2022-02-12:
@hobiseven: we can think about that once this all works as expected. Please test rcKeith's config with my Marlin2ForPipetBot_edge branch and send a picture of the status screen. If you do not need the LCD_SHOW_SECONDARY_AXES feature, there is a foamcutter config since 2021: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/FoamCutter . If you have improvements or if there exists a project page for a community around a machine that needs different configs, feel free to open a pull request at https://github.com/MarlinFirmware/Configurations .

rcKeith, 2022-02-12:
Hi
I've been working on some other projects for a couple of days but I have
some more updates to come probably tomorrow. I've also managed to test some
spare stepper motors connected to the board and all worked fine after some
tweaks to the acceleration settings. The LCD_SHOW_SECONDARY_AXIS_LINE seems
to be the best option. Using the other version still truncates the bottom
characters.  Thanks for your help and support. Hopefully next week I can
test my foam cutter for real.
Keith

https://rckeith.co.uk <http://www.rckeith.co.uk>

rcKeith, 2022-02-16:
Hi DerAndere
I've been testing today on my foam cutter using the Arduino Mega 2560 and the RAMPS board.  All works well except for homing. After some investigation and connecting a stepper motor to the Z-axis(which we don't use) when a G28 is issued it tries to home Z. It seems like it will always home Z first even though we don't have a Z-axis.
I've looked and the code and can't see where I can disable this if I can?

Would it be possible to change the planner.cpp from  
 _MAX(sq(steps_dist_mm.x) + sq(steps_dist_mm.y), sq(steps_dist_mm.i) + sq(steps_dist_mm.j))
to 
 _MAX(sq(steps_dist_mm.x) + sq(steps_dist_mm.y), sq(steps_dist_mm.z) + sq(steps_dist_mm.i))

so the Z-axis could be used. This would need a NUM_AXES changing to 4  for the FOAMCUTTER_XYUV
in a few places.
Thanks
Keith

hobiseven, 2022-02-16:
Hi

I did the initial change in the planner and as far as I remember no issue to change the code as you propose. I also got the g28 issue back then but never really looked at it. 

Alternatively you can change your g28 to g28 x then g28 u and after that g28 y and g28 v.

Also there might be a piece of code called g28.cpp, or at least a piece of code where g28 is coded. Using the #ifdef foamcutter_xyuv you could write a custom g28. I think homing x, u , y and v would make more sense. 

Cheers

rcKeith, 2022-02-16:
Hi Olivier
I've just recompiled and with the changes and it seems to work OK. Yes, I have tried g28 x but it always homes Z first. I thought it was because Z_SAFE_HOMING was defined but I have that commented out.  I will do some more testing tomorrow.
Keith

DerAndere1, 2022-02-16:
Check if you have disabled HOME_Z_FIRST. In any case it will always first move the Z axis up a bit before homing other axes. This should not be a Z homing move though.

With NUM_AXES 4, the default G28 homing sequence is (tested on hardware): 
1. Z cleance move, 2. home axes XYZI

Maybe there is some magic happening with NUM_AXES 5

If for FOAMCUTTER_XYUV, the best sequence is XUYV, lets indeed modify file G29.cpp as suggested by hobiseven.

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send M502 and M500 to ensure your Configurations are applied.
- Issue the command M111 S247 to enable maximum logging.
- Perform a G28 to do your standard homing procedure.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a .TXT file and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

DerAndere1, 2022-02-17:
I implemented the homing sequence for G28 suggested by hobiseven in Marlin2ForPipetbot_edge

hobiseven, 2022-02-18:
@rcKeith Glad to hear that all is ok on your setup now. We might need to open an other issue for the hot wire. I used to control this using Jedicut + parallel port emulator in the Mega + ramps, so from a hardware point of view, that should be OK. Watchout for the voltage limit on the RAMPS mosfet. I have  a 48V DC PSU right now, and I will move that to an AC 220 chinese cheap solid state thyristor kind of thing. Let us know if you get proper PWM out of the RAMPS.  You might want to check the display also for PWM control...

rcKeith, 2022-02-18:
Hi Olivier
I don't think the hot wire will be a problem because it is used the same way as the GRBL Mega 5X I have on the website.  It's just treated as a spindle and works well on the RAMPS and MKS Gen L.  I've used it on all my projects for a while here's an example https://youtu.be/-nVzzA1CwF4
I don't use Jedicut I much prefer DevWing Foam 2 and for a free program, https://www.diyrcwings.com/app/ is much better in my opinion.
I'll keep you updated.
Keith

DerAndere1, 2022-02-18:
@rcKeith I would like to know if `G28 U` still works if you revert 
https://github.com/rcKeith/Marlin/blob/d5798334c88050fe2624778ec0d263be908a4598/Marlin/src/gcode/calibrate/G28.cpp#L438-L445
to back to
```
    if (z_homing_height && (NUM_AXIS_GANG(doX, || doY, || TERN0(Z_SAFE_HOMING, doZ), || doI, || doJ, || doK, || doU, || doV, || doW))) {
      // Raise Z before homing any other axes and z is not already high enough (never lower z)
      if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Raise Z (before homing) by ", z_homing_height);
      do_z_clearance(z_homing_height);
      TERN_(BLTOUCH, bltouch.init());
    }
```
?

rcKeith, 2022-02-19:
Hi DerAndere
I had reverted back I should have made that clear when I said it was all working.  So it's using XYUV axes.  Using the XYZU helped me see where the problem was and then I moved it back to XYUV.
So all good now. Thanks.
Keith

DerAndere1, 2022-02-19:
@rcKeith Great, so we have a working system that can serve as a baseline to benchmark further improvements. I am still trying to understand the code, Whenever you have time, could you try to copy the contents from my modyfied G28.cpp (raw file link: https://raw.githubusercontent.com/DerAndere1/Marlin/Marlin2ForPipetBot/Marlin/src/gcode/calibrate/G28.cpp ) into your current G28.cpp file to replace its content and test command `G28` (without specifying any axes)? It should home axes in sequence 1. X, 2. U, 3. Y, 4. V. Does it work?  If that is not what you want, you can then undo the change and reflash you current version afterwards, thx.

DerAndere1, 2022-02-19:
Ups. I updated the link in the above comment.

rcKeith, 2022-02-20:
Hi DerAndere
Homing of each individual axis does work already but I will try the updated
file. I think the current sequence is 1. X  2. Y  3. V 4. U

Keith

https://rckeith.co.uk <http://www.rckeith.co.uk>

[1 more reply]

closed 2024-06-14 by DerAndere1