DerAndere1/Marlin2ForPipetBot #67: 5-Axis really required on Foam Cutter?[FR] (feature summary)

ABLomas, 2023-01-16:
### Describe the feature you want

There are many 4-axis boards (at least on definition - creality 1.x series (AVR), MKS E3 and so on, so on) - on paper they look like perfect candidate for foam cutter.
But in reality code requires 5 axis, one of them is unused.
I tried to add fake definitions for mks robin e3 - a bit too many checks, can't even select the same pins for "fake axis".

There is another issue with endstops - foam cutters work perfectly without any homing endstops. Still, code requires some axis min/max definitions (this one needs more testing, may be related to my (poor) 5-axis hack).

While second one is related to safety - first one is clearly artificial. Can we get rid of this requirement?

DerAndere1, 2023-01-31:
It would be great if someone could prepare a pull request targeting upstream Marlin to compile without Y or Z axis. But it is a tedious task. For now much easier is to define a fake Z axis. You can have a look at my FOAMCUTTER_XYUV pull request to see how it can be done: https://github.com/MarlinFirmware/Marlin/commit/b06e378703d9046b01c24d73bfb647b08517a91f

DerAndere1, 2023-02-02:
My last answer was a bit incomplete. The default foamcutter example config for RAMPS relies on Marlins implicit pin-redifinitions. If I remember correctly, current Marlin redefines unused extruder pins as I axis STEP/DIR/ENAABLE pins. 
The following is a more general approach for a fake Z axis: 
1. identy one or more free pin(s) that can be configured as an output pin, e.g. pin 36 in the case of the RAMPS - based foamcutter by rcKeith. 
2. In your motherboard's pins_YOUR_MOTHERBOARD.h file, change the `Z_ENABLE_PIN`, `Z_STEP_PIN`, `Z_DIR_PIN` to that free pin:

```
#define Z_ENABLE_PIN 26
#define Z_STEP_PIN 26
#define Z_DIR_PIN 26
```

3. Change the  Z stop pins to -1:
```
#define Z_MAX_PIN -1
#define Z_MIN_PIN -1
```

ABLomas, 2023-02-02:
Yes, i did all this (on 32b board, too), this is simply additional job which is required to fool some checks, which are not required in first place ;-)
(btw, working good so far)

closed 2024-06-14 by DerAndere1